DIGImend/digimend-kernel-drivers #198: Huion Kamvas Pro 22 touch strip issue

not-surt, 2018-09-20:
I've got my Huion Kamvas Pro 22 set up and working pretty much perfectly with the exception that the touch strip produces garbage events.

Get a seemly random selection of multiple events per touch and up or down strokes don't seem to make much difference.

Attached xinput test output.
[touch_bar_up.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2403421/touch_bar_up.txt)
[touch_bar_down.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2403422/touch_bar_down.txt)

Further data attached here: #195

JedTheKrampus, 2018-10-08:
That's consistent with my experience.

spbnick, 2018-10-08:
What would be the functionality of those touchstrips and how do they work on Windows? I assume they're duplicate, not independent, and both produce the same input, at least?

not-surt, 2018-10-08:
Yeah, the buttons and touch strips are mirrored.
From the usbhid-dump output seems that when touched the 6th byte gives an absolute position value from decimal 1 to 23, 1 at top, 23 at bottom, 0 when not touched.

spbnick, 2018-10-13:
We can probably make it work, but I need to understand better how you want to use those touchstrips. So, do they work as mouse wheels (scrolling up/down), or do they produce key presses?

not-surt, 2018-10-13:
Well the device itself seems to just spit out an absolute touch position.

The windows driver allows mapping up strokes and down strokes.
The mapping will fire immediately once the finger starts moving and can fire multiple times during a stroke depending on how far along the strip the finger travels, like rolling a mousewheel multiple ticks.
You can slide finger up-and-down without releasing to fire a mix of the up and down mappings.

JedTheKrampus, 2018-10-13:
The touch strip events should map to Buttons or StripLeftUp/StripLeftDown events with xsetwacom. That way we can set them to do whatever we want, like changing brush opacity or size in Krita or rotating the canvas. Or, as you say, scrolling.

not-surt, 2018-11-13:
Updated mine with the new tilt supporting firmware (https://www.reddit.com/r/huion/comments/9vsscd/huion_kamvas_pro_22_just_got_tilt_update/) and recorded a usbhid-dump. Difficult to feel exactly when the pen tip will tilt off the screen so not sure how useful it will be.
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2575611/pen_tilt.txt)