DIGImend/digimend-kernel-drivers #200: Huion H420 doesn't work Ubuntu 16.04

Goysa2, 2018-10-01:
Hi I am completely new to this and I have no idea how to make my graphic tablet work.

I don't know what drivers I have and I don't have, I don't understand how to install and uninstall drivers. The only thing I am 100% is that the tablet didn't work "plug and play" style (nothing happened when I plugged it in, and nothing happens after hours of looking up solutions online).

I did download the latest DIGImend driver (v8) extracted it, did the make/ make install/ procedure. Didn't work. Tried to do `rmmod hid-huion ` and I get this error: ` rmmod: ERROR: Module hid_huion is not currently loaded `. And so I am completely lost and looking for guidance!

Thank you!

Goysa2, 2018-10-01:
Note: if I do `xinput list` the tablet doesn't show up

NuDogma, 2018-11-13:
Do you have SecureBoot enabled? If so, you cannot install kernel modules unless they are signed.

nullbasis, 2018-12-06:
I have the same problem. I installed v8 of the Digimend driver (.deb version in Software Center) and my Huion H420 doesn't work. How can I tell if I have SecureBoot enabled, and how can I disable it?

NuDogma, 2018-12-06:
In the BIOS setup of the computer, there is usually an option to enable/disable secure boot. If you put these words in a search engine (disable secureboot bios) you'll find more information.

nullbasis, 2018-12-07:
Turns out I didn't even have SecureBoot enabled. Still nothing. I guess it's back to the Googles for me.