DIGImend/digimend-kernel-drivers #201: Is the tilt functionality on new Huion tablets supported?

Cpt-Ktw, 2018-10-07:
Huion got pretty confusing with their naming scheme while updating their product line and unfortunately DIGImend page doesn't make it clear enough. 

Are the newer Huion tablet models with pen tilt functionality supported by DIGImend?
If not, can I expect that in the future?

spbnick, 2018-10-07:
Hi @Cpt-Ktw,

I'm interested in implementing this, but Huion has not agreed to pay me for work on this so far. They paid for supporting newer tablets in general, but not tilt support yet. I asked twice if they would, about two months ago, but got no answer. I'll try to raise them again.

See also #181

andresm, 2018-12-29:
@Cpt-Ktw , I have added tilt support for the Huion H610pro V2. I only tested it on a H610pro V2 but you can give it a try with others Huion tablets with tilt support:

https://github.com/andresm/digimend-kernel-drivers/commits/master

spbnick, 2019-05-05:
Latest development version of the drivers supports Huion tilt reporting.

closed 2019-05-05 by spbnick