DIGImend/digimend-kernel-drivers #202: Using xsetwacom Area param results in cursor stuck in lower left corner of the screen

JedTheKrampus, 2018-10-08:
I used xinput_calibrator to generate new Area parameters, but when I apply them to the stylus with xsetwacom all points on the tablet get mapped to the lower corner of the screen. Setting ResetArea returns the tablet to normal, but the small amount of parallax I was trying to deal with remains.

Example of a command that will cause the problem: `$ xsetwacom --set "HID 256c:006e Pen stylus" Area 75 95239 23 53302`

I'm using the latest deb release on a Huion Kamvas Pro 22.

In the meantime I will just have to lower my chair a bit so I'm looking at the tablet straight on.

Thanks for supporting the new tablets, by the way. Overall it's working great.

ThePreviousOne, 2018-10-28:
how did you get these numbers?
I haven't used `xsetwacom` for this before so I could be wrong but (assuming 1080p resolution) max X should always be 1919 and max Y 1079

except in this case https://github.com/linuxwacom/xf86-input-wacom/wiki/Area-mapping

ThePreviousOne, 2018-10-28:
nevermind found some actual examples. Using MapToOutput seem to be the correct way (or at least the easy way to do things https://askubuntu.com/questions/547990/how-to-configure-tablet-area-on-ubuntu-14-04

JedTheKrampus, 2018-10-28:
I got those numbers from xinput_calibrator. Using MapToOutput is better than nothing, but it's not ideal as there's still some parallax. If the Area parameter worked correctly with my tablet, I could wrap xinput_calibrator with a python script and set up a hotkey to run and apply the calibration to deal with the parallax that a display tablet introduces.