DIGImend/digimend-kernel-drivers #203: Huion WH1409, unable to install driver

Yann79, 2018-10-10:
Good evening,
I feel under the weather. I have bought a splendid Huion WH1409 pen tablet for my childrends. Unfortunately, I am not able to install Digimend drivers, neither on a Kubuntu 18.04 nor a Ubuntu 18.04 distro. On both PCs I even lost the out of the box good working functions coordinate & pressure (when connected by USB) after roll backs.

Starting from roll back, the actual situation on my Kubuntu PC is the following :

```
my@Saloon:~$ uname -a
Linux Saloon 4.15.0-37-generic #40-Ubuntu SMP Tue Oct 2 15:36:54 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

my@Saloon:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description: Ubuntu 18.04.1 LTS
Release: 18.04
Codename: bionic

my@Saloon:~$ lsusb
Bus 002 Device 002: ID 8087:8001 Intel Corp.
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8009 Intel Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 007: ID 256c:006e
Bus 003 Device 003: ID 058f:6366 Alcor Micro Corp. Multi Flash Reader
Bus 003 Device 002: ID 046d:08d7 Logitech, Inc. QuickCam Communicate STX
Bus 003 Device 005: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

my@Saloon:~$ xinput –list
⎡ Virtual core pointer id=2 [master pointer (3)]
⎜ ↳ Virtual core XTEST pointer id=4 [slave pointer (2)]
⎜ ↳ Logitech M315/M235 id=9 [slave pointer (2)]
⎜ ↳ PenTablet Pad id=12 [slave pointer (2)]
⎣ Virtual core keyboard id=3 [master keyboard (2)]
↳ Virtual core XTEST keyboard id=5 [slave keyboard (3)]
↳ Power Button id=6 [slave keyboard (3)]
↳ Power Button id=7 [slave keyboard (3)]
↳ Sleep Button id=8 [slave keyboard (3)]
↳ gspca_zc3xx id=10 [slave keyboard (3)]
↳ AT Translated Set 2 keyboard id=13 [slave keyboard (3)]
↳ PenTablet Pen id=11 [slave keyboard (3)]

my@Saloon:~$ dkms status
nvidia-340, 340.107, 4.15.0-37-generic, x86_64: installed
virtualbox, 5.2.10, 4.15.0-36-generic, x86_64: installed
virtualbox, 5.2.10, 4.15.0-37-generic, x86_64: installed
virtualbox, 5.2.10, 4.15.0-37-lowlatency, x86_64: installed

my@Saloon:~$ sudo evtest
[sudo] Mot de passe de my :
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0: Power Button
/dev/input/event1: Sleep Button
/dev/input/event2: Power Button
/dev/input/event3: AT Translated Set 2 keyboard
/dev/input/event4: Logitech M315/M235
/dev/input/event5: HDA Intel PCH Front Mic
/dev/input/event6: HDA Intel PCH Rear Mic
/dev/input/event7: HDA Intel PCH Line
/dev/input/event8: HDA Intel PCH Line Out Front
/dev/input/event9: HDA Intel PCH Line Out Surround
/dev/input/event10: HDA Intel PCH Line Out CLFE
/dev/input/event11: HDA Intel PCH Line Out Side
/dev/input/event12: HDA Intel PCH Front Headphone
/dev/input/event13: PenTablet Pen
/dev/input/event14: PenTablet Pad
/dev/input/event15: PenTablet Mouse
/dev/input/event16: PenTablet Keyboard
/dev/input/event17: PenTablet Consumer Control
[... 200 lines cut ...]
    MatchIsKeyboard "on"
    MatchDevicePath "/dev/input/event*"
    Driver "evdev"
EndSection

Section "InputClass"
    Identifier "Huion scroll"
    MatchProduct "TABLET"
    MatchIsPointer "off"
    MatchIsKeyboard "off"
    MatchIsTouchpad "off"
    MatchIsTablet "off"
    MatchIsTouchscreen "off"
    MatchDevicePath "/dev/input/event*"
    Driver "evdev"
EndSection
```

It looked pretty good up to now, but unfortunatly, I am not able to make devices appear :

`my@Saloon:~$ xsetwacom –list`

…still returns empty result… and tablet still doesn't give any sign of functionality coordinate or pressure…

What is the way to to finish this install ? Can you help please ?

DanielJoyce, 2018-11-06:
Yeah, something weird goin on. The latest driver fixed the right click issue on the XP-PEN for me, but everything else disappeared.

spbnick, 2018-11-10:
We never had WH1409 officially supported, but let's try making it work. Please don't install the modules using DKMS *only*, because that won't install the requisite udev rules and the rebinding script. Please download the latest master, follow the README.md, and either build and install the .deb package, or use `make dkms_install` or `make install`. Then unload the previously-loaded module with `modprobe -r hid_uclogic`, reconnect the tablet and post the output of `cat /sys/module/hid_uclogic/version` to verify that the latest version from master has loaded.

Also please follow what's going on in #217.

spbnick, 2018-11-17:
Please see the updates in #217.