DIGImend/digimend-kernel-drivers #204: XP_Pen 13.3v2, almost everything works except for button 3 on stylus sends BTN_TOUCH when clicked.

DanielJoyce, 2018-10-12:
That button appears to be sending BTN_TOUCH events along with button 1 ( the nib ). Could be a driver issue, or maybe a bad pen.

Hmm, going from the XP-Pen site, this might be intentional?

Everything else seems to work just fine.

DanielJoyce, 2018-10-12:
Also, stylus does not show up in ubuntu desktop Wacom Settings. Works, but I have to set it up manually via xsetwacom.

spbnick, 2018-10-13:
The desktop Wacom settings won't work with those tablets unless support for them is added to libwacom, and then it probably won't that well, because Huion reuses VID:PID pairs.

By "Button 3" do you mean the top side button on the pen?

DanielJoyce, 2018-10-13:
Yes, top half of rocker-button on stylus.

DanielJoyce, 2018-10-15:
[stream.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2480826/stream.txt)

Capture of button clicks, button 0 ( nib ) messages start with 07 A1, button two ( lower half of stylus rocker ) sends 07 A2, and button 3 ( upper half of stylus rocker ) sends 07 A4.

DanielJoyce, 2018-10-15:
It comes up as UGEE brand, 0x28bd 0x000b

DanielJoyce, 2018-10-25:
The product page for the XP-PEN 13.3 mentions easy one-click erasing. I'm thinking they have their tablet  firmware send something non-standard for third-button click, or they provide a wonky HID descriptor and then their driver handles it special. So it might be something as trivial as providing a HID override map in the driver, or special casing the button handling.

DanielJoyce, 2018-10-25:
Analyzing the second hid dump, I get
```
0x05, 0x0D,        // Usage Page (Digitizer)
0x09, 0x02,        // Usage (Pen)
0xA1, 0x01,        // Collection (Application)
0x85, 0x07,        //   Report ID (7)
0x09, 0x20,        //   Usage (Stylus)
0xA1, 0x00,        //   Collection (Physical)
0x09, 0x42,        //     Usage (Tip Switch)
0x09, 0x44,        //     Usage (Barrel Switch)
0x09, 0x45,        //     Usage (Eraser)  
```

So its saying it has an eraser but thats the second barrel button ( upper half of rocker ).

DanielJoyce, 2018-10-25:
So I may be able to improve my xorg mapping, and get it working as an eraser ( as it reports eraser ), or I can submit a patch to digimend and have it remap the hid descriptor I think...

spbnick, 2018-10-28:
Looking at stream.txt. I see absolutely normal button reports. If the driver initializes correctly, normal button clicks should be reported. However, I see that your tablet has a VID:PID pair that the driver doesn't know about and that is the problem, most likely.

I threw together a basic patch to try to support your tablet. Could you please try to download, install, and use this version of the driver to verify: https://github.com/spbnick/digimend-kernel-drivers/archive/add_xp_pen_13_3_v2.zip

Also, could you please post the output of `lsusb -v` for your tablet, and, please download, build, and run the fresh version of uclogic-probe from https://github.com/DIGImend/uclogic-tools/archive/master.zip Please post the output here.

Thank you.

spbnick, 2018-10-29:
@DanielJoyce well, what's in HID report descriptor is entirely to the kernel to interpret, and AFAIK, there's not much, if anything, to handle buttons like that in the kernel.

DanielJoyce, 2018-11-03:
I know, which I was wondering if the driver can remap the behavior.

DanielJoyce, 2018-11-03:
```
$ sudo lsusb -v -d 28bd:000b

Bus 003 Device 003: ID 28bd:000b  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x28bd 
  idProduct          0x000b 
  bcdDevice            0.00
  iManufacturer           5 XP-PEN
  iProduct                6 ARTIST 13.3
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           91
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower               70mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     139
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               2
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     140
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               2
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      59
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               2
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
Device Status:     0x0000
  (Bus Powered)

```

DanielJoyce, 2018-11-03:
```
$ sudo ./uclogic-probe  003 003
M 58 00 50 00 2D 00 50 00 45 00 4E 00
P 41 00 52 00 54 00 49 00 53 00 54 00 20 00 31 00 33 00 2E 00 33 00
S 64 0C 03 C0 72 8C 40 06 00 FF 1F EC 09
S 65 04 03 00 74
S 6E 04 03 30 00
S 79 28 03 46 00 36 00 30 00 20 00 5F 00 55 00 47 00 5F 00 56 00 41 00 52 00 31 00 33 00 5F 00 52 00 30 00 37 00 31 00 33 00
S 7A 28 03 46 00 36 00 30 00 20 00 5F 00 55 00 47 00 5F 00 56 00 41 00 52 00 31 00 33 00 5F 00 52 00 30 00 37 00 31 00 33 00
S 7B 28 03 46 00 36 00 30 00 20 00 5F 00 55 00 47 00 5F 00 56 00 41 00 52 00 31 00 33 00 5F 00 52 00 30 00 37 00 31 00 33 00
S C8 28 03 46 00 36 00 30 00 20 00 5F 00 55 00 47 00 5F 00 56 00 41 00 52 00 31 00 33 00 5F 00 52 00 30 00 37 00 31 00 33 00
S C9 28 03 46 00 36 00 30 00 20 00 5F 00 55 00 47 00 5F 00 56 00 41 00 52 00 31 00 33 00 5F 00 52 00 30 00 37 00 31 00 33 00
S CA 28 03 46 00 36 00 30 00 20 00 5F 00 55 00 47 00 5F 00 56 00 41 00 52 00 31 00 33 00 5F 00 52 00 30 00 37 00 31 00 33 00
```

DanielJoyce, 2018-11-03:
The device usb ids look the same, so the previous tablet config should work. xsetwacom list-devices no longer lists anything though.

DanielJoyce, 2018-11-03:
```
$ xinput
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ ELAN24EE:00 04F3:24EE                   	id=12	[slave  pointer  (2)]
⎜   ↳ DELL07E6:00 06CB:76AF Touchpad          	id=13	[slave  pointer  (2)]
⎜   ↳ SynPS/2 Synaptics TouchPad              	id=18	[slave  pointer  (2)]
⎜   ↳ XP-PEN ARTIST 13.3 Pen Pen (0)          	id=20	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Power Button                            	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ Integrated_Webcam_HD: Integrate         	id=10	[slave  keyboard (3)]
    ↳ Integrated_Webcam_HD: Integrate         	id=11	[slave  keyboard (3)]
    ↳ Intel HID events                        	id=14	[slave  keyboard (3)]
    ↳ Intel HID 5 button array                	id=15	[slave  keyboard (3)]
    ↳ Dell WMI hotkeys                        	id=16	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=17	[slave  keyboard (3)]
    ↳ XP-PEN ARTIST 13.3 Pen                  	id=19	[slave  keyboard (3)]

```

DanielJoyce, 2018-11-04:
It appears something has subtly changed because the old x config to set the wacom driver for the device no longer seems to work.

DanielJoyce, 2018-11-04:
```
[ 2104.427558] input: XP-PEN ARTIST 13.3 as /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb3/3-2/3-2.1/3-2.1:1.0/0003:28BD:000B.0003/input/input32
[ 2104.489908] hid-generic 0003:28BD:000B.0003: input,hidraw2: USB HID v1.00 Mouse [XP-PEN ARTIST 13.3] on usb-0000:39:00.0-2.1/input0
[ 2104.490564] input: XP-PEN ARTIST 13.3 as /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb3/3-2/3-2.1/3-2.1:1.1/0003:28BD:000B.0004/input/input33
[ 2104.491427] hid-generic 0003:28BD:000B.0004: input,hiddev0,hidraw3: USB HID v1.00 Device [XP-PEN ARTIST 13.3] on usb-0000:39:00.0-2.1/input1
[ 2104.492073] hid-generic 0003:28BD:000B.0005: hiddev1,hidraw4: USB HID v1.00 Device [XP-PEN ARTIST 13.3] on usb-0000:39:00.0-2.1/input2
[ 2104.494060] PKCS#7 signature not signed with a trusted key
[ 2104.494490] PKCS#7 signature not signed with a trusted key
[ 2104.494585] PKCS#7 signature not signed with a trusted key
[ 2104.537068] uclogic 0003:28BD:000B.0005: interface is invalid, ignoring
[ 2104.594337] input: XP-PEN ARTIST 13.3 Pen as /devices/pci0000:00/0000:00:1c.4/0000:03:00.0/0000:04:02.0/0000:39:00.0/usb3/3-2/3-2.1/3-2.1:1.1/0003:28BD:000B.0004/input/input34
[ 2104.662077] uclogic 0003:28BD:000B.0004: input,hidraw2: USB HID v1.00 Device [XP-PEN ARTIST 13.3] on usb-0000:39:00.0-2.1/input1
[ 2104.665856] uclogic 0003:28BD:000B.0003: interface is invalid, ignoring
```

Okay, I think its a secureboot issue maybe?

DanielJoyce, 2018-11-04:
```
$ xinput
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ ELAN24EE:00 04F3:24EE                   	id=12	[slave  pointer  (2)]
⎜   ↳ DELL07E6:00 06CB:76AF Touchpad          	id=13	[slave  pointer  (2)]
⎜   ↳ SynPS/2 Synaptics TouchPad              	id=18	[slave  pointer  (2)]
⎜   ↳ XP-PEN ARTIST 13.3 Pen stylus           	id=19	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Power Button                            	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ Integrated_Webcam_HD: Integrate         	id=10	[slave  keyboard (3)]
    ↳ Integrated_Webcam_HD: Integrate         	id=11	[slave  keyboard (3)]
    ↳ Intel HID events                        	id=14	[slave  keyboard (3)]
    ↳ Intel HID 5 button array                	id=15	[slave  keyboard (3)]
    ↳ Dell WMI hotkeys                        	id=16	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=17	[slave  keyboard (3)]
```

```
$ cat /etc/X11/xorg.conf.d/52-xp-pen-13.3V2.conf 
# XP-Pen 13.3 config

#Section "InputClass"
#    Identifier "XP-Pen 13.3 keyboard"
#    Driver "wacom"
    #MatchProduct: "ARTIST 13.3"
    #MatchVendor: "XP-PEN"
#    MatchDevicePath "/dev/input/event*"
#    MatchIsKeyboard "on"
#    MatchUSBID "28bd:000b"
#EndSection

#Section "InputClass"
#    Identifier "XP-Pen 13.3v2 pointer"
#    Driver "wacom"
#    MatchDevicePath "/dev/input/event*"
#    #MatchIsPointer "on"
#    MatchUSBID "28bd:000b"
#EndSection

Section "InputClass"
    Identifier "XP-Pen 13.3"
    Driver "wacom"
#    MatchProduct: "ARTIST 13.3"
#    MatchVendor: "XP-PEN"
    MatchDevicePath "/dev/input/event*"
#    MatchIsTablet "on"
    MatchUSBID "28bd:000b"
EndSection

```

DanielJoyce, 2018-11-04:
Pad and pad buttons no longer appear or work.

spbnick, 2018-11-10:
@DanielJoyce, thank you for the detailed information, but it's a bit hard for me to follow what comes after what. Let's try to back-track a little.

I see that the driver is loading, which is good. The "interface is invalid, ignoring" message simply says that some of the USB interfaces the tablet provides don't make sense to the driver, so it doesn't add them to the system. That should be OK so far, but let's see what the pad buttons actually do.

Did I get it right, that you have the pen working with the wacom driver and everything is fine with it, but the buttons on the side of the tablet don't work? Is there anything else that doesn't work?

Regarding the buttons, please run `sudo usbhid-dump -m 28bd:000b -es` with your tablet connected, and press the buttons one by one, top-to-bottom, with the tablet oriented to have the buttons on the left. Post the output here.

DanielJoyce, 2018-11-12:
Everything was working with the stock digimend drivers that came with ubuntu 18.04 except the 'right mouse click' button on the stylus ( top half of rocker button on the stylus ). It was all picked up by the wacom drivers and working except for that part of the rocker. I just had to create the proper xorg conf file to load the wacom driver, and the wacom config file.

With the patched drivers you gave me, the tablet and pen still work, but I can't get the wacom driver config to work anymore. None of screen buttons work either ( since wacom can no longer find the devices ).

DanielJoyce, 2018-11-12:
```
$ cat out.txt 
003:006:001:STREAM             1541997436.641599
 06 01 00 00 00

003:006:001:STREAM             1541997436.857522
 06 00 00 00 00

003:006:001:STREAM             1541997437.475545
 06 02 00 00 00

003:006:001:STREAM             1541997437.659706
 06 00 00 00 00

003:006:001:STREAM             1541997438.305575
 06 04 00 00 00

003:006:001:STREAM             1541997438.511588
 06 00 00 00 00

003:006:001:STREAM             1541997438.977734
 06 08 00 00 00

003:006:001:STREAM             1541997439.155773
 06 00 00 00 00

003:006:001:STREAM             1541997439.499721
 06 10 00 00 00

003:006:001:STREAM             1541997439.673697
 06 00 00 00 00

003:006:001:STREAM             1541997440.035659
 06 20 00 00 00

003:006:001:STREAM             1541997440.167804
 06 00 00 00 00
 
```

DanielJoyce, 2018-11-12:
```
sudo evtest 
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event2:	Sleep Button
/dev/input/event3:	Power Button
/dev/input/event4:	AT Translated Set 2 keyboard
/dev/input/event5:	Video Bus
/dev/input/event6:	SynPS/2 Synaptics TouchPad
/dev/input/event7:	Intel HID events
/dev/input/event8:	Intel HID 5 button array
/dev/input/event9:	Dell WMI hotkeys
/dev/input/event10:	ELAN24EE:00 04F3:24EE
/dev/input/event11:	DELL07E6:00 06CB:76AF Touchpad
/dev/input/event12:	HDA Intel PCH Headphone Mic
/dev/input/event13:	HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event14:	HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event15:	HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event16:	HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event17:	HDA Intel PCH HDMI/DP,pcm=10
/dev/input/event18:	Integrated_Webcam_HD: Integrate
/dev/input/event19:	Integrated_Webcam_HD: Integrate
/dev/input/event20:	XP-PEN ARTIST 13.3
/dev/input/event21:	XP-PEN ARTIST 13.3 Pen
/dev/input/event22:	XP-PEN ARTIST 13.3 Mouse
/dev/input/event23:	XP-PEN ARTIST 13.3 Mouse
```

DanielJoyce, 2018-11-12:
Uninstalling and using older version of kernel drivers that come with 18.04:
```
sudo evtest 
[sudo] password for daniel: 
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event2:	Sleep Button
/dev/input/event3:	Power Button
/dev/input/event4:	AT Translated Set 2 keyboard
/dev/input/event5:	XP-PEN ARTIST 13.3
/dev/input/event6:	XP-PEN ARTIST 13.3
/dev/input/event7:	Video Bus
/dev/input/event8:	SynPS/2 Synaptics TouchPad
/dev/input/event9:	Intel HID events
/dev/input/event10:	Intel HID 5 button array
/dev/input/event11:	Dell WMI hotkeys
/dev/input/event12:	HDA Intel PCH Headphone Mic
/dev/input/event13:	HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event14:	HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event15:	HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event16:	HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event17:	HDA Intel PCH HDMI/DP,pcm=10
/dev/input/event18:	ELAN24EE:00 04F3:24EE
/dev/input/event19:	DELL07E6:00 06CB:76AF Touchpad
/dev/input/event20:	Integrated_Webcam_HD: Integrate
/dev/input/event21:	Integrated_Webcam_HD: Integrate
```

Wacom mappings work again.

DanielJoyce, 2018-11-12:
Uninstalled digimend, grepping for 'hid', i see the following:
```
hid_multitouch         20480  0
intel_hid              16384  0
sparse_keymap          16384  2 intel_hid,dell_wmi
mac_hid                16384  0
hid_generic            16384  0
usbhid                 49152  0
i2c_hid                20480  0
hid                   118784  4 i2c_hid,usbhid,hid_multitouch,hid_generic
```

Doesn't appear to use uc-logic drivers at all. Or is it because they are compiled into the kernel? If only I could get right-click working... Is there some other maintainer I should contact?

Barracuda77, 2018-12-01:
@DanielJoyce What command do you run to get this?
Event: time 1539317779.526757, -------------- SYN_REPORT ------------
Event: time 1539317779.530752, type 4 (EV_MSC), code 4 (MSC_SCAN), value d0045
Event: time 1539317779.530752, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1539317779.530752, -------------- SYN_REPORT ------------
Event: time 1539317779.644894, type 4 (EV_MSC), code 4 (MSC_SCAN), value d0045
Event: time 1539317779.644894, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0

Barracuda77, 2018-12-08:
@DanielJoyce Did you manage to resolve this? I have the Artist 22E pro and facing the same issue.

DanielJoyce, 2018-12-11:
No, I'm just gonna make do with the rocker switch being slightly 'broken' for right now. The Digimend drivers provide worse support for this tablet than the kernel drivers which seem to work fine except for part of the rocker switch.

Barracuda77, 2018-12-11:
@DanielJoyce I managed to fix it. I am on Ubuntu 18.04 LTS I am using evdev drivers.
also, I am using a udev .hwdb file to remap my tablet buttons as described here. except my scan codes are slightly different and most probably yours will be too.
https://gist.github.com/loentar/1aa83e9e76fa34735133dc4d07eeb529

and to that file, I just added a command to remap that rocker as a right click. 
KEYBOARD_KEY_d0045=273

DanielJoyce, 2018-12-12:
Ahhh, awesome!!!!!! :D :D

You can add templates then to the xwacom to get gnome wacom mapping working too... 

I'll dig around on that half.

Is there an upstream place to submit the evdev mappings?

Barracuda77, 2018-12-12:
@DanielJoyce Interested to know more about the xwacom templates. update me please with that.
as for a place to submit evdev mappings am still looking.

DanielJoyce, 2018-12-12:
/usr/share/libwacom/layouts/

This where the templates are kept, stored as svgs.

https://github.com/linuxwacom/libwacom/tree/master/data/layouts

Following the above, just create a new file, add the mappings, and put in /usr/share/libwacom/layouts/

DanielJoyce, 2018-12-12:
You just need to remap the rocker switch in your evdev hwdb entry, then once you add a libwacom config file you can remap the buttons there if you want, providing a bit more flexibillity...

Would be nice if we could remap buttons per-application, outside of said application ( some don't support it inside )...

DanielJoyce, 2018-12-15:
Hmm, that right-click remapping isn't working. Even after sudo systemd-hwdb update
 and udevadm trigger

DanielJoyce, 2018-12-15:
Hmm, I can get d0045 to emit 273 but x doesn't pick it up.

DanielJoyce, 2018-12-15:
Apparently its bound to the keyboard and not the device and not the mouse, while the nib and lower rocker switch is bound to the mouse.

DanielJoyce, 2018-12-15:
Yeah, its reported on the wrong 'channel'

DanielJoyce, 2018-12-15:
YAY!

```
$ cat /etc/udev/hwdb.d/10-xp-pen-keyboard.hwdb 
evdev:input:b0003v28BDp000B*
 KEYBOARD_KEY_d0045=332
```

Works!

332 is BTN_STYLUS2, which is 'right-click' for tablets. The events are seperate but unified by X. This works! 

I noticed the lower half of the rocker sent 331, or 0x14b

```
#define BTN_STYLUS		0x14b
#define BTN_STYLUS2		0x14c
```

So I had the upper rocker button send 332 and it works! :)

Barracuda77, 2018-12-15:
@DanielJoyce Great you made it work. Very strange as I tried 332 first and it didnt work. so I used the same code the mouse is producing and it did. obviously, each model is a little bit different.

DanielJoyce, 2018-12-15:
What is are you using? I have Ubuntu using xfree86 with libinput and systemd.

Also I only override one code in hwdb. And I use libwacom to map everything else.

Barracuda77, 2018-12-17:
@DanielJoyce I am on Ubuntu 18.04 LTS using evdev. I am overriding all the keys with hwdb. but it is not optimum. very interested in how you use libwacom to map the rest. I mainly use Blender and 3D coat for sculpting and I use Krita occasionally for some quick sketching and rough paints.

Barracuda77, 2018-12-28:
@DanielJoyce I tried to run it using the Wacom driver but the Gnome UI didn't recognize it. it worked but I couldn't remap any keys using the ui tool as it didn't see it. although xsetwacom list did list it.

DanielJoyce, 2019-01-02:
You need to create a mapping file for libwacom to pick it up

Gist of what I am using

https://gist.github.com/DanielJoyce/84b9fb1780f82ed92766313d3d2292e3

Barracuda77, 2019-01-03:
@DanielJoyce Will try this. thanks a lot.

Barracuda77, 2019-01-07:
@DanielJoyce Hi, I have a question for you how did you get the scan codes in the last line for the .tablet file?
Thanks a lot for your help.

fahrstuhl, 2019-06-21:
Hi,
I had the same problem and tried to fix it with hwdb remapping, but it didn't work for me at all.

As already noted in this issue, the second button on the stylus gets reported as eraser by the USB descriptor. I adapted a [Digimend driver for my Artist 13.3](https://github.com/fahrstuhl/digimend-kernel-drivers/blob/master/hid-xppen.c) and changed the USB descriptor and the stylus buttons work perfectly now.

I wanted to go ahead and change the descriptor for the frame buttons, too, but couldn't figure out what the best change would be.

DanielJoyce, 2019-10-28:
Will try when I get home. have you put in a PR?

DanielJoyce, 2019-10-28:
Ahh, was wondering why some of my stuff broke when I reinstalled my laptop, forgot to install digimend. :D

fahrstuhl, 2019-10-28:
I have not put in a PR yet. Back then I couldn't figure out what the frame buttons should be mapped to best and haven't touched it since. Currently I remap the frame buttons on the fly using evdevremapkeys instead of doing it in the driver.

DanielJoyce, 2019-10-28:
Usually they should be handled by other parts of libwacom.

DanielJoyce, 2019-10-28:
Something changed in libwacom and things don't work quite as well with that driver...

DanielJoyce, 2019-10-29:
Yeah pad Switches no longer work, Ubuntu 19.04 based distro. Also tablet is now recognized as different devices including touch in their name, though touch is not supported. Will try taking new dumps....

DanielJoyce, 2019-10-29:
Stylus now shows up in Wacom settings manager and pad setup says there is a pad but it refuses to load the template file.

Since so much of this worked out of the box last time with vanilla Wacom will try there first.

DanielJoyce, 2019-10-30:
Ive come to th conclusion that none of these libraries do what I need. The wacom stuff and control panel assume only one config for all apps. It also seems very picky at device detection though everything seems to be there under evdev if I use that.

So I'm gonna investigate making my own tablet software like sc-controller, with an easy to set up flow, mapping keys per app, and uinput and other support.

fahrstuhl, 2019-11-01:
I think the best way to go for general support of the 13.3 is for me to create a PR for the eraser fix and add a [libwacom description](https://github.com/linuxwacom/libwacom/blob/master/data/wacom.example) of the tablet to get the buttons working, no change to the USB descriptor necessary.