DIGImend/digimend-kernel-drivers #210: Huion H430p disappears from xinput after first reboot

jvyden420, 2018-10-21:
OS: Ubuntu 18.10 x86_64 
Kernel: 4.18.0-10-generic
I installed the latest release as of 10/21/18. After the first reboot, the tablet disappears.
```
jvyden@jvyden-kubuntu:~$ xinput --list | grep HUION
⎜   ↳ HUION Huion Tablet Pad                    id=9    [slave  pointer  (2)]
    ↳ HUION Huion Tablet Pen                    id=8    [slave  keyboard (3)]
jvyden@jvyden-kubuntu:~$ 
```

Those 2 options dont work. I used to have a device showing up as id=26 before, but now that is simply just gone. What happened? What can I do?

jvyden420, 2018-10-21:
FYI, the tablet still works. I just need it to show up in xinput so i can make sure it doesn't use my second monitor.

jvyden420, 2018-10-21:
This is getting a little interesting. It reappeared while i wasn't paying attention. The hell is going on here?

jvyden420, 2018-10-21:
```
jvyden@jvyden-kubuntu:~$ xinput --list | grep HUION
⎜   ↳ HUION Huion Tablet Pad                    id=9    [slave  pointer  (2)]
⎜   ↳ HUION Huion Tablet Pen Pen (0)            id=26   [slave  pointer  (2)] # <-- what i want
    ↳ HUION Huion Tablet Pen                    id=8    [slave  keyboard (3)]
jvyden@jvyden-kubuntu:~$ 

```

spbnick, 2018-10-22:
@jvyden420, the devices will only appear after the pen is moved over the surface (with the libinput driver, IIRC). Could it be that you simply haven't touched the pen yet, at that moment?

spbnick, 2018-10-28:
I see. We'll have to track what's going on in `dmesg` and X.org log when this happens. If you see this happening again, please see if you can see any signs in there, and if you do, please post them here.

jvyden420, 2018-10-28:
Oops! forgot 'bout this.
Narrowed down the issue. It works after pressing buttons. Is this a thing that can be fixed in the driver?

jvyden420, 2018-11-07:
closing for inactivity

closed 2018-11-07 by jvyden420