DIGImend/digimend-kernel-drivers #211: Huion H1060P

CG-3D, 2018-10-23:
Hi all,

I recently got hold of a Huion H1060P. Works like a charm in KDE Neon using DIGImend v8 kernel drivers / xsetwacom, except for the tilt support, which should have been supported after updating the firmware.

I received the (Windows) firmware updater (H1060P_Tilt_20180808.exe) through Huion support. This updater however will not connect to the device ('connect Error') in Windows 10 Pro with the 14.7.6 drivers. I tried all sorts of combinations on 4 different systems (different cable, disabled Secure Boot, etc.) but to no avail.

Any info that could help me resolve or understand this problem would be appreciated greatly :).

CG-3D, 2018-10-24:
Update: I received a new updater (same name as the first) from Huion support and it worked fine. Tablet now supports tilt in Windows (Krita) just fine. If there is any way I can help to get tilt support in Linux, I would love to help.

diegogangl, 2018-11-07:
I just got a H1060p and can confirm it works like a charm. I haven't gotten the tilt working yet though. 
Using Fedora 29 / Kernel 4.18.16 / Digimend v8

diegogangl, 2018-11-13:
Looks like I talked too soon. The tablet  is not recognized at boot. I get this in dmesg:

`
[   17.275598] uclogic 0003:256C:006E.0001: device parameters not found
[   17.275600] uclogic 0003:256C:006E.0001: tablet enabling failed
`

I found a workaround by unpluggin the tablet and running:

`sudo modprobe -r hid-kye hid-uclogic hid-polostar hid-viewsonic`

When I plug it again it works. And I get this in dmesg:
```
[  190.448867] usb 3-1: new full-speed USB device number 5 using xhci_hcd
[  190.603872] usb 3-1: New USB device found, idVendor=256c, idProduct=006e, bcdDevice= 0.00
[  190.603876] usb 3-1: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[  190.650490] input: HID 256c:006e Pen as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:256C:006E.000B/input/input23
[  190.650697] input: HID 256c:006e Pad as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:256C:006E.000B/input/input24
[  190.650796] uclogic 0003:256C:006E.000B: input,hidraw0: USB HID v1.11 Keypad [HID 256c:006e] on usb-0000:00:14.0-1/input0
[  190.650873] uclogic 0003:256C:006E.000C: No inputs registered, leaving
[  190.650927] uclogic 0003:256C:006E.000C: hidraw8: USB HID v1.11 Device [HID 256c:006e] on usb-0000:00:14.0-1/input1
```
Specs:
Fedora 29 / Kernel  4.18.17
Digimend built from Git Master (via DKMS)

diegogangl, 2018-11-20:
## A more complete dmesg:

```
[    1.175025] uclogic 0003:256C:006E.0001: device parameters not found
[    1.175027] uclogic 0003:256C:006E.0001: tablet enabling failed
[    1.175142] uclogic 0003:256C:006E.0002: No inputs registered, leaving
[    1.175243] uclogic 0003:256C:006E.0002: hidraw0: USB HID v1.11 Device [HID 256c:006e] on usb-0000:00:14.0-1/input1
[   11.077433] uclogic 0003:256C:006E.0001: device parameters not found
[   11.077434] uclogic 0003:256C:006E.0001: tablet enabling failed
[  190.626212] uclogic 0003:256C:006E.000B: input,hidraw0: USB HID v1.11 Keypad [HID 256c:006e] on usb-0000:00:14.0-1/input0
[  190.626317] uclogic 0003:256C:006E.000C: No inputs registered, leaving
[  190.626368] uclogic 0003:256C:006E.000C: hidraw8: USB HID v1.11 Device [HID 256c:006e] on usb-0000:00:14.0-1/input1
```
---

## Output of /sbin/modinfo hid-uclogic

```
filename:       /lib/modules/4.18.18-300.fc29.x86_64/extra/hid-uclogic.ko.xz
version:        9
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     0485A47017CB313B2F84B27
alias:          hid:b0003g*v000028BDp00000042
alias:          hid:b0003g*v000028BDp00000094
alias:          hid:b0003g*v000028BDp00000075
alias:          hid:b0003g*v000028BDp00000071
alias:          hid:b0003g*v000028BDp00000074
alias:          hid:b0003g*v00002179p00000053
alias:          hid:b0003g*v00005543p00003031
alias:          hid:b0003g*v00005543p00000047
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        
retpoline:      Y
name:           hid_uclogic
vermagic:       4.18.18-300.fc29.x86_64 SMP mod_unload 

```

---

Output of cat /sys/module/hid_uclogic/version: 9

---

## lsusb

```
Bus 003 Device 006: ID 256c:006e  
```

(it doesn't show up with a name)

---
## Descriptors

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2600162/descriptors.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2600163/hid_report_descriptors.txt)

---
## Raw input samples

[coordinates.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2600167/coordinates.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2600169/pen_buttons.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2600170/pen_pressure.txt)
[tablet_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2600171/tablet_buttons.txt)

5nefarious, 2018-12-26:
I am experiencing the same issue on a Huion H610PRO. The workaround suggested by @diegogangl seems to work for me as well.

jatin11gidwani, 2019-02-25:
I am having the same problem that the tablet is not connecting to the updater, can you share the Google drive link of the new updater file you got ?

alihakki, 2019-03-08:
hi, I'm thinking of getting the H1060P. i m very amateur, is this tablet working on linux?

diegogangl, 2019-04-07:
@spbnick is there any way we could pool some money together to pay for some work on this tablet? Even just fixing the detection on booting would be great.

alihakki, 2019-04-13:
i have some problem with fedora, i test it on kde neon, some time do it again cant find tablet and pen when unplug, plug usb. i unplug usb and i wait some time and i plug it again it work. last night i tired too much unplug plug, restart pc, power off on pc nope not work, i go sleep and this morning it was working when i open pc. O_o

alihakki, 2019-04-13:
@diegogangl i solved this problem, tablet now working every reboot not need any command or unplug plug. only need re Generate initramfs O_o fedora dont do it automatic apply settings files like kdeneon or ubuntu.

this command can be not need but i do when tablet working 
$ sudo modprobe -r hid-kye hid-uclogic hid-polostar hid-viewsonic
unplug plug tablet 
$ xinput
slave  pointer  (2)]
⎜   ↳ COMPANY USB Device                      	id=11	[slave  pointer  (2)]
⎜   ↳ HID 256c:006e Pen stylus                	id=12	[slave  pointer  (2)]
⎜   ↳ HID 256c:006e Pad pad                   	id=13	[slave  pointer  (2)]

## Backup old initramfs nouveau image ##
sudo mv /boot/initramfs-$(uname -r).img /boot/initramfs-$(uname -r)-nouveau.img
## Create new initramfs image ##
sudo dracut /boot/initramfs-$(uname -r).img $(uname -r)
sudo reboot

spbnick, 2019-04-13:
@diegogangl, I haven't found a way to pool money for features. The closest thing was https://www.bountysource.com/

diegogangl, 2019-04-14:
@alihakki Tried and can confirm it works, great job! :clap: Though one thing I found is that you have to re-do it after updating the kernel. 

@spbnick Yeah, I was thinking about bountysource or something like that. It'd be great if we could pool money to complete this tablet's support.

berarma, 2020-12-04:
Maybe blacklisting some of the modules creating a file `/etc/modprobe.d/huionh1060p.conf` with contents:

```
blacklist hid-kye
#blacklist hid-uclogic
blacklist hid-polostar
blacklist hid-viewsonic
```

This could be a permanent solution.

How's the support for this model nowadays?

phseiff, 2020-12-15:
I would also like to know that, since I am thinking about buying one as an Ubuntu user >.<

berarma, 2020-12-15:
> I would also like to know that, since I am thinking about buying one as an Ubuntu user ʘ‿ʘ

I've bought it and it works OOTB on Linux 5.7, but not the soft buttons. Pressure works but I think tilt does not. The Gnome Wacom setup doesn't work because this device hasn't been added to libwacom. So the hard keys would need to be configured with xsetwacom.

I've installed DIGImend but I think it doesn't fix anything for me at the moment. Maybe it fixes tilt, I haven't checked.

CG-3D, 2020-12-15:
Actually, I have installed the latest Digimend v10 drivers and I can confirm the tilt works like a charm with my Huion H1060P in Krita 4.4.1 on Kubuntu 20.10.