DIGImend/digimend-kernel-drivers #213: Support for Huion Kamvas GT-156HD V2

Barracuda77, 2018-11-01:
Hello, I am planning on buying the Kamvas GT-156HD v2. But as it is quite an investment. I wanted to make sure if this driver supports it? I am on Ubuntu 18.04 LTS.
Any insights on this will be very appreciated  as I searched a lot for this and didn't find anything useful.
Regards,
Ahmed

spbnick, 2018-11-10:
We don't have that tablet officially supported. However, there's a high possibility the pen input will work. The buttons, and especially the touch strip might not.

To make sure you can try reaching Huion support. They actually paid to implement support for their latest, most popular models (but not exactly this one), in this driver, and some people within the company might be able to test your model. If not, then your request will at least inform them of people's interest and we might get financing for further work.

We should be able to implement support for your tablet eventually, but with the current queue of devices and lack of financing (apart from a bit of Patreon support), this can take indefinite time.

Barracuda77, 2018-11-10:
Thank you @spbnick for your reply and insight. I will definitely email Huion to ask them.
I was actually directed here by a youtube review that had a question about Linux drivers and the reviewer asked Huion and they directed him here.
and I had one more question, would it be possible to donate or put funds towards a specific driver?
Best,
Ahmed

spbnick, 2018-11-10:
@Barracuda77, so far I had a feeling that at the price I would find fair for this work, it won't be possible to collect enough funds. However, I think I ought to at least try and see. I'll think about that. If you know of any suitable funding platform for this kind of thing, please post your ideas here, or send me an e-mail.

Barracuda77, 2018-11-10:
@spbnick I sent you an email.

Barracuda77, 2018-12-11:
@andrelkin I ended up buying a different tablet. as when I was in the market I found better discounts on the XP-Pen Artist 22E pro.

paint-rulz, 2019-02-14:
A vote for this too. I don't know if it is really cheeky to ask this, but what payment would be necessary for this driver? ... and, just to make sure I have the right device: [https://www.amazon.co.uk/HUION-15-6-inch-V2-Upgraded-8192-Pressure-14-Customizable/dp/B0778JWV68/ref=sr_1_11?s=electronics&keywords=15.6+graphics+monitor](url)

paint-rulz, 2019-02-15:
@andrelkin me also. Thanks.

FeralBytes, 2019-10-15:
@spbnick I think once you push a release for version 10 of digimend drivers this can be closed because you added the new USB ID for it 256c:006d. By pulling from this git source "clone". I was able to dkms build and dkms install, restarted. Upon doing so my Kamvas GT-156HD V2 was detected although not working. So I did the following:
```
sudo usbhid-dump -es -m 256c:006d
```
and that gave me output for the stylus.
I then created a file at /etc/X11/xorg.conf.d/52-tablet.conf  the only file in this directory and I inserted the following contents to get it working after a restart.
```
# Huion tablets
Section "InputClass"
    Identifier "Huion Stylus"
    #MatchProduct "HUION"
    #MatchIsTablet "on"
    MatchDevicePath "/dev/input/event*"
    MatchUSBID "256c:006d"
    Driver "wacom"
EndSection

Section "InputClass"
    # Not Working.
    Identifier "Huion Buttons"
    #MatchProduct "HUION"
    MatchUSBID "256c:006d"
    MatchIsKeyboard "on"
    MatchDevicePath "/dev/input/event*"
    Driver "evdev"
EndSection

Section "InputClass"
    # Not Working.
    Identifier "Huion Scroll"
    #MatchProduct "HUION"
    MatchUSBID "256c:006d"
    MatchIsPointer "off"
    MatchIsKeyboard "off"
    MatchIsTouchpad "off"  
    # I think MatchIsTouchpad needs to be on.
    MatchIsTablet "off"
    MatchIsTouchscreen "off"
    MatchDevicePath "/dev/input/event*"
    Driver "evdev"
EndSection
```

After restart libwacom driver picked it up and allowed the calibration, and adjustments for the stylus pressure. Never got the Eraser or buttons working. Not concerned with the eraser. But if you have insight into how to get the buttons working I would appreciate it. Touchpad would be nice too.
OS:  Linux MSI-GP62-2QE 5.0.0-23-generic #24~18.04.1-Ubuntu SMP Mon Jul 29 16:12:28 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux