DIGImend/digimend-kernel-drivers #214: Huion Kamvas Pro 13 functionality issues

Eqlipse, 2018-11-01:
Hello! I'm trying to get my recently-acquired art hardware working on my Kubuntu system, but I seem to be running into an issue.

The tablet only seems to be recognized as a generic "Tablet Monitor Pad pad," and the stylus as "Tablet Monitor Pen stylus." While I can bind the buttons just fine with xwacomset, it doesn't seem to recognize the scrollbar. Also, while binding the pen to the HDMI output does seem to work, I don't appear to get the full pressure sensitivity range.

I tried to investigate this a bit with uclogic-probe, and huion-probe, but they both return an error:
"Failed to get manufacturer string descriptor: Pipe error"

Running lsusb -v give me this:
```
Bus 003 Device 002: ID 256c:006e  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x256c 
  idProduct          0x006e 
  bcdDevice            0.00
  iManufacturer           5 (error)
  iProduct                6 Tablet Monitor
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      18
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               2
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      97
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               2
Device Status:     0x0000
  (Bus Powered)
```
Any help offered would be greatly appreciated ^_^

ThePreviousOne, 2018-11-05:
uclogic-probe release version is outdated try using git master

Eqlipse, 2018-11-07:
What are you talking about? As far as I know, the _only_ version available is the one listed here: https://github.com/DIGImend 
Which is what I'm using.

Eqlipse, 2018-11-07:
I tried re-installing Uclogic tools again, and got it working after a reboot.
Here's the output:
```P 54 00 61 00 62 00 6C 00 65 00 74 00 20 00 4D 00 6F 00 6E 00 69 00 74 00 6F 00 72 00
S 65 04 03 09 00
S 6E 04 03 31 00
S 79 02 03
S 7A 08 03 01 10 00 00 00 00
S 7B 0C 03 48 00 4B 00 20 00 4F 00 6E 00
S C8 13 03 80 E5 00 18 81 00 FF 1F D8 13 03 05 00 01 08 00 40
S C9 24 03 48 00 55 00 49 00 4F 00 4E 00 5F 00 4D 00 31 00 38 00 32 00 5F 00 31 00 38 00 31 00 30 00 31 00 39 00
S CA 46 03 48 00 55 00 49 00 4F 00 4E 00 20 00 41 00 6E 00 69 00 6D 00 61 00 74 00 69 00 6F 00 6E 00 20 00 54 00 65 00 63 00 68 00 6E 00 6F 00 6C 00 6F 00 67 00 79 00 20 00 43 00 6F 00 2E 00 2C 00 6C 00 74 00 64 00
```

Eqlipse, 2018-11-07:
Again, I'm trying to get the scrollbar to work on the tablet. Just to clarify what I'm working on here.

Eqlipse, 2018-11-18:
(relatively) Fresh install of Ubuntu 18.10, returning to the same issue....
```
lsusb | grep 256c:006e
Bus 003 Device 002: ID 256c:006e

sudo uclogic-probe 3 2
 P 54 00 61 00 62 00 6C 00 65 00 74 00 20 00 4D 00 6F 00 6E 00 69 00 74 00 6F 00 72 00
S 65 04 03 09 00
S 6E 04 03 31 00
S 79 02 03
S 7A 08 03 01 10 00 00 00 00
S 7B 0C 03 48 00 4B 00 20 00 4F 00 6E 00
S C8 13 03 80 E5 00 18 81 00 FF 1F D8 13 03 05 00 01 08 00 40
S C9 24 03 48 00 55 00 49 00 4F 00 4E 00 5F 00 4D 00 31 00 38 00 32 00 5F 00 31 00 38 00 31 00 30 00 31 00 39 00
S CA 46 03 48 00 55 00 49 00 4F 00 4E 00 20 00 41 00 6E 00 69 00 6D 00 61 00 74 00 69 00 6F 00 6E 00 20 00 54 00 65 00 63 00 68 00 6E 00 6F 00 6C 00 6F 00 67 00 79 00 20 00 43 00 6F 00 2E 00 2C 00 6C 00 74 00 64 00

xsetwacom --list devices
Tablet Monitor Pen stylus               id: 10  type: STYLUS    
Tablet Monitor Pad pad                  id: 11  type: PAD

xinput list 11
Tablet Monitor Pad pad                          id=11   [slave  pointer  (2)]
        Reporting 8 classes:
                Class originated from: 11. Type: XIButtonClass
                Buttons supported: 16
                Button labels: None None None None None None None None None None None None None None None None
                Button state:
                Class originated from: 11. Type: XIKeyClass
                Keycodes supported: 248
                Class originated from: 11. Type: XIValuatorClass
                Detail for Valuator 0:
                  Label: Abs X
                  Range: 0.000000 - 0.000000
                  Resolution: 0 units/m
                  Mode: absolute
                  Current value: 0.000000
                Class originated from: 11. Type: XIValuatorClass
                Detail for Valuator 1:
                  Label: Abs Y
                  Range: 0.000000 - 0.000000
                  Resolution: 0 units/m
                  Mode: absolute
                  Current value: 0.000000
                Class originated from: 11. Type: XIValuatorClass
                Detail for Valuator 2:
                  Label: None
                  Range: 0.000000 - 1.000000
                  Resolution: 1 units/m
                  Mode: absolute
                  Current value: 0.000000
                Class originated from: 11. Type: XIValuatorClass
                Detail for Valuator 3:
                  Label: None
                  Range: 0.000000 - 1.000000
                  Resolution: 1 units/m
                  Mode: absolute
                  Current value: 0.000000
                Class originated from: 11. Type: XIValuatorClass
                Detail for Valuator 4:
                  Label: None
                  Range: 0.000000 - 1.000000
                  Resolution: 1 units/m
                  Mode: absolute
                  Current value: 0.000000
                Class originated from: 11. Type: XIValuatorClass
                Detail for Valuator 5:
                  Label: None
                  Range: 0.000000 - 1.000000
                  Resolution: 1 units/m
                  Mode: absolute
                  Current value: 0.000000
```

Out of curiosity, what would be the _default_ values for a strip? As in, what command calls would I make to configure it? I use a script to handle the setting of buttons whenever I start my system...
```
#!/bin/bash
#
tablet="Tablet Monitor Pad pad"
pen="Tablet Monitor Pen stylus"

bindings=(
"+ctrl z" # undo
"+e" # erase
"+ctrl"
"+shift"
"+m" # mirror
)

buttons=(1 2 3 8 9)

i=0
for binding in "${bindings[@]}"
do
  xsetwacom --set "$tablet" Button ${buttons[$((i++))]} key $binding
done

xsetwacom set "$pen" MapToOutput HDMI-A-0
xsetwacom set "$pen" PressureCurve 0 0 100 50
#xinput set-prop "$tablet" --type=float "Coordinate Transformation Matrix" 0.5 0 0.5 0 1 0 0 0 1
#xsetwacom --set "$tablet" Area 200 0 65100 38500
#xsetwacom --set "$tablet" Button 2 button +2
#xsetwacom --set "$tablet" Button 3 button +3
```

cyberpaint, 2018-11-26:
I also recently bought the Kamvas Pro 13 and having quite some problems getting it to work on Ubuntu 18.10 (still new to linux). Can't even get it to be listed with xsetwacom --list 
**How did you set it up?** 

Which tutorial did you follow?

wwklnd, 2019-01-03:
I have recently bought a Kamvas Pro 13, and can't get it to work in Manjaro. The closest I've gotten is Benthor's python based workaround for the GT-191, but there the mouse pointer doesn't line up with pen placement. I've run the diagnostics listed on the DIGImend site, I'll attach the outputs I got. I've tried all the solutions I can find, and I'm not getting much further it seems.
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2724421/frame_buttons.txt)
[frame_dials.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2724422/frame_dials.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2724423/pen_buttons.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2724424/pen_pressure.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2724425/pen_tilt.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2724426/pen_coords.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2724427/probe.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2724428/hid_report_descriptors.txt)
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2724429/descriptors.txt)

GBeauregard, 2019-12-02:
> I have recently bought a Kamvas Pro 13, and can't get it to work in Manjaro. The closest I've gotten is Benthor's python based workaround for the GT-191, but there the mouse pointer doesn't line up [...]

@wwklnd Did you ever get it working? I'm having trouble getting the Kamvas Pro 13 working in Gentoo. xsetwacom lists the devices like the OP here, but I can't get the stylus to actually work for anything

dwilches, 2020-04-25:
Finally got my Kamvas Pro 13 to work well in Ubuntu, try installing version 20.04. Also, I had to uninstall the official drivers for my AMD card and let Ubuntu use its own.
Things that work now and that didn't work before for me:
- I can use 2 screens at the same time, with input going only to the tablet's screen
- Mapping of the 5 buttons is working now
Nonetheless, I've not found yet how to configure the touch bar.

wwklnd, 2020-05-29:
Jumping in here again with an update: I've gotten my Kamvas Pro 13 to work on both my laptop and my desktop with just the latest DIGImend kernel drivers under Manjaro, to some extent. The pen and all buttons on it work, as does the display, but the touch strip and four middle buttons do *not* work. How would I go about configuring those? I seem to have missed something, since it sounds like some others have gotten them to work.

dwilches, 2020-05-29:
@wwklnd I managed to get the buttons on the display working, this is what I did:
```
$ xsetwacom --list devices
Tablet Monitor Pen stylus               id: 16  type: STYLUS    
Tablet Monitor Touch Ring pad           id: 17  type: PAD       
Tablet Monitor Pad pad                  id: 18  type: PAD       

$ xsetwacom set 18 Button 1 'key ctrl z'
$ xsetwacom set 18 Button 2 'key tab'
$ xsetwacom set 18 Button 3 'key f'
$ xsetwacom set 18 Button 8 'key shift f'
$ xsetwacom set 18 Button 9 'key q'
```
Notice that you can set a shortcut even for the bottom button (index 9 above). I couldn't get the strip working though.

I put together [a script](https://github.com/dwilches/huion-kamvas-pro-13) that configures the screen and the buttons depending on the software that you want to use, so I'm linking it here in case it's useful.

wwklnd, 2020-05-29:
@dwilches Oh, interesting! `xsetwacom --list devices` only lists my Wacom Intuos 4 stuff, I didn't think to set up my Kamvas using that as well. Might give that a whirl!

wwklnd, 2020-05-29:
This is likely relevant to the strip not working for the record: https://gitlab.gnome.org/GNOME/gnome-control-center/-/issues/118