DIGImend/digimend-kernel-drivers #215: I can't get it to work with manjaro!

SlightlyButtered, 2018-11-02:
Hello there,

 I've been very stressed recently trying to get this to work on manjaro, After hours and hours of searching online I've turned up dry, Manjaro support team was somewhat helpful but still didn't solve anything, And I can't find online resources on how to install this on manjaro! All I ask is to provide a detailed explanation on how to install it on manjaro, And if it is already listed... GREAT! Just link/show me where to go and I will be on my merry way.

Kind regards,
SlightlyButtered

narutse, 2018-11-08:
I got mine to work! Did you install your kernel headers? I follow the instruction here: https://bbs.archlinux.org/viewtopic.php?id=239082 but replaced linux-headers with the one corresponding to my kernel.

spbnick, 2018-11-09:
I feel your pain, @SlightlyButtered, but we can't really help you unless you tell us what exactly goes wrong. How do you install the driver, what results do you get? The best would probably be a copy-paste of your terminal session as you're trying to install it.

spbnick, 2019-01-05:
Closing as reporter is not responding. @SlightlyButtered, please reopen, if you still need help.

closed 2019-01-05 by spbnick