DIGImend/digimend-kernel-drivers #216: Huion H640P Doesn't make it into evtest output on CentOS

rubypanther, 2018-11-05:
Running CentOS 7.5.1804 with a new Huion H640P and it doesn't really work.

I've tried it on two systems running the same OS, one is a recent Thinkpad laptop and the other is an older AMD-based desktop. Results were same.

Out of the box it partially works using hid-generic. It reports x and y that change as expected, so the hardware seems OK.

After installing current digimend-kernel-drivers from github, it creates some noise is the logs and shows up in lsusb but doesn't show up in evtest at all.

Here the is the dmesg input upon insert: 
https://gist.github.com/rubypanther/d40aab981792cdd2ab9a478d74eca020

X log:
https://gist.github.com/rubypanther/89716b09d8100e171c298200be869ebd

evtest output, note that in X log above it is trying to set up at /dev/input/event20 but evtest only lists up to 19. Those were on the same insert. I'm not sure if the dbus log was the same insert.
https://gist.github.com/rubypanther/5a9683be35abae478539461e2291080e

 cat /sys/module/hid_uclogic/version
returns: 9

I'd love to get this working, please let me know if there is any more information needed! Please forgive me in advance if it takes me a couple days to reply, though. :)

spbnick, 2018-11-10:
This looks like the device file (/dev/input/event*) is being yanked from the system just as X.org is trying to set it up. Do you see any messages in `dmesg` after the ones you've posted? Anything about device being disconnected? Perhaps the tablet's firmware resets?

When and where did you buy the tablet? Could it be using newer, and buggier firmware?

rubypanther, 2018-11-10:
There was nothing more in dmesg after that.

I bought it new recently on amazon, so it could be new firmware.

spbnick, 2018-11-10:
OK, I'll try my h640p on CentOS 7.5 today, just in case. Meanwhile, do you think you could capture the USB traffic you get from the moment you plug the tablet in and until X.org fails to use the tablet? You can use Wireshark for that.

rubypanther, 2018-11-10:
[usb_huion_pcapng.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/2568309/usb_huion_pcapng.zip)
Here is the wireshark output

spbnick, 2018-11-10:
Thank you. Still looking at the capture and am still to try my h640p on CentOS. Meanwhile found and fixed a DKMS issue breaking the driver installation: dell/dkms#66.

spbnick, 2019-01-05:
Closing this issue as reporter communication has ceased. @rubypanther, please reopen, if you still need help.

rubypanther, 2019-01-05:
I never ceased communication, what information did you need?

rubypanther, 2019-01-05:
I don't have a button to re-open, but it sounded like you'd found the bug?

Is it correct that this was not fixed?

If you want me to fix it let me know, when you said you found it I stopped worrying about what the actual code changed needed was, because it would take a bunch of time to become familiar with the code.

But it definitely should get fixed, RHEL and Centos are fairly important distros.

rubypanther, 2019-01-11:
Still here, this is still a bug, please reopen.

spbnick, 2019-01-14:
Yes, I found the issue, but am not sure how to fix it. The problem is that upstream the input_configured handler has a particular interface after a particular kernel version and RHEL and CentOS have that change backported to an earlier kernel version. I couldn't find a way to detect that change except by checking the kernel version, and I cannot change the version number condition just for RHEL and CentOS. If you can come up with a good solution to this, I'd be glad to merge it.