DIGImend/digimend-kernel-drivers #217: Huion Giano WH1409 does not work anymore with newest digimend driver

hvfrancesco, 2018-11-05:
My Huion Giano WH1409 works well with v7 of Digimend driver, but it doesn't with newer version.
the tablet is recognized and it appears in xsetwacom --list output, but gives no sign of life

hvfrancesco, 2018-11-05:
evtest shows no output

hvfrancesco, 2018-11-05:
ubuntu 14.04
3.13.0-161-generic kernel

ThePreviousOne, 2018-11-06:
I would try testing with a newer (still maintained) kernel

ThePreviousOne, 2018-11-06:
or maybe not #203

spbnick, 2018-11-07:
Hi folks, I'll try to look into this on the weekend, hold tight!

tomvdab, 2018-11-07:
I bought a Huion Giano Wh1409 v2 some days ago and it works perfectly with Digimend v8 + arch Linux with it's most recent kernel. Thx for your work with Digimend btw :)

hvfrancesco, 2018-11-08:
tried on a freshly installed Ubuntu 18.04 with 4.15.0-38-generic kernel, unfortunately same results. The tablet works well with distribution provided driver, but if I switch to v.8 dkms provided on digimend site, it is not responding anymore.

tomvdab, 2018-11-08:
I use linux kernel 4.18.16. Before installing Digimend v8 dkms the tablet was not functioning. After installing Digimend dkms it worked.

ChiSamurai, 2018-11-09:
I'm on Linux Mint 18.3 with kernel 4.10.0-38 and got the same issues. With the out of the box driver it works like a mouse, with digimend drivers (tried v6 - v8) xsetwacom --output lists two devices (Pen & Pencil) but no reaction at all :-(

I commented another issue some days before and attached some additional infos here: https://github.com/DIGImend/digimend-kernel-drivers/issues/51#issuecomment-435598029-permalink

#issuecomment-435598029-permalink

tomvdab, 2018-11-09:
I have only this configuration for X11 and everything is working:

`/etc/X11/xorg.conf.d/50-tablet.conf`

```
Section "InputClass"
    Identifier "Tablet"
    Driver "wacom"
    MatchDevicePath "/dev/input/event*"
    MatchUSBID "256c:006e"
EndSection
```
I just followed the instructions on the [README](https://github.com/DIGImend/digimend-kernel-drivers)
It's a bit strange that on Ubuntu the tablet works as a mouse out of the box. On Arch Linux the tablet didn't function at all before installing Digimend.

spbnick, 2018-11-10:
Well, you can see why we don't have it listed as "supported" - nobody actually went and made sure it is, and nobody figured out how it works yet, so nobody keeps it working. That's why it broke, perhaps.

First of all, please try installing the current development version: https://github.com/DIGImend/digimend-kernel-drivers/archive/master.zip

If that doesn't work, please reconnect your tablet and post the following:
* the output of `sudo dmesg` (last 100 lines or so is enough)
* the output of `/sbin/modinfo hid-uclogic`
* the output of `cat /sys/module/hid_uclogic/version`
* the output of `sudo evtest`, just what it prints at the start, without any event dumps

And then let's go from here.

tomvdab, 2018-11-10:
If you guys want me to post info about my working setup, feel free to ask! To simplify troubleshooting, hvfrancesco, did you connect your tablet with usb-cable only? When you have the wireless dongle attatched to  your computer, VID:PID shows up two times in lsusb, which might complicate things.

hvfrancesco, 2018-11-11:
thank you @spbnick, I try to include a little more information, and I attacch as well the Xorg.0.conf with development driver and with stock driver. Thank you a lot for your help
[Xorg.0.log-dev.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2569242/Xorg.0.log-dev.txt)
[Xorg.0.log-stock.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2569243/Xorg.0.log-stock.txt)

:

first the results of uclogic-probe and usbhid dump using development drivers:

lsusb

Bus 003 Device 007: ID 256c:006e
------------------------------------------------------------------------

sudo ./uclogic-probe 3 7
M 54 00 41 00 42 00 4C 00 45 00 54 00
P 50 00 65 00 6E 00 20 00 54 00 61 00 62 00 6C 00 65 00 74 00
S 64 12 03 A0 D7 C4 86 03 00 FF 07 A0 0F 08 00 00 00 00 00
S 65 04 03 20 A0
S 6E 02 00
S 79 02 00
S 7A 08 03 01 08 00 00 00 00
S 7B 0C 03 48 00 4B 00 20 00 4F 00 6E 00
S C8 12 03 A0 D7 00 C4 86 00 FF 07 A0 0F 03 0C 00 00 10 00
S C9 24 03 48 00 55 00 49 00 4F 00 4E 00 5F 00 54 00 31 00 35 00 33 00 5F 00 31 00 36 00 30 00 36 00 30 00 37 00
S CA 46 03 48 00 55 00 49 00 4F 00 4E 00 20 00 41 00 6E 00 69 00 6D 00 61 00 74 00 69 00 6F 00 6E 00 20 00 54 00 65 00 63 00 68 00 6E 00 6F 00 6C 00 6F 00 67 00 79 00 20 00 43 00 6F 00 2E 00 2C 00 6C 00 74 00 64 00
------------------------------------------------------------------------

sudo usbhid-dump -m 256C:006E -es
Starting dumping interrupt transfer stream
with 1 minute timeout.

003:007:000:STREAM             1541929568.007272
 07 80 86 3D C1 43 00 00

003:007:000:STREAM             1541929568.009254
 07 80 86 3D C1 43 00 00

003:007:000:STREAM             1541929568.013257
 07 80 86 3D C1 43 00 00

003:007:000:STREAM             1541929568.015258
 07 80 78 3D C1 43 00 00

003:007:000:STREAM             1541929568.019258
 07 80 6E 3D CC 43 00 00

003:007:000:STREAM             1541929568.021259
 07 80 64 3D CC 43 00 00

003:007:000:STREAM             1541929568.025256
 07 80 5A 3D CC 43 00 00

003:007:000:STREAM             1541929568.027256
 07 80 51 3D D4 43 00 00

003:007:000:STREAM             1541929568.033256
 07 80 40 3D E2 43 00 00

003:007:000:STREAM             1541929568.037222
 07 80 38 3D E7 43 00 00

003:007:000:STREAM             1541929568.039223
 07 80 30 3D EC 43 00 00

003:007:000:STREAM             1541929568.053236
 07 80 03 3D F7 43 00 00

003:007:000:STREAM             1541929568.055238
 07 80 F3 3C FF 43 00 00

[... 107 lines cut ...]
003:004:000:STREAM             1541928865.252557
 07 80 96 63 74 36 00 00

003:004:000:STREAM             1541928865.270551
 07 80 2B 63 B4 36 00 00

003:004:000:STREAM             1541928865.274548
 07 80 10 63 C6 36 00 00

003:004:000:STREAM             1541928865.294530
 07 80 45 62 62 37 00 00

003:004:000:STREAM             1541928865.310531
 07 80 E0 61 CC 37 00 00

003:004:000:STREAM             1541928865.316530
 07 80 C6 61 EC 37 00 00

003:004:000:STREAM             1541928865.346542
 07 80 79 61 2B 38 00 00

003:004:000:STREAM             1541928865.352542
 07 80 67 61 36 38 00 00

------------------------------------------------------------------------

hvfrancesco, 2018-11-12:
I'm really sorry, I made a mess with copy and paste on my previous comment, the actual output of
sudo usbhid-dump -m 256C:006E -es with the development driver is as follows (it seems there should not be the 0A code at the beginning):

Starting dumping interrupt transfer stream
with 1 minute timeout.

003:004:001:STREAM             1542013945.696547
 0A C0 25 1E A5 39 00 00

003:004:001:STREAM             1542013945.700532
 0A C0 25 1E B0 39 00 00

003:004:001:STREAM             1542013945.730560
 0A C0 3B 1E DC 39 00 00

003:004:001:STREAM             1542013945.742548
 0A C0 57 1E DA 39 00 00

003:004:001:STREAM             1542013945.754564
 0A C0 84 1E D7 39 00 00

003:004:001:STREAM             1542013945.766574
 0A C0 B9 1E DA 39 00 00

003:004:001:STREAM             1542013945.846586
 0A C1 9E 20 88 39 63 05

003:004:001:STREAM             1542013945.858603
 0A C1 22 21 51 39 B7 05

003:004:001:STREAM             1542013945.928641
 0A C1 FB 25 E0 36 57 07

003:004:001:STREAM             1542013945.956641
 0A C1 B3 28 9A 35 5B 07

003:004:001:STREAM             1542013945.970644
 0A C1 56 2A AA 34 3B 07

003:004:001:STREAM             1542013945.976636
 0A C1 08 2B 35 34 F3 06

003:004:001:STREAM             1542013945.980655
 0A C1 62 2B F4 33 A7 06

003:004:001:STREAM             1542013945.988641
 0A C1 7B 2C 1A 33 D7 04

003:004:001:STREAM             1542013945.994647
 0A C1 39 2D 75 32 A7 02

003:004:001:STREAM             1542013945.998658
 0A C1 96 2D 1F 32 C3 01

003:004:001:STREAM             1542013946.000663
 0A C1 F4 2D C4 31 0B 01

003:004:001:STREAM             1542013946.004666
 0A C1 50 2E 65 31 7B 00

003:004:001:STREAM             1542013946.010638
 0A C0 05 2F A0 30 00 00

003:004:001:STREAM             1542013946.012640
 0A C0 5C 2F 3A 30 00 00

003:004:001:STREAM             1542013946.016657
 0A C0 B3 2F D4 2F 00 00

003:004:001:STREAM             1542013946.018643
 0A C0 07 30 6C 2F 00 00

003:004:001:STREAM             1542013946.022654
 0A C0 59 30 02 2F 00 00

003:004:001:STREAM             1542013946.024655
 0A C0 A9 30 9D 2E 00 00

003:004:001:STREAM             1542013946.028646
 0A C0 F7 30 39 2E 00 00

003:004:001:STREAM             1542013946.030638
 0A C0 45 31 D6 2D 00 00

hvfrancesco, 2018-11-13:
I've tried to dump the hid report descriptor, maybe it's of some use
[wh_1409.code.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2575137/wh_1409.code.txt)
[wh_1409.spec.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2575138/wh_1409.spec.txt)
[wh_1409.xml.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2575139/wh_1409.xml.txt)
[wh1409_desc.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2575140/wh1409_desc.txt)

spbnick, 2018-11-13:
Thank you for the dumps, @hvfrancesco. Alright, the reports starting with 0A seem to be due to the tablet not being initialized by the driver. Perhaps the development driver didn't load in that case.

So far it appears that even though this tablet says it's a new version, it's still using the old protocol. This throws a wrench into the auto-detection method used in the driver, but perhaps we can salvage the situation.

Now, let's verify this is the case. Install the latest development version ("master") of the driver, reboot or run `sudo modprobe -r hid-uclogic`, connect the tablet, make sure the new driver loaded (`cat /sys/module/hid_uclogic/version` should output `9`). Then run `sudo usbhid-dump -m 256C:006E -es` and do a stroke with the pen. Post a few lines of the output.

As an additional verification, run `uclogic-probe` on your tablet, discard the output (we don't need it anymore), run usbhid-dump again, do a stroke, and post a few lines of the output.

As the last verification, please execute `echo "module hid_uclogic +pfl" | sudo tee /sys/kernel/debug/dynamic_debug/control` to enable debug logging, reconnect the tablet, and post about 100 lines from the end of `dmesg` output.

Thank you.

hvfrancesco, 2018-11-13:
thanks a lot @spbnick ,

here it is

------------------------------------------------------------------------------
cat /sys/module/hid_uclogic/version
9
------------------------------------------------------------------------------
THIS IS THE OUTPUT AFTER LOADING THE DRIVER

sudo usbhid-dump -m 256C:006E -es
Starting dumping interrupt transfer stream
with 1 minute timeout.

003:007:001:STREAM             1542100182.824954
 0A C0 4A 1F 4A 4E 00 00

003:007:001:STREAM             1542100182.828946
 0A C0 4A 1F 4A 4E 00 00

003:007:001:STREAM             1542100182.830947
 0A C0 4A 1F 4A 4E 00 00

003:007:001:STREAM             1542100182.834947
 0A C0 3F 1F 4A 4E 00 00

003:007:001:STREAM             1542100182.836944
 0A C0 35 1F 3B 4E 00 00

003:007:001:STREAM             1542100182.840944
 0A C0 2C 1F 3B 4E 00 00

003:007:001:STREAM             1542100182.842946
 0A C0 22 1F 3B 4E 00 00

003:007:001:STREAM             1542100182.846946
 0A C0 1A 1F 3B 4E 00 00

003:007:001:STREAM             1542100182.848945
 0A C0 11 1F 35 4E 00 00
----------------------------------------------------------------------------------------

THIS IS THE OUTPUT AFTER RUNNING uclogig-probe

sudo usbhid-dump -m 256C:006E -es
Starting dumping interrupt transfer stream
with 1 minute timeout.

003:007:000:STREAM             1542100310.622020
 07 80 06 40 39 41 00 00

003:007:000:STREAM             1542100310.685985
 07 80 4B 42 74 40 00 00

003:007:000:STREAM             1542100310.826000
 07 81 EF 46 DE 3E 07 03

003:007:000:STREAM             1542100310.877999
 07 81 F7 48 23 3E 97 04

003:007:000:STREAM             1542100310.949998
 07 81 A8 4D FB 3B A3 06

003:007:000:STREAM             1542100310.967985
 07 81 CF 4E 95 3B BF 06

003:007:000:STREAM             1542100311.010004
 07 81 52 51 FB 3A 0B 06
-----------------------------------------------------------------------------------------------

[... 119 lines cut ...]
[  552.543672] input: TABLET Pen Tablet Mouse as /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.1/3-11.1:1.1/0003:256C:006E.0014/input/input60
[  552.544098] uclogic 0003:256C:006E.0014: input,hiddev0,hidraw1: USB HID v1.11 Mouse [TABLET Pen Tablet] on usb-0000:00:14.0-11.1/input1
[  552.547911] uclogic_params_init:894: uclogic 0003:256C:006E.0015: pen v2 parameters found
[  552.547923] uclogic_probe:205: uclogic 0003:256C:006E.0015: parameters:
               .invalid = false
               .desc_ptr =           (null)
               .desc_size = 0
               .pen_unused = false
               .pen.desc_ptr = 000000002c2c3263
               .pen.desc_size = 96
               .pen.id = 8
               .pen.inrange = none
               .pen.fragmented_hires = true
               .frame.desc_ptr = 000000006e089d9b
               .frame.desc_size = 70
               .frame.id = 247
               .frame.re_lsb = 0
               .frame.dev_id_byte = 0
               .pen_frame_flag = 0x20
[  552.548347] input: TABLET Pen Tablet Pen as /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.1/3-11.1:1.0/0003:256C:006E.0015/input/input62
[  552.548650] input: TABLET Pen Tablet Pad as /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.1/3-11.1:1.0/0003:256C:006E.0015/input/input63
[  552.548913] uclogic 0003:256C:006E.0015: input,hidraw2: USB HID v1.10 Keypad [TABLET Pen Tablet] on usb-0000:00:14.0-11.1/input0
francesco@francesco:/mnt/sviluppo/uclogic-tools-master$ 

thank you again a lot for your efforts

spbnick, 2018-11-13:
Thank you, @hvfrancesco, this confirms it: even though the tablet responds to new-model initialization from the driver, it doesn't actually switch to the new mode and continues in default mode, on another interface. It does respond to old-model initialization and that's what we can use. Must have been a model released some time during the transition. This will need special handling in the driver, which I'll try to find time for, this week.

hvfrancesco, 2018-11-13:
I think you're absolutelu right: as a check I just inverted the order of v2 and v1  probe in uclogic_params_init(), recompiled, and the tablet is correctly working with the modified driver

spbnick, 2018-11-13:
Yes, the problem is that would break the newer tablets, so we'll have to do something smart here.

hvfrancesco, 2018-11-13:
sure, thanks again. I'm sorry I can't help more, but I'm taking the occasion to learn a bit of new stuff too

ChiSamurai, 2018-11-15:
Wow, thanks for all the work!! I'd wish I could help you anyhow on this... at least by visiting your patreon site for a donation :-)

spbnick, 2018-11-17:
I'm thinking about making the driver detect the WH1409 quirk using the value of a string descriptor, which seems to contain firmware version. Everyone here, who wants their WH1409 supported by the driver, please post the output of `uclogic-probe` for your tablet, so that I can verify I'm not missing your version. Thank you.

@hvfrancesco, I already have yours, so no need to do anything.

ChiSamurai, 2018-11-17:
Thanks a lot! I'll post mine tomorrow as soon as I'm back home!

tomvdab, 2018-11-17:
[uclogic-probe_Huion_wh1409_v2.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2592124/uclogic-probe_Huion_wh1409_v2.txt)

spbnick, 2018-11-18:
@tomvdab, thank you! Your uclogic-probe output looks very much different from @hvfrancesco one. The firmware is much newer and the resolution is higher. I would say that your tablet might just work with the new driver. Have you tried that?

@hvfrancesco, do you think you could try asking Huion for a firmware update? That might actually fix your issue.

I'll wait a bit for more uclogic-probe outputs to decide what to do.

ChiSamurai, 2018-11-18:
[uclogic-decode.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2592930/uclogic-decode.txt)
[uclogic-probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2592931/uclogic-probe.txt)

spbnick, 2018-11-18:
Thank you, @ChiSamurai! Do you think you could run the [latest development version of uclogic-probe](https://github.com/DIGImend/uclogic-tools/archive/master.zip) instead? The output of the last release is not enough, unfortunately.

I need to make a new release.

ChiSamurai, 2018-11-18:
@spbnick No problem, here are the new outputs:

[uclogic-decode.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2592977/uclogic-decode.txt)
[uclogic-probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2592978/uclogic-probe.txt)

spbnick, 2018-11-18:
Thank you, @ChiSamurai, your uclogic-probe output seems to be the same as @hvfrancesco has, which gives me a bit more confidence my approach is going to work.

tomvdab, 2018-11-18:
@spbnick, I installed the latest driver (commit 9b90aa8fa9d98eacd955d10fddac644208b30b66) and everything seems to be working.

@hvfrancesco, yes, I own the Huion wh1409 v2. [Here](https://www.huiontablet.com/huion-wh1409-v2.html) you find a comparison of the newer and the older model. The v2 model has tilt function but I suppose Digimend doesn't implement that feature (yet).

hvfrancesco, 2018-11-20:
I've contacted Huion support to see if it existed an update for the firmware of the first version of WH-1409. Up to now the only answer I got from them was a suggestion to connect the tablet under windows or mac os to check if it was working. I wrote back explaining once again that the problem was not in the hardware and what the issue was, but I've had the impression I can't expect much from them.

spbnick, 2018-11-20:
Thank you for the update, @hvfrancesco. OK, let's proceed with our plan. I'll write some code checking for the firmware version and changing the initialization logic for your model.

Mraize, 2018-11-22:
Hey, here's my tablet's dmesg and probe/decode output, should be the first version, or late in production.

[huionwh1409_dmesg_probe_decode.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2608778/huionwh1409_dmesg_probe_decode.txt)

spbnick, 2018-11-22:
Thank you, @Mraize, your tablet is the same as the ones @hvfrancesco and @ChiSamurai have.

spbnick, 2018-11-24:
OK, I implemented support for Huion WH1409 and pushed it to #225. Please download and test the modified code available at https://github.com/spbnick/digimend-kernel-drivers/archive/add_support_for_huion_wh1409.zip

Please report your results here!

hvfrancesco, 2018-11-24:
@spbnick the code seems to work perfectly! the tablet is correctly initialized and performs as expected. Only thing not working are the buttons on the frame, but that's not a big issue for me. Thanks

spbnick, 2018-11-24:
Glad to hear that, @hvfrancesco! The buttons should work, could you describe how you expect them to work, and also provide the output of `evtest` for the "pad" device, where you press them all in turn.

hvfrancesco, 2018-11-24:
I was trying to map them with xsetwacom as the pad is correctly using wacom X11 driver.

 xsetwacom --list
TABLET Pen Tablet Pen stylus    	id: 8	type: STYLUS    
TABLET Pen Tablet Pad pad       	id: 9	type: PAD  

It does not give any error message when, for example, do:
xsetwacom --set "TABLET Pen Tablet Pad pad" Button 10 "key y"
but there's no effect, and so I've tried to check with evtest and there seems to be no effect either when pressing them. Maybe I'm doing something wrong here. This is the output:

sudo evtest 
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Power Button
/dev/input/event1:	Power Button
/dev/input/event2:	AT Translated Set 2 keyboard
/dev/input/event3:	ImExPS/2 Logitech Explorer Mouse
/dev/input/event4:	TABLET Pen Tablet Pen
/dev/input/event5:	TABLET Pen Tablet Pad
/dev/input/event6:	TABLET Pen Tablet Mouse
/dev/input/event7:	TABLET Pen Tablet Keyboard
/dev/input/event8:	TABLET Pen Tablet Consumer Control
/dev/input/event9:	TABLET Pen Tablet System Control
/dev/input/event10:	HDA Intel PCH Front Mic
/dev/input/event11:	HDA Intel PCH Rear Mic
/dev/input/event12:	HDA Intel PCH Line
/dev/input/event13:	HDA Intel PCH Line Out
/dev/input/event14:	HDA Intel PCH Front Headphone
/dev/input/event15:	Eee PC WMI hotkeys
/dev/input/event16:	HDA NVidia HDMI/DP,pcm=3
/dev/input/event17:	HDA NVidia HDMI/DP,pcm=7
/dev/input/event18:	HDA NVidia HDMI/DP,pcm=8
/dev/input/event19:	HDA NVidia HDMI/DP,pcm=9
Select the device event number [0-19]: 5
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6e version 0x110
Input device name: "TABLET Pen Tablet Pad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 256 (BTN_0)
    Event code 257 (BTN_1)
    Event code 258 (BTN_2)
    Event code 259 (BTN_3)
    Event code 260 (BTN_4)
    Event code 261 (BTN_5)
    Event code 262 (BTN_6)
    Event code 263 (BTN_7)
    Event code 264 (BTN_8)
    Event code 265 (BTN_9)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 331 (BTN_STYLUS)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max        1
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max        1
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)

ChiSamurai, 2018-11-24:
Tablet indeed seems to work like a charm. I did not try the buttons yet, but I'll do it in a minute.
I had one issue after installing the drivers: X11 was not starting and dropped an error: 

> "NoMatchProduct" is not a valid keyword in this section.

The section was in /usr/share/X11/xorg.conf.d/50-digimend.conf, Line 18:

> NoMatchProduct "MousePen"

After commenting it out, X11 was starting normally.

spbnick, 2018-11-24:
@hvfrancesco, @ChiSamurai, could you please post the output of `sudo usbhid-dump -es -m 256C:006E` for your tablet, where you press the buttons one-by-one, top-to-bottom, with this driver installed?

@ChiSamurai, I must be using a new X.org feature in the driver (`NoMatchProduct`), which doesn't work for you. Could you tell me which distribution version you're using?

Mraize, 2018-11-24:
@spbnick , getting the same as @ChiSamurai, thats what mine spits out
[usbhid-dump.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2612608/usbhid-dump.txt)

tomvdab, 2018-11-24:
Commit 99bd2a3d7205fba77c05255ff13a43b7ac3e42a6 seems to be working for my Huion WH1409 V2, including all buttons. I'm on Arch linux kernel 4.19.2 .

spbnick, 2018-11-25:
Thanks, @tomvdab, that's good to know. Even though this change wasn't addressing your tablet (it should be working without it), that means that I haven't broken it :)

ChiSamurai, 2018-11-25:
@spbnick I'm on Linux Mint 18.3 Cinnamon 64bit (Cinnamon Version 3.6.7, Kernel 4.15.0-39-generic).
X11 Version:

> xdpyinfo|grep -i version
version number:    11.0
X.Org version: 1.18.4

Here's my usbhid dump:
[usbhid-dump.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2612980/usbhid-dump.log)

Since the tablet has two columns of buttons I was going top-left, top-right to finally bottom-left, bottom-right.

spbnick, 2018-11-26:
@hvfrancesco, @ChiSamurai, the reports from usbhid-dump look good, I'm not sure what's happening there.

Could you please execute `echo "module hid_uclogic +pfl" | sudo tee /sys/kernel/debug/dynamic_debug/control` to enable debug logging, reconnect the tablet, and post about 100 lines from the end of dmesg output?

Please also make sure that you don't get any output from evtest when running it for the device named "TABLET Pen Tablet Pad" (or similar, ending with "Pad"), when pressing the frame buttons.

Thank you.

ChiSamurai, 2018-11-28:
@spbnick 
Sorry for the delay, but here is mine:
[debug.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2626076/debug.log)

DamnWidget, 2018-11-30:
Hi @spbnick 
I am in the same boat, the Stylus works perfectly with your PR code but pad buttons are dead, this is the relevant output from dmesg when I unplug and plug the WiFi receiver.

```
[141461.966057] usb 1-2: new high-speed USB device number 12 using xhci_hcd
[141462.147752] hub 1-2:1.0: USB hub found
[141462.147826] hub 1-2:1.0: 4 ports detected
[141462.466066] usb 1-2.1: new full-speed USB device number 13 using xhci_hcd
[141462.607062] uclogic_params_huion_init:760: uclogic 0003:256C:006E.0017: transition firmware detected, not probing pen v2 parameters
[141462.607780] uclogic_params_huion_init:795: uclogic 0003:256C:006E.0017: pen v1 parameters found
[141462.608467] uclogic_params_frame_init_v1_buttonpad:482: uclogic 0003:256C:006E.0017: generic buttons enabled
[141462.608476] uclogic_params_huion_init:806: uclogic 0003:256C:006E.0017: buttonpad v1 parameters found
[141462.608490] uclogic_probe:205: uclogic 0003:256C:006E.0017: parameters:
                .invalid = false
                .desc_ptr =           (null)
                .desc_size = 0
                .pen_unused = false
                .pen.desc_ptr = ffffa061b40c8660
                .pen.desc_size = 92
                .pen.id = 7
                .pen.inrange = inverted
                .pen.fragmented_hires = false
                .frame.desc_ptr = ffffa061b40c8120
                .frame.desc_size = 70
                .frame.id = 0
                .frame.re_lsb = 0
                .frame.dev_id_byte = 0
                .pen_frame_flag = 0x20
[141462.608990] input: TABLET Pen Tablet Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.1/1-2.1:1.0/0003:256C:006E.0017/input/input95
[141462.609427] input: TABLET Pen Tablet Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.1/1-2.1:1.0/0003:256C:006E.0017/input/input96
[141462.609910] uclogic 0003:256C:006E.0017: input,hidraw1: USB HID v1.10 Keypad [TABLET Pen Tablet] on usb-0000:00:14.0-2.1/input0
[141462.613261] uclogic_probe:205: uclogic 0003:256C:006E.0018: parameters:
                .invalid = false
                .desc_ptr =           (null)
                .desc_size = 0
                .pen_unused = true
                .pen.desc_ptr =           (null)
                .pen.desc_size = 0
                .pen.id = 0
                .pen.inrange = normal
                .pen.fragmented_hires = false
                .frame.desc_ptr =           (null)
                .frame.desc_size = 0
                .frame.id = 0
                .frame.re_lsb = 0
                .frame.dev_id_byte = 0
                .pen_frame_flag = 0x00
[141462.613736] input: TABLET Pen Tablet Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.1/1-2.1:1.1/0003:256C:006E.0018/input/input97
[141462.614168] uclogic 0003:256C:006E.0018: input,hiddev1,hidraw2: USB HID v1.11 Mouse [TABLET Pen Tablet] on usb-0000:00:14.0-2.1/input1
[141462.615697] uclogic_probe:205: uclogic 0003:256C:006E.0019: parameters:
                .invalid = false
                .desc_ptr =           (null)
                .desc_size = 0
                .pen_unused = true
                .pen.desc_ptr =           (null)
                .pen.desc_size = 0
                .pen.id = 0
                .pen.inrange = normal
                .pen.fragmented_hires = false
                .frame.desc_ptr =           (null)
                .frame.desc_size = 0
                .frame.id = 0
                .frame.re_lsb = 0
                .frame.dev_id_byte = 0
                .pen_frame_flag = 0x00
[141462.616164] input: TABLET Pen Tablet Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.1/1-2.1:1.2/0003:256C:006E.0019/input/input99
[141462.686048] input: TABLET Pen Tablet Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.1/1-2.1:1.2/0003:256C:006E.0019/input/input100
[141462.686269] input: TABLET Pen Tablet System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.1/1-2.1:1.2/0003:256C:006E.0019/input/input101
[141462.686585] uclogic 0003:256C:006E.0019: input,hidraw3: USB HID v1.11 Keyboard [TABLET Pen Tablet] on usb-0000:00:14.0-2.1/input2
[141462.786056] usb 1-2.4: new high-speed USB device number 14 using xhci_hcd
[141462.920400] usb-storage 1-2.4:1.0: USB Mass Storage device detected
[141462.920940] scsi host1: usb-storage 1-2.4:1.0
[141462.935259] udevd[14255]: failed to execute '/lib/udev/hid-rebind' 'hid-rebind': No such file or directory
[141462.935665] udevd[14257]: failed to execute '/lib/udev/hid-rebind' 'hid-rebind': No such file or directory
[141462.935989] udevd[14260]: failed to execute '/lib/udev/hid-rebind' 'hid-rebind': No such file or directory
[141463.101222] [1711]: Watching system buttons on /dev/input/event24 (TABLET Pen Tablet Keyboard)
[141463.131076] [1711]: Watching system buttons on /dev/input/event26 (TABLET Pen Tablet System Control)
[141463.237610] [1711]: Watching system buttons on /dev/input/event25 (TABLET Pen Tablet Consumer Control)
```

Let me know what else can I do to help you help me

spbnick, 2018-12-01:
Thank you everyone for the information. I'll try with one of the tablets using the older protocol, at home. Meanwhile, somebody, please post the output of `sudo bash -c 'cat /sys/kernel/debug/hid/0003:256C:006E.*/rdesc'` and then also the output of `sudo bash -c 'cat /sys/kernel/debug/hid/0003:256C:006E.*/events'` where you try to press one or two buttons on the tablet frame. I either messed up the report descriptor or massaging the reports here. Thank you.

DamnWidget, 2018-12-01:
Here you are:
Rdesc: https://gist.github.com/DamnWidget/cf82be4d5ddafe10a7ca8b05a036a5f4
Events: https://gist.github.com/DamnWidget/0875b3f9c114ca359e489cc76684a345

In the events report I pressed from button 1 to 12 sequentially in order

spbnick, 2018-12-01:
Thank you, @DamnWidget, I think I found the problem. Stand by for a fix!

spbnick, 2018-12-01:
OK, fixed version available at https://github.com/spbnick/digimend-kernel-drivers/archive/add_support_for_huion_wh1409.zip - please test!

I lost assignment of the virtual frame report ID during latest refactoring. Good that we caught it.

DamnWidget, 2018-12-01:
Confirmed, it works flawlessly with this last update. I think you can merge the PR into the mainline.

Thank you very much for take care of this one.

spbnick, 2018-12-01:
Thank you everyone for your help. Official support for Huion WH1409 is now merged to the drivers.

hvfrancesco, 2018-12-01:
thanks a lot Nikolai!

leoplaw, 2019-10-06:
My Huion 1409 was working out of the box with Kernel 4.20.6-1-default (OpenSuse Tumbleweed).
I recently upgraded the entire system and now have Kernel 5.2.11-1-default. The Huion 1409 table no longer works.

So, I pulled Digimend V9, compiled and installed, still no joy. I then pulled the latest update, compiled and installed, still broken.

I thoroughly went over all of the instructions.

It seems that the X.org config needs the identifier to be "PenTablet" and not "TABLET".

Section "InputClass"
    Identifier "InputTable"

with this correctly set, xinput reports the following pointers

PenTablet  Pen stylus                     id=9    [slave  pointer  (2)]
PenTablet  Pad pad                        id=10   [slave  pointer  (2)]
PenTablet  Consumer Control        id=12   [slave  pointer  (2)]

and Keyboards

PenTablet  Keyboard                       id=11   [slave  keyboard (3)]
PenTablet  System Control                 id=13   [slave  keyboard (3)]

However, the tablet still does not work.
xsetwacom --list
PenTablet  Pen stylus                   id: 9   type: STYLUS    
PenTablet  Pad pad                      id: 10  type: PAD   

lsusb
Bus 002 Device 003: ID 256c:006e  

I also followed the instructions to remove the kernel modules.
sudo modprobe -r hid-kye hid-uclogic hid-polostar hid-viewsonic

and then
dksm install digimend/9 --force

reboot, still nothing works.

After much experimenting I came to the conclusion the issue is with the module hid-uclogic and removed hid-uclogic with modprobe and then specifically installed the compiled V9 resulting in the following error.

insmod /lib/modules/5.2.11-1-default/extra/hid-uclogic.ko
insmod: ERROR: could not insert module /lib/modules/5.2.11-1-default/extra/hid-uclogic.ko: Unknown symbol in module

At this point I don't know what further to do, and have resorted to booting from Kernel 4.20.6-1-default again to make the Huion usable again.

I can provide whatever output you need to solve the issue.