DIGImend/digimend-kernel-drivers #218: XP Pen Deco 03 Support

Lite5h4dow, 2018-11-15:
Just wondering if you were planning on supporting the XP Pen Deco 03? would love to sculpt on my linux machine

bianchifan, 2019-02-23:
One more!

airosa-id, 2019-12-05:
Yes! Yes! Please!

tuanpham96, 2020-03-28:
Same here please!

applesaucyy, 2020-07-30:
Here's some basic diagnostic info for any willing devs. Many would appreciate, considering the xp-pen provided beta drivers still need a lot of work 
Manufacturer: XP-PEN
Model Name: Deco 03
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5003260/descriptors.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5003261/hid_report_descriptors.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5003264/pen_coords.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5003266/pen_tilt.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5003267/pen_pressure.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5003268/pen_buttons.txt)
[frame_dials.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5003269/frame_dials.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5003272/frame_buttons.txt)

zetagabriels, 2020-11-27:
Hello, I am interested in contributing for this tablet, as the existing Deco 03 driver is very lackluster, and tends to "stick" when using commands containing `ctrl` or `shift`. Do you have any recommendations on where I should start looking before I begin?

bianchifan, 2021-01-21:
There is a brand new UI for the Linux Beta driver on XP-Pen Homepage, much more functions..
https://www.xp-pen.com/download/file/id/1936/ext/zip.html
To install, I must change owner and so on, at least it runs fine now without root/SU

zetagabriels, 2021-11-06:
I've opened a pull request (#577) for stylus support. Pad buttons and wheel are WIP.