DIGImend/digimend-kernel-drivers #221: XP-Pen Artist 15.6

xiyizi, 2018-11-20:
Bit of a noob with all this stuff, but  if this helps anyone then perhaps it's all for the best.

So, in **/usr/share/X11/xorg.conf.d/50-tablet.conf** I put:
```
Section "InputClass"

  Identifier "Tablet"
    Driver "evdev"
    MatchDevicePath "/ dev / input / event *"
    MatchUSBID "28 bd: 000 c"

EndSection
```

And then in: **/usr/share/X11/xorg.conf.d/99-modified-calibration.conf**
```

Section "InputClass"

  Identifier "calibration"
    MatchUSBID "28 bd: 000 c"
    Option "MinX" "160"
    Option "MaxX" "34519"
    Option "MinY" "164"
    Option "MaxY" "19549"
    Option "SwapXY" "0" # unless it was already set to 1
    Option "InvertX" "0" # unless it was already set
    Option "InvertY" "0" # unless it was already set 

EndSection

```

The pressure sensitivity and such appears to work. Buttons seem to be working too. Not sure how to change what they do without catching the input from **xinput** or something.

The calibration is off unless I unify my monitors. Possibly halving one of the Max[x/y] values depending on monitor orientation would do it? 

Not entirely sure this is the right place to post this but thought someone might find this info useful.

Barracuda77, 2018-12-08:
Hi, I have the Artist 22E pro. and for calibration, I used the xinput transformation matrix explained here
https://wiki.archlinux.org/index.php/Calibrating_Touchscreen 
and it works great no offset at all.
but I have one question for you does your top stylus button work as a right click?

airfield20, 2018-12-27:
I have the artist 22E as well and the x configuration from the README file worked great. from there I was able to open up the wacom settings panel on ubuntu and calibrate the device/change button configurations.

Barracuda77, 2018-12-27:
@airfield20 Can you elaborate more on your setup please? which README file are you referring to? and which driver is running the Tablet and are you on Ubuntu 18 or earlier release? cause a lot of people managed to get the tablet running on 16 but on 18 I had to fiddle around a bit.
Thanks a lot!

airfield20, 2018-12-27:
I am running on ubuntu 16.04 . I used this 
`Section "InputClass"
    Identifier "Tablet"
    Driver "wacom"
    MatchDevicePath "/dev/input/event*"
    MatchUSBID "<VID>:<PID>"
EndSection`

and then after a reboot when i typed `xsetwacom list` it showed up as a wacom input device.

airfield20, 2018-12-30:
I didnt have to use the wacom driver to do it. Since Im on 16.04 I used the unity settings panel which worked for me.

paint-rulz, 2019-02-17:
Hi :-) So wacom drivers need installing, and then the 2 .conf files placing, whichever version of the xp-pen 15.6 it is? Ptxconf works on Manjaro to fix the display for present Ugee, so may be useful here?

rthickling, 2019-07-15:
[ptxconf](http://wenhsinjen.github.io/ptxconf/) worked well for me with a XP-Pen Artist 15.6 and Dell Inspiron 14 3000 using Ubuntu 18.04.

Wcubed, 2022-02-11:
I made a small calibration script as well. In case the other options don't work or result in offsets: https://github.com/Wcubed/calibrate_pen_display