DIGImend/digimend-kernel-drivers #222: About the Huion H610Pro V2 comparing with One by Wacom.

sksdutra, 2018-11-20:
Hello, I have a One by Wacom, but bought a **Huion H610Pro V2 (8192)**.

So I installed the Digimend drivers on my Manjaro system (XFCE4) and create the file 52-tablet on my system and the script to configure the lateral buttons, but testing on Krita and comparing it to One by Wacom's performance, I noticed that the cursor slides annoyingly slower than when moved with Wacom's and is much more difficult to control the pressure.

Is there any way I can change tablet options to improve the experience with it (Cursor speed and pressure)? I only use Linux and I already think in sell it, considering that the experience was not very pleasant as using Wacom.

spbnick, 2018-11-24:
@sksdutra Unless you're using the tablet in relative mode, the cursor speed on the screen is only dependent on the ratio between the screen size and the tablet size. So, if e.g. you were using the small Wacom One, which is 8.3"x5.7" and you're now using Huion H610Pro V2, which is 10"x6.25", on the same screen, then the Huion will feel "slower", because you will have to move your hand further to move the same distance on the screen. The trade-off, of course, is precision.

That is unless you talk about latency, i.e. the time it takes for the cursor on the screen to react to pen movement. That could be different in theory, but I don't think that should happen.

Regarding pressure sensitivity, you can adjust pressure curve in applications you're using, or change the `PressureCurve` with xsetwacom. See https://wiki.archlinux.org/index.php/Wacom_tablet#Pressure_curve and http://www.davidrevoy.com/article182/calibrating-wacom-stylus-pressure-on-krita for explanations and examples.

sksdutra, 2018-11-24:
@spbnick thank you for your considerations.

closed 2018-12-01 by spbnick