DIGImend/digimend-kernel-drivers #223: Monoprice 10594 Frame Buttons

lunix4all, 2018-11-21:
I'd like to bring this issue up again: https://github.com/DIGImend/digimend-kernel-drivers/issues/42

I have a Monoprice 10594, which is just a rebranded Huion H610 Pro. The pen works for drawing in Krita, but none of the buttons on the side work. This seems strange, since apparently both the Huion H610 and H610 Pro have been listed on the supported tablets page for quite some time. Is there anyway I can make my tablet use the H610 Pro drivers?

Thanks in advance!

lunix4all, 2018-11-26:
Is there any way to get the buttons working without using the Wacom driver?

spbnick, 2018-12-01:
@lunix4all, using the Wacom driver is the simplest approach so far. If you install the latest development version of the drivers, you won't have to configure the Wacom driver to handle your tablet, only to configure the buttons with "xsetwacom".