DIGImend/digimend-kernel-drivers #226: Xinput coordinate transform matrix.

Cpt-Ktw, 2018-11-26:
I find it extremely confusing. 
What should my transform matrix look like if I have :

1. Dual monitor setup one 1920*1080 and another 3840*2160(i need to be active)

2. 16/10 tablet active area while the monitor is 16/9

spbnick, 2018-12-01:
This is an involved case, and unfortunately I don't have enough time to remember how the math works and help you. However, if you search the web for something like "calculate coordinate transformation matrix for xinput" or such, you should be able to find some articles, answers, scripts, and even spreadsheets to help you figure it out.

spbnick, 2019-01-05:
Closing, as there's no further communication from the reporter. @Cpt-Ktw, please reopen, if you still need help.

closed 2019-01-05 by spbnick