DIGImend/digimend-kernel-drivers #227: Latest Ugee M708 has different hardware, diagnostics attached.

Kirtai, 2018-11-30:
I just bought a new Ugee M708 which didn't work with the digimend drivers due to a hardware refresh.

New USB id is 28bd:0907

This particular version comes with a batteryless pen that claims 8192 pressure values.
See attached files for the detailed information on this hardware.
I have access to both an old and new model M708 if that's useful.

The default linux/X drivers in Debian Buster (testing) partially work:
Tip pressure, absolute pen movement, frame buttons send keys, tip button is left click, lower barrel button is middle click, upper barrel button is ERASER, not right click for some weird reason.

"xsetwacom list" shows the stylus but not the buttons.

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2631711/descriptors.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2631712/frame_buttons.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2631713/hid_report_descriptors.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2631714/pen_buttons.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2631715/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2631716/pen_pressure.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2631717/pen_tilt.txt)

spbnick, 2018-12-01:
Thank you for the report, @Kirtai. Could you please give a link to the product that you purchased, and also please post the output of the latest development version of `uclogic-probe` for your tablet?

Kirtai, 2018-12-01:
I got the tablet from Amazon UK [here](https://www.amazon.co.uk/gp/product/B072M76LNZ/) and the uclogic-probe output is attached.
[uclogic-probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2635937/uclogic-probe.txt)

spbnick, 2018-12-01:
@Kirtai, please try this version of the driver: https://github.com/spbnick/digimend-kernel-drivers/archive/support_ugee_m708_v2.zip

@icefairy64, please open a new issue and attach diagnostics, as your tablet's USB product ID is different.

icefairy64, 2018-12-01:
@spbnick, done: #233

Kirtai, 2018-12-03:
With that driver, the tablet is unresponsive.
Attached are logs from X and dmesg
[Xorg.0.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2637736/Xorg.0.log)
[messages.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2637737/messages.txt)

spbnick, 2018-12-03:
Thank you, Kirtai, I'll try to figure out what's going on. Do you have another Ugee tablet, one with product ID 1227, which I see in the log?

Kirtai, 2018-12-03:
No, it's the same tablet.

This is strange.
With the driver you provided installed, when I plug in the the first time, it attaches as ID 1227 .
If I remove it then plug it in again, it attaches as ID 0907 and sometimes doesn't detach when I unplug it.

With your driver removed, it appears as ID 0907 every time and detaches properly.

Attached is lsusb output from first and second connection.
[lsusb.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2638964/lsusb.txt)

spbnick, 2018-12-03:
Ha-ha, wonderful. I'll have to think on what to do about this.

Kirtai, 2018-12-04:
OK, I've been experimenting with this tablet and have got some things working you may find useful.

If I install the X11 config you provided in your driver in [this entry](https://github.com/DIGImend/digimend-kernel-drivers/issues/227#issuecomment-443449075), the tablet is partially recognized by the wacom driver.

xsetwacom recognises the Stylus and what appears to be a spurious "Mouse touch" device.

- Setting things like the active area of the Stylus work both with xsetwacom & with the KDE config tool.
- The lower stylus barrel button can be remapped from middle click to right click.
- The upper stylus barrel button can NOT be remapped and does not seem to do anything. While pressed, in Krita it prevents the brush from drawing and in Mypaint, it seems to stop pressure working.
- The tablet buttons can NOT be detected or changed in any way

I do not have any digimend drivers installed according to lsmod. (I think)

Attached files contain the results of

- xsetwacom list devices
- xsetwacom -s --get "10 inch PenTablet stylus" all
- xsetwacom -s --get "10 inch PenTablet Mouse touch" all
- lsmod

and the X11 config I'm using.

Thanks for helping with this :)

[m708v2-list.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2643074/m708v2-list.txt)
[m708v2-stylus-s.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2643075/m708v2-stylus-s.txt)
[m708v2-Mouse-s.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2643077/m708v2-Mouse-s.txt)
[lsmod.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2643080/lsmod.txt)
[50-digimend.conf.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2643081/50-digimend.conf.txt)

Kirtai, 2018-12-04:
Also added the X and dmesg logs with the partial wacom driver

[dmesg.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2643107/dmesg.txt)
[Xorg.0.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2643108/Xorg.0.log)

spbnick, 2018-12-08:
@Kirtai, @blackorange1905, please my another attempt at making Ugee M708 V2 work: https://github.com/spbnick/digimend-kernel-drivers/archive/support_ugee_m708_v2_v2.zip

Kirtai, 2018-12-12:
With the new driver the tablet is unresponsive. It has a light on it that flickers whenever an event is sent, this remained dark the whole time. It only flickers once when I plug it in.

Weirdly, the KDE config tool and xsetwacom both do show information on the tablet.

Attached are the log files.
[dmesg.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2672929/dmesg.txt)
[x.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2672932/x.log)
[m708v2-list.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2672933/m708v2-list.txt)
[m708v2-stylus.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2672934/m708v2-stylus.txt)
[xinput.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2672935/xinput.txt)
[m708v2-pad.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2672936/m708v2-pad.txt)
[lsmod.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2672943/lsmod.txt)

spbnick, 2018-12-12:
Oh well, at least it doesn't change its VID:PID anymore. Please write if you notice that again. I wonder why it's initializing two of the three interfaces twice. Could you please execute `echo "module hid_uclogic +pfl" | sudo tee /sys/kernel/debug/dynamic_debug/control`, reconnect your tablet and then post the last 30 or so lines of `sudo dmesg` output?

Thank you.

Kirtai, 2018-12-13:
Here it is

[dmesg-debug.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2677733/dmesg-debug.txt)

spbnick, 2018-12-14:
Thank you. This seems OK so far, except for some reason the other two interfaces are initialized *twice*. Not sure why. Could you give me some `usbhid-dump` output for your tablet, from you making a stroke with the pen with this driver installed? And also `evtest` output for the pen and the pad devices (for the stylus and the frame buttons respectively)?

Kirtai, 2018-12-14:
Here you go.

dump-pen contains a pen stroke.
dump-buttons contains the stylus button presses.
dump-pad shows the frame buttons from top to bottom.

The evtests didn't show anything while using the device, just the descriptor. 

[dump-pad.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2680063/dump-pad.txt)
[dump-buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2680049/dump-buttons.txt)
[dump-pen.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2680050/dump-pen.txt)
[evtest-pad.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2680051/evtest-pad.txt)
[evtest-pen.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2680052/evtest-pen.txt)

spbnick, 2018-12-15:
Thank you, @Kirtai. This looks like somebody butchered the usual UC-Logic firmware. It uses the old protocol's report ID for the pen, but reports with the new protocol. At the same time, it responds to frame button initialization, but doesn't actually switch the protocol, only the interface number and continues reporting like a generic keyboard, with fixed shortcuts. Not to mention the trick where it resets and begins to identify as a different tablet, which we've seen before.

I think I can make the pen work with somewhat invasive changes to the driver, but so far the frame buttons would have to stay that way. That is unless you'd like to try capturing the traffic from the Windows driver with Wireshark, so that I can see how it actually initializes and uses the tablet. Write if you'd like to do that and need help. Meanwhile I'll see if I can throw together another test version for you.

spbnick, 2018-12-15:
OK, here's another attempt, trying to make the pen work, but without the frame buttons: https://github.com/spbnick/digimend-kernel-drivers/archive/support_ugee_m708_v2_v3.zip

Please try it, and if it doesn't work, please post the `dmesg` output with driver debugging enabled, when plugging in the tablet, the output of `sudo bash -c "cat /sys/kernel/debug/hid/0003:28BD:0907.*/rdesc"`, and then `sudo evtest` for the pen interface.

Kirtai, 2018-12-16:
Done. I noticed that this time evtest shows two mouse devices and no stylus & that the frame keys work until the pen touches the tablet at which point everything stops working. evtest shows nothing at all for the stylus.

I'm going to try the windows/wireshark thing with the laptop I've been testing this with. I've been meaning to do this for some other hardware I have anyway. Anything I should be particularly aware of?

[debug.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2684044/debug.txt)
[dmesg.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2684045/dmesg.txt)

Kirtai, 2018-12-16:
Here's a windows wireshark capture of plugging in the tablet, moving the stylus and pressing the frame buttons.

If you need more details, just let me know.

[windows.pcapng.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/2684069/windows.pcapng.zip)

Kirtai, 2019-01-11:
Do you require any more information or traces?

spbnick, 2019-01-11:
Thank you, @Kirtai, but this is enough for now. I approached this a couple times and have some ideas what to do, but I need more time to handle it. At the moment my main concern is contributing the current driver to the upstream kernel and that will take a while.