DIGImend/digimend-kernel-drivers #228: DKMS install on Raspberry Pi

ALTracer, 2018-11-30:
Just pulled master from DIGImend/digimend-kernel-drivers to my Raspberry Pi 3 running Raspbian 9 Stretch armhf, installed dkms and raspberrypi-kernel-headers, and went `sudo colormake dkms_install`. Except our 4.4.76+ something modules already have `hid_uclogic` version 9, and that module doesn't work with Huion H1060P. Luckily DKMS offered me to `--force`.
Solution: `sudo dkms remove -m digimend/9 --all`, `sudo dkms install --force -m digimend/9`.
Result: tablet working as mouse and in GIMP. I just wish there was an easy way to configure it via some widgets.

spbnick, 2018-12-01:
Thank you for the report @ALTracer, is your problem that `sudo make dkms_install` didn't install some modules? If so, did you see [DKMS issues preventing correct installation](https://github.com/DIGImend/digimend-kernel-drivers#dkms-issues-preventing-correct-installation) chapter in the README.md?

spbnick, 2019-01-05:
Closing this issue, as the reporter is not responding. @ALTracer, please reopen if you still need help.

closed 2019-01-05 by spbnick

ALTracer, 2020-05-29:
@spbnick Issue solved by updating dkms from upstream APT.
Raspbian Buster, v4.19.118, `dpkg-buildpackage --no-sign` creates functional packages from v9 tag and v10 current master branch. Confirmed working with Huion H1060P, no wacom.