DIGImend/digimend-kernel-drivers #229: Before rebooting HUION 1060 plus worked, after getting 'input irq status -75 received'

nate-byrnes, 2018-11-30:
I must be missing something, I've gone through #77 and #82 and cannot seem to get back to a working config... When I first installed the git drivers the tablet worked. Later in the day the computer locked up in Krita and I rebooted. After reboot, the tablet didn't work, and even after ```make uninstall && make install```, and trying the dkms debian package, I am still getting the irq status dmesg.... what am I missing? 

Thanks in advance.

spbnick, 2018-12-01:
Let's see what version of the driver is loaded. Please plug in the tablet and post the output of  `/sbin/modinfo hid-uclogic` and `cat /sys/module/hid_uclogic/version`

nate-byrnes, 2018-12-01:
```
filename:       /lib/modules/4.15.0-39-generic/updates/dkms/hid-uclogic.ko
version:        8
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     99F6B16F19F4390474B2D9F
alias:          hid:b0003g*v000028BDp00000042
alias:          hid:b0003g*v000028BDp00000075
alias:          hid:b0003g*v000028BDp00000071
alias:          hid:b0003g*v000028BDp00000074
alias:          hid:b0003g*v00002179p00000053
alias:          hid:b0003g*v00005543p00003031
alias:          hid:b0003g*v00005543p00000047
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        hid
retpoline:      Y
name:           hid_uclogic
vermagic:       4.15.0-39-generic SMP mod_unload 
```

I do not have a version file in that directory. There is a srcversion file though.... see below:

```
root@tool:~/digimend-kernel-drivers# ls /sys/module/hid_uclogic/
coresize  holders   initstate  refcnt    srcversion  uevent
drivers   initsize  notes      sections  taint

root@tool:~/digimend-kernel-drivers# cat /sys/module/hid_uclogic/srcversion 
1DD838FD354887326DC6A47
```

spbnick, 2018-12-08:
The fact that you don't have the "version" file means that the stock driver coming with the kernel is getting loaded. Please download and install [the latest development release](https://github.com/DIGImend/digimend-kernel-drivers/archive/master.zip), see if you get the `version` file, and if you don't try to load the compiled driver manually by running `sudo insmod hid-uclogic.ko` in the source directory, then see if you get the `version` file, and post last 100 lines of `sudo dmesg` output.

spbnick, 2019-01-05:
Closing this issue, as the reporter is not responding. @nate-byrnes, please reopen, if you still need help.

closed 2019-01-05 by spbnick

nate-byrnes, 2019-01-05:
Sorry about the radio silence. IIRC the re-install after a more robust manual cleanup of the prior efforts got me up and running. I likely just dorked up my install. Your reply did get pointed in the right direction. Thanks very much!!

spbnick, 2019-01-05:
Ah, great, thanks for the update :)