DIGImend/digimend-kernel-drivers #233: Add support for new revision of XP-Pen Deco 01

icefairy64, 2018-12-01:
I have recently got a new XP-Pen Deco 01, which appears to have different USB ID (`28bd:0902`).

The only problem that I notice for now is that `xsetwacom` does not appear to handle frame buttons properly (I could `--get` only buttons 1-8).

Collected diagnostics are in the attachment:

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2636236/descriptors.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2636237/frame_buttons.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2636238/hid_report_descriptors.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2636239/pen_buttons.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2636240/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2636241/pen_pressure.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2636242/pen_tilt.txt)

spbnick, 2018-12-08:
If the frame buttons are the only problem, please post the output of `xsetwacom list`, and then `xinput test` for your tablet's "pad" device, where you press the buttons in order, top-to-bottom.

icefairy64, 2018-12-10:
I've run these command while still forcing `wacom` driver to handle this tablet, and got only `TOUCH` and `STYLUS` devices in `xsetwacom list` command output:

[xsetwacom_list.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2664758/xsetwacom_list.txt)

Running `xinput test` on both devices and pressing frame buttons did not result in any output.

When I removed this tablet from overrides in `xorg.conf.d`, there were no more devices on `xsetwacom list` output, but `xinput test` actually did output something useful this time:

[xinput_test_nowacom.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2664775/xinput_test_nowacom.txt)

Please note that tablet sent some keystrokes on some buttons (as far as I understand, they were Ctrl+Z on 5, Ctrl+S on 6, Ctrl+ [+] on 7 and Ctrl+ [-] on 8).

Also, I don't seem to be able to remap stylus buttons either, but I will try to investigate this further, as I did not have much time to experiment with `xsetwacom` yet.

Z-Map, 2019-01-05:
The issue seems to be related to [this one](https://github.com/DIGImend/digimend-kernel-drivers/issues/178)
The  USB ID is the same.
I will investigate too when I can.

kaiec, 2019-03-19:
I got the same device from Amazon Germany and can confirm all of the behaviour above. Let me know if I can help testing.

loopside, 2019-06-10:
Is the support coming? I just noticed that my Deco 01 v2 (28BD:0902) works exactly the same after uninstalling digimend to what it was when I had it installed (top pen button not working, tablet buttons working, but Keyboard entry in xinput that handles it says that this device has no buttons).

Proprietary drivers aren't perfect too because you either have to have them opened on your desktop in order to work on your tablet or create wacky-hacky solutions to hide it. And you can't map work area which makes the tablet literally useless in e.g. osu! Leaving me with option to e.g. use proprietary driver for drawing and stuff and OOTB support for osu! or idle.

If it's important, my OS is Arch with GNOME, using AUR package but I assume it's just pure digimend packaged to be easily installable for Arch. If you need some additional logs or info from me, I'll hop in and send you if I don't forget about it. Thanks in advance!