DIGImend/digimend-kernel-drivers #235: How do I get the buttons to work H950P

FazalAAli, 2018-12-05:


FazalAAli, 2018-12-05:
I'm going to use the idea in this guide when I get home and report back:

https://askubuntu.com/questions/152297/how-to-configure-extra-buttons-in-logitech-mouse

FazalAAli, 2018-12-05:
Update: That hasn't worked. Someone halp

spbnick, 2018-12-08:
@FazalAAli, make sure your tablet is handled by the Wacom driver and then follow the README.md configuration instructions: https://github.com/DIGImend/digimend-kernel-drivers#configuring-wacom-xorg-driver

spbnick, 2019-01-05:
Closing this issue, as the reporter is not responding. @FazalAAli, please reopen, if you still need help.

closed 2019-01-05 by spbnick

realmadpuppy, 2019-02-07:
Hello I was on the iirc chat and left messages and left the window open for a day and a half without response, But adding to this thread, I have the same problem in Ubuntu 18.10 I got the pad working great with the .8 release of DIGimend (.9 didn't work at all) But, I would love to get full functionality with my H950P and the buttons just don't work, Interestingly, I was in the KDE joystick config and the buttons post in that program.I used the Ulogic probe and Xinpit utilites and the post from top to bottom a 1,2,3,8,9,10,11,12. and if I press them while on the desktop random menu's pop up and windows close and browser tabs change so it is doing something, what I would like to know is do I uninstall the .8 drivers and install and configure the H950P with the wacom utilities? I really don't want to toast what is already working well to get the buttons configured properly but if it will work overall better with the wacom drivers and configuration I would like to give it a shot.
thank you in advance.