DIGImend/digimend-kernel-drivers #236: Huion H610 Pro V2 - Tablet Diagnostics

Browncoat97, 2018-12-07:
[The tablet on Amazon for reference.](https://smile.amazon.com/gp/product/B07DPC98DT/ref=od_aui_detailpages01?ie=UTF8&psc=1)

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2655357/descriptors.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2655363/hid_report_descriptors.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2655365/pen_coords.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2655366/pen_tilt.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2655367/pen_pressure.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2655368/pen_buttons.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2655369/frame_buttons.txt)

spbnick, 2018-12-08:
Thank you for the diagnostics, @Browncoat97! Could you please also post the output of [uclogic-probe](https://github.com/DIGImend/uclogic-tools) for your tablet?

Browncoat97, 2018-12-08:
Absolutely!
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2660116/probe.txt)
[decode.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2660117/decode.txt)

spbnick, 2018-12-09:
Thank you! This looks like it should work OK. Do you have any problem with making your tablet work?

andresm, 2018-12-10:
It works fine for me, but without tilt support.

Browncoat97, 2018-12-11:
The activity light comes on when I press any of the buttons and when the pen is near the tablet, however nothing actually happens when I do these things.

I've tried on Manjaro kernel 4.19.6 and Fedora kernel 4.16.11.

Here's the xinput compared to my H420 that is working if that helps.

H610 Pro V2
Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ USB Optical Mouse                       	id=9	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ HUION Huion Tablet Pen                  	id=8	[slave  keyboard (3)]
    ↳ Metadot - Das Keyboard Das Keyboard     	id=10	[slave  keyboard (3)]
    ↳ Metadot - Das Keyboard Das Keyboard     	id=11	[slave  keyboard (3)]
    ↳ Eee PC WMI hotkeys                      	id=12	[slave  keyboard (3)]

H420
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ USB Optical Mouse                       	id=9	[slave  pointer  (2)]
⎜   ↳ HUION H420 Consumer Control             	id=8	[slave  pointer  (2)]
⎜   ↳ HUION H420 Pad                          	id=15	[slave  pointer  (2)]
⎜   ↳ HUION H420 Pen Pen (0)                  	id=16	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ Metadot - Das Keyboard Das Keyboard     	id=10	[slave  keyboard (3)]
    ↳ Metadot - Das Keyboard Das Keyboard     	id=11	[slave  keyboard (3)]
    ↳ Eee PC WMI hotkeys                      	id=12	[slave  keyboard (3)]
    ↳ HUION H420 Consumer Control             	id=13	[slave  keyboard (3)]
    ↳ HUION H420 Pen                          	id=14	[slave  keyboard (3)]

@andresm Did your tablet 'work out of the box'?  If you don't mind can you tell me what driver your'e using or what method you got yours to work? I could be missing something painfully obvious.

andresm, 2018-12-12:
I use Debian Stretch and the digimend-dkms 8 deb package. I also use the wacom tablet driver for the shortcut buttons. I think the light is controlled by the tablet hardware, so that is no indication that the software is working.

I had a similar problem with the tablet. The cause was that I forgot to unload the old drivers:

sudo modprobe -r hid-kye hid-uclogic hid-polostar hid-viewsonic

closed 2018-12-23 by Browncoat97

Browncoat97, 2018-12-23:
Thank you for helping me and leading me in the right direction.

If anyone else is 'new' and reading this,

1. Install the digimend-kernel-drivers-dkms-git and linux-headers with your package manager.
2. sudo modprobe -r hid-kye hid-uclogic hid-polostar hid-viewsonic  (if something did not install from above you will get a Fatal error, try again)
3. reboot
4. have fun

spbnick, 2018-12-29:
@andersm, I answered under your commit.

vicarma, 2019-05-08:
I am trying to get my daughters Huion H610ProV2 to work also on Debian Stretch, but so far we are unable to even have the raspberry pi 3b+ recognise the tablet.  I have followed the threads for updating the kernels, but I receive the following response:

pi@raspberrypi:~ $ sudo apt-get install -y "linux-headers-$(uname -r)"
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-4.14.98-v7
E: Couldn't find any package by glob 'linux-headers-4.14.98-v7'
E: Couldn't find any package by regex 'linux-headers-4.14.98-v7'

As a newcomer to this linux environment, I do not know how to proceed and would appreciate some help.

disappear123df, 2019-06-09:
how did you get yours to work im trying everything it doesn't show anything when i do xsetwacom list
but if i move the pen it move in the screen  i have no pressure sensing and do i have to put those file you posted at xorg.conf.d

andresm, 2019-06-09:
@vicarma: sudo apt-get install raspberrypi-kernel-headers. But I am not sure you should be compiling kernel modules from source if you are new to linux.

@disappear123df: I guess xsetwacom wont work unless you activate the wacom driver, so yes, I guess you have to place those files to configure xorg(I did).