DIGImend/digimend-kernel-drivers #238: Huion New 1060+ working 99% with 2 minor issues

aceman500, 2018-12-10:
Thank you for your commitment to the digimend drivers!

My huion new 1060+ is working well with 2 minor issues:
1 If the tablet is plugged in on bootup it does not function.
lsmod shows hid_uclogic  has loaded but tablet does not work. Removing that module and plugging the tablet back in fixes it. Also if its not plugged in at boot then you plug it in it works.

2 It appears the table goes to sleep after inactivity. (you need to touch the table to wake it)
After that its seems to forget the monitor mapping and need to re run xsetwacom MapToOutput every time.

System:
Linux Fedora 28
Kernel ver: 4.18.16
using Xorg driver

spbnick, 2020-04-26:
Was able to reproduce this on a Fedora 29 VM, investigating.

spbnick, 2020-04-26:
This seems to be a problem with the way Fedora loads kernel modules at boot. When booting, it loads the stock kernel hid-uclogic module, even though the depmod configuration says it should load the one from the "extras" directory instead, the one installed by digimend-kernel-drivers. After it booted, you can unload the stock module with `modprobe -r hid-uclogic` and load the new module with `modprobe hid-uclogic`. This needs to be raised with Fedora developers.

spbnick, 2020-04-26:
I asked Red Hat kernel developers. Let's see if I get any answer.

spbnick, 2020-05-17:
A fix for the first issue is up in #394.

spbnick, 2020-05-17:
One approach to solve the second issue: https://askubuntu.com/questions/508236/how-can-i-run-code-whenever-a-usb-device-is-unplugged-without-requiring-root/516336#516336

spbnick, 2020-05-18:
#394 is merged.

commit c17d0128 (2020-05-18): Add installed modules to initramfs with dracut

Fedora seems to be adding the stock tablet drivers to its initramfs. As
a result, newer tablets with VID:PID pairs already supported by the
stock modules won't work, if plugged before boot, since they will be
handled by the stock module.

Explicitly add our modules to Fedora initramfs, so those tablets work
even if plugged in before boot.

Fixes #238
closed 2020-05-18 by spbnick