DIGImend/digimend-kernel-drivers #239: How do I enable proprietary mode in a HUION tablet?

sprite-1, 2018-12-17:
While trying to figure out a way to get my HUION tablet to work with a WINE application, I stumbled across [this thread](https://forums.linuxmint.com/viewtopic.php?t=154032) which talks about a "proprietary mode". I downloaded the huion-tools linked there (now called uclogic-tools) and installed it but I'm not sure what to do from this point onwards.

How exactly do I enable the proprietary mode?

spbnick, 2018-12-26:
Hi @sprite-1, I find it hard to imagine how enabling proprietary mode alone could help you set up the tablet with WINE. I feel you're going the wrong way. Still, there are at least two proprietary modes that Huion tablets support now, and one of them you can enable by running uclogic-probe on your tablet. Not sure what you're gonna do from there.

I would recommend you instead give the digimend-kernel-drivers a try and report with the results and the information on your tablet here.

sprite-1, 2018-12-26:
Heya @spbnick,

> I find it hard to imagine how enabling proprietary mode alone could help you set up the tablet with WINE. I feel you're going the wrong way.

Truth be told, I'm not exactly sure what I'm doing 😅 I do have the digimend drivers installed and the tablet works perfectly with Krita. However, I'm trying to get MediBang Paint (which is running through WINE) to detect the pressure sensitivity which it's not doing right now.

> and one of them you can enable by running uclogic-probe on your tablet. Not sure what you're gonna do from there.

I have done this but it didn't seem to do anything.

spbnick, 2018-12-26:
I see. I didn't have time yet to get any tablet working with applications running under Wine. Since you got your tablet working with Krita, you might have good luck asking Wine folks for help.

sprite-1, 2018-12-26:
Alright, that's understandable. I was just solely reliant on that one thread I linked that talked about being able to make it work under WINE by enabling some proprietary mode.