DIGImend/digimend-kernel-drivers #245: "NoMatchProduct" is not a valid keyword in this section.

Narann, 2019-01-02:
I use the v9 .deb package on Ubuntu 16.04.5 LTS.

X can't start because `NoMatchProduct` keyword in its xorg configuration file:

```
"NoMatchProduct" is not a valid keyword in this section.
```
Full X log:
```
X.Org X Server 1.18.4
Release Date: 2016-07-19
[    82.560] X Protocol Version 11, Revision 0
[    82.575] Build Operating System: Linux 4.4.0-128-generic x86_64 Ubuntu
[    82.605] Current Operating System: Linux foo 4.15.0-43-generic #46~16.04.1-Ubuntu SMP Fri Dec 7 13:31:08 UTC 2018 x86_64
[    82.605] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-4.15.0-43-generic ...
[    82.648] Build Date: 10 August 2018  09:33:05AM
[    82.676] xorg-server 2:1.18.4-0ubuntu0.8 (For technical support please see http://www.ubuntu.com/support) 
[    82.689] Current version of pixman: 0.33.6
[    82.716] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[    82.716] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    82.774] (==) Log file: "/var/log/Xorg.0.log", Time: Wed Jan  2 15:05:22 2019
[    82.789] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    82.820] Parse error on line 18 of section InputClass in file /usr/share/X11/xorg.conf.d/50-digimend.conf
	"NoMatchProduct" is not a valid keyword in this section.
[    82.836] (EE) Problem parsing the config file
[    82.836] (EE) Error parsing the config file
[    82.836] (EE) 
Fatal server error:
[    82.868] (EE) no screens found(EE) 
[    82.883] (EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
[    82.944] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[    82.976] (EE) 
[    82.991] (EE) Server terminated with error (1). Closing log file.
```

I juste replaced `/usr/share/X11/xorg.conf.d/50-digimend.conf` by `/usr/share/X11/xorg.conf.d/50-digimend.conf.bak` fix the problem. I don't use any tablet in the file.

Others seems to have this problem too:

https://forums.linuxmint.com/viewtopic.php?t=283685&p=1566899#p1567021
https://www.linuxmintusers.de/index.php?topic=52520.0

I don't know if `NoMatchProduct` is new or deprecated but this seems to be the problem.

As I have a solution you can close this ticket if you want but if we can use a more common keyword that would be great.

Thanks again for your hard work ! :)

spbnick, 2019-01-02:
Thank you for the report, @Narann! Yes, I saw this problem before and forgot to fix it before the release :(. You can safely remove that line as you probably don't have the tablet it's for. I'll think about this on the weekend and will probably remove it in the tree as well.

BruceZ2015, 2019-11-05:
I just got this same error after downloading the digimend-dkms_9_all.deb driver on Sunday and installing it for my Huion H610PRO V2 on ubuntu 16.04 kernel 4.4.0-166-generic.  Error found only after great distress.  Commenting out line

NoMatchProduct "MousePen"

allowed system to boot.  I had no cursor, but this was the case without the driver as well.  I could scroll, open things, and draw.