DIGImend/digimend-kernel-drivers #246: Help needed - Raspbian and Huion H950P (not working)

NunoHLF, 2019-01-03:
Hi, currently trying to install Huion H950P in a raspberry pi running Raspbian with no luck. Tried to follow the manual installation instructions, used the deb file auto install, updated kernel headers... this is what I get: 

pi@raspberrypi:~ $ sudo apt-get install raspberrypi-kernel-headers
Reading package lists... Done
Building dependency tree       
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1.20181112-1).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
pi@raspberrypi:~ $ sudo dkms install --force -m digimend/9
Error! echo
Your kernel headers for kernel 4.14.90-v7+ cannot be found at
/lib/modules/4.14.90-v7+/build or /lib/modules/4.14.90-v7+/source.
pi@raspberrypi:~ $ 

I am new to Linux, so any help you can provide would be more then welcome! :)
Thank you very much in advance!

spbnick, 2019-01-03:
Never used Raspbian myself, could you ask Raspbian folks on how to make DKMS work?

NunoHLF, 2019-01-03:
hi @spbnick - thanks for your prompt reply. I spoke with the Raspbian team, as per your suggestion, and they were able to help. I was having some issues because I was using rpi-update to update my kernel - I need to run uptdates from apt instead. Now everything is working as it should, and I just need to map the tablet buttons - but I already found the tutorial for that. Thanks for your great job with the drivers!

closed 2019-01-03 by NunoHLF

spbnick, 2019-01-03:
Glad to hear you figured it out :)

vicarma, 2019-05-06:
@NunoHLF Hi, I have a very similar issue with my daughters Huion graphics pad and we have tried also (unsuccessfully) to make the changes to the kernel.  I have very similar response from RaspPi and would really appreciate if you could share the final code that you had that enabled the kernel to be updated.

yannickmodahgouez, 2019-12-16:
hey @NunoHLF could you put a link here to the instruction gave to you by the Raspbian team ? Thanks !