DIGImend/digimend-kernel-drivers #251: Regression: since Linux 4.18rc1 Trust Flex Graphics Tablet incorrectly detected as mouse, so no Pen input device, no pressure detection, etc.

RussianNeuroMancer, 2019-01-11:
Hello!

I got recommendation to contact DIGIMend project in [this bugreport](https://gitlab.freedesktop.org/libinput/libinput/issues/64#note_100216). Actual regression that I want to report is described in [upstream bugtracker](https://bugzilla.kernel.org/show_bug.cgi?id=202195).

JacekJagosz, 2019-05-01:
It is still a problem for me on kernel 5.1, without or even with Digimend drivers installed.
I know it probably needs to be fixed in the kernel, but how could I report it?
output of `xsetwacom list`:
`UGTABLET TABLET WP5540 stylus   	id: 15	type: STYLUS    
UGTABLET TABLET WP5540 Mouse touch	id: 16	type: TOUCH `

RussianNeuroMancer, 2019-07-16:
It's already reported as you can see here: https://bugzilla.kernel.org/show_bug.cgi?id=202195

JacekJagosz, 2019-07-16:
I am following that, but still nothing has been done for months. Nobody took any action.
I really beg any of you who is capable to try and fix it.
Otherwise I will need to buy a new tablet, even though I really liked this one.

RussianNeuroMancer, 2019-07-16:
> I really beg any of you who is capable to try and fix it.

I tried to contact @spbnick via e-mail but he didn't answer. Maybe my message got into spam folder, I don't know.

You could try to contact Benjamin via e-mail. Issue was introduced after his commit and I just don't know anyone else, besides @spbnick, who is capable of resolving this in upstream.

spbnick, 2019-07-17:
Hi @JacekJagosz, @RussianNeuroMancer! I got @RussianNeuroMancer e-mail and took a look at the issue, but I simply didn't have time nor energy to tackle it yet. I'm [taking a break from DIGImend](https://www.patreon.com/posts/taking-pause-28216343) at the moment, so it will still take a while for me to get to it. Sorry about that.

JacekJagosz, 2019-07-17:
@spbnick As long as someday somebody fixes it I (and probably everybody interested) am fine with it.
I was just worried that a bug introduced in 4.18 and not even fixed in 5.2 isn't popular enough to ever get fixed.
If in a few months you were able to fix it, I would be really happy!

DanielJoyce, 2019-10-29:
Oh poop I'm seeing the same problem. My tablet is now seen as a touch mouse ...

RussianNeuroMancer, 2020-01-28:
@spbnick hi, I wonder if you have time currently to look into this issue?

RussianNeuroMancer, 2020-03-03:
Unfortunately, with Linux 5.5/5.6 (maybe earlier, didn't check) this tablet regressed further: https://bugzilla.kernel.org/show_bug.cgi?id=202195#c8 (Comment 8).

spbnick, 2020-03-03:
I don't have time right now, but I put it into the queue for the next release, and should get to it in a few weeks.

spbnick, 2020-06-06:
Looking at the bug report I see that the tablet in question was never supported by the DIGImend drivers and was handled by the generic HID driver. The tablet reuses the name of [the tablet](https://digimend.github.io/tablets/Waltop_Q_Pad/) with completely different hardware, although the manufacturer assigned it a different item number (21259 vs 16937).

The change in the kernel causing the tablet to break is unfortunate, but I believe @bentiss was heading in the right direction there and we probably shouldn't be reverting it. Instead, we should have the tablet supported properly. I made a brief stab at that in #412. Please test and report results there.

RussianNeuroMancer, 2020-06-09:
Please check if device is detected properly:

```
[ 6910.241693] usb 1-1.1.3: new full-speed USB device number 15 using xhci_hcd
[ 6910.376697] usb 1-1.1.3: New USB device found, idVendor=2179, idProduct=0004, bcdDevice= 0.00
[ 6910.376722] usb 1-1.1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 6910.376732] usb 1-1.1.3: Product: TABLET WP5540
[ 6910.376741] usb 1-1.1.3: Manufacturer: UGTABLET
[ 6910.396172] input: UGTABLET TABLET WP5540 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/0003:2179:0004.0011/input/input51
[ 6910.454522] input: UGTABLET TABLET WP5540 Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/0003:2179:0004.0011/input/input52
[ 6910.455985] hid-generic 0003:2179:0004.0011: input,hiddev5,hidraw9: USB HID v1.00 Mouse [UGTABLET TABLET WP5540] on usb-0000:00:14.0-1.1.3/input0
[ 6910.637814] hid_uclogic: loading out-of-tree module taints kernel.
[ 6910.637964] hid_uclogic: module verification failed: signature and/or required key missing - tainting kernel
[ 6910.819595] input: UGTABLET TABLET WP5540 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/0003:2179:0004.0011/input/input53
[ 6910.878134] input: UGTABLET TABLET WP5540 Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/0003:2179:0004.0011/input/input54
[ 6910.879679] input: UGTABLET TABLET WP5540 Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/0003:2179:0004.0011/input/input55
[ 6910.880233] uclogic 0003:2179:0004.0011: input,hiddev5,hidraw9: USB HID v1.00 Mouse [UGTABLET TABLET WP5540] on usb-0000:00:14.0-1.1.3/input0
```
It's seems like uclogic driver is loaded, but "libinput debug-events" doesn't report pressure (and there is no pressure in Krita too).

spbnick, 2020-06-09:
Alright, a guess didn't work. Could you please post the output of `uclogic-probe` for your tablet, plus the `usbhid-dump -es` output for your tablet when you do a stroke with the pen? See details in https://digimend.github.io/support/howto/trbl/diagnostics/

spbnick, 2020-06-09:
Also, please comment in the PR instead.