DIGImend/digimend-kernel-drivers #253: Xp-pen g640s

dovin0, 2019-01-14:
The list of supported tablets shows the xp-pen g640 model but not the g640s. As far as I know, they differ only in appearance, is this model supported?

OffifialMITX, 2019-10-31:
Found this asking myself the same question, chances are that the g640s doesn't even need digimend drivers, since, according to [their own website](https://www.xp-pen.de/product/284.html), it should be supported by default with Android starting with version 6.

Franzcoz, 2020-07-26:
Hi, apparently according to Digimend site this tablet is not supported, and it differs from G640 in the buttons pad. I own this tablet and it is recognized as 3 input devices by libinput driver, it works well, only thing is that it's rotated in 90 degrees, but with a .conf file that has a rotation option "CW" it's no longer rotated.
The other thing is that it doesn't work with wacom drivers, only partially using .conf files (pad doesn't get recognized). so if there is interest in making it work so people that own this tablet can configure it via a GUI I could help in all I can.