DIGImend/digimend-kernel-drivers #254: Huion 1409 - can't get it to work

yodermk, 2019-01-19:
Hi, thanks for all this!  I have a 1409 that worked to some extent with these drivers when I first got it a year and a half ago, though I never had any sort of Wacom support and buttons never worked. Then it stopped working. I noticed it became supported again so I got excited.

Installed the newest driver (I'm on Arch Linux) via DKMS. It does seem to be detected, and for the first time ever, "xsetwacom --list" *does* list the pen and the stylus. But nothing actually works.

In /usr/share/X11/xorg.conf.d, I have the stock 10-evdev.conf and 40-libinput.conf, which have the tablet sections, and added the xorg.conf from the digimend repo as 50-huion.conf.  I also copied the udev rules from the repo into /etc/udev/rules.d/90-hid.rules

Am I doing anything clearly wrong?

Thanks much!

binarydepth, 2020-02-20:
I can't get it to work on Fedora 31 either
**/usr/share/X11/xorg.conf.d**
has **10-evdev.conf  10-quirks.conf  10-radeon.conf  40-libinput.conf  50-digimend.conf 70-wacom.conf  71-libinput-overrides-wacom.conf**

```
dkms status
digimend, 9, 5.3.7-301.fc31.x86_64, x86_64: installed
```

I'm not sure if the digimend driver is being loaded instead of wacom.

binarydepth, 2020-02-20:
It works with the cable.