DIGImend/digimend-kernel-drivers #256: Reading and injecting pressure input?

teleportingtortoise, 2019-01-28:
Sorry if this is the wrong place to ask about this, but I was wondering if there is any way to both read and inject pressure input from the command line to a specified pen tablet device. For some time I've been hoping to convert other input sources into pressure input so I can do things like simulate foot pedal controlled airbrushes and the like, but haven't had much luck even figuring out where to start. I've tried sniffing around in dev input but, while I can see cursor events, I can't seem to find pressure events (or simply can't tell them apart in the hex output). No worries if this is beyond the scope of this issue tracker and thank you for your time and for the great work on these drivers.

spbnick, 2019-01-30:
This is somewhat beyond the scope, yes, but what you're trying to do is interesting, so I'd like to try and help you. First, let me ask you a question to understand better what you're trying to do.
Which actual, physical input devices do you have, and how do you want them to appear to the applications, exactly?

spbnick, 2019-02-08:
I think your best bet would be using something like https://python-evdev.readthedocs.io/en/latest/ - you can benefit from the kernel drivers for your tablet, implement a custom driver for your pedal and speak to the kernel in a relatively high-level protocol.

My plan would be to write a Python program which grabs exclusive access to the tablet device (so that nothing tries to interpret it by itself), interfaces with the pedal (or whatever else you have) somehow (I don't know how it would work), then creates a uinput device emulating a tablet (it can look exactly like your tablet's device would already look), and then reads both tablet and pedal input, mixes them together, and produces the uinput device's input.

Does this make sense?

teleportingtortoise, 2019-02-17:
Yeah I think I follow. I've definitely got some learning ahead of me since I'm new to coding and don't know much about how evdev actually works under the hood, but having a direction and an idea of how to approach it helps a lot, thank you!

tobiasBora, 2019-03-26:
Hello,
I just found this thread, and it seems close to what I just wrote. You can try to get inspired by this script [here](https://unix.stackexchange.com/questions/508612/simulate-graphics-tablet-using-libevent) that simulates a graphics tablet from scratch, and this script [here](https://unix.stackexchange.com/questions/507687/graphic-tablet-veikk-pressure-sensitivity-on-linux) that takes input from an existing device (a mouse-like device) and converts it into a tablet. Hope it can help!

teleportingtortoise, 2019-06-21:
@tobiasBora Thank you! These with the documentation really helps to give me a head start.