DIGImend/digimend-kernel-drivers #257: xsetwacom on Huion Inspiroy H640P issues

sprite-1, 2019-01-30:
By default `xsetwacom` doesn't detect my tablet's side buttons but pressure sensitivity works with the pen and I can use it, albeit limited, on drawing programs such as Krita. However, I recently discovered a way to get `xsetwacom` to detect it by adding a file in `/usr/share/X11/xorg.conf.d/52-tablet.conf` and inside that file is this:

```
Section "InputClass"
	Identifier "Huion on wacom"
	#MatchIsTablet "on"
	MatchProduct "Tablet Monitor"
	MatchDevicePath "/dev/input/event*"
	Driver "wacom"
	# Apply custom Options below.
EndSection
```

When I set that, finally `xsetwacom --list devices` sees it. A weird side effect of that is when I do `xinput list`, it moves both the pad and the pen under the `Virtual core pointer` category. Before that, one of them was in the `Virtual core keyboard` category.

In any case, I then tried to remap `Button 1` to `key F13` using:
```
xsetwacom --set "Tablet Monitor Pad pad" Button 1 "key F13"
```
but it gave me an error saying "Warning: unable to map 'F13' to a keycode." when according to [this](https://unix.stackexchange.com/a/301970/325974) and [this](http://wiki.linuxquestions.org/wiki/List_of_Keysyms_Recognised_by_Xmodmap), it should support from F1 to F35.

Then I tried to remap it again, this time to `key F11` and it proceeded without errors. However, pressing any buttons on the side of my tablet doesn't seem to trigger F11 at all (fullscreen shortcut for most apps).

So now I'm a bit lost. Where can I go from here?

realmadpuppy, 2019-02-08:
wow, I tried what you did and it kept X from loading, I had to reboot into recovery, drop into the console and delete the lines I added to get it working. granted, My tablet is the H950P but everything I have read so far claims that they are nearly identical except for size ans the extra 2 buttons on the frame.

sprite-1, 2019-02-08:
Whee did you place it? I have it inside `52-tablet.conf`

[image]

realmadpuppy, 2019-02-08:
On my install of Kubuntu 18.10 there is no config file named **"52-tablet.conf"**  what I have is **"70-wacom.conf"** so I slotted what you wrote verbatim at the bottom of the that file making sure **"EndSection"** was the last line. Also, I have this line added to this file with no results:
[image]
Which I think you will notice is the same ID as the H640P (I think the H640P and the H950P are essentially the same tablet sans the size and extra frame buttons) 
I have been using Linux for a long time but I am just a slightly better than average end user, I am probably totally off base here and screwed up the by placing stuff where it didn't belong.
By the way, thanks for replying. It is pretty dead around here and at the iirc chat, but, I didn't realize that the main dev is out of town for attending conventions.

realmadpuppy, 2019-02-08:
I'm sorry I just noticed the text under your screenshot, yes, I didn't have that right, when I **"xinput list"**
this is what I get:
[image]
So, maybe changing the name to **"pad"** would fix the issue. IDK, like I said before I may be in the wrong config file altogether even though I think the naming convention on mine just looks like what (K)Ubuntu uses in its latest iteration of the xwacom drivers.

sprite-1, 2019-02-08:
I didn't have the 52-tablet.conf either, I just made it myself based on some post I found elsewhere that I can't track down now

realmadpuppy, 2019-02-08:
Just one more question? did you have no **.conf** file at all or did you add the new **.conf** file along side another **.conf** file?

sprite-1, 2019-02-08:
For your case,I think you would need to use `HID 256c:006e` as the tablet identifier so it covers both pad and pen

realmadpuppy, 2019-02-08:
So let me get this straight? I'll do it with your code modified:

Section "InputClass"
	Identifier "256c:006e"
	#MatchIsTablet "on"
	MatchProduct "pen"
	MatchDevicePath "/dev/input/event*"
	Driver "wacom"
	# Apply custom Options below.
EndSection

did I get that right??

sprite-1, 2019-02-08:
Don't forget the HID part

```
Section "InputClass"
	Identifier "Huion on wacom"
	#MatchIsTablet "on"
	MatchProduct "HID 256c:006e"
	MatchDevicePath "/dev/input/event*"
	Driver "wacom"
	# Apply custom Options below.
EndSection
```

realmadpuppy, 2019-02-08:
ok, you change the **"MatchProduct"** line not the **"Identifier"** line. I got confused because you used the phrase **"tablet Identifier"** and I focused on the word **Identifier** :P I'll give this a shot tomorrow to see if it works. 
Thank You so much for the help, I may have more questions in the future if you don't mind. 
I'm not sure where you live in the world so I will say have a nice day. :)

realmadpuppy, 2019-02-08:
ALRIGHTY! It didn't work but I processed all your information and came up with a plan and it worked!
I created a file called **52-tablet.conf** Inserted the code you gave me:
[image]
Section "InputClass"
	Identifier "Huion on wacom"
	#MatchIsTablet "on"
	MatchProduct "HID 256c:006e"
	MatchDevicePath "/dev/input/event*"
	Driver "wacom"
	# Apply custom Options below.
EndSection

and when I ran **"xsetwacom --list devices"** I got this:
[image]

I'm done tonight, I want to go to bed happy! I'll test setting up the buttons tomorrow and see if I run into the same problems you are having.

**PS, I didn't understand what the number at the start of the config file meant I thought it was some version number, turns out that it is position that the config file gets read so naming the file _"52-tablet.conf"_ will be read before _"70-wacom.conf"_ on my system.**

sprite-1, 2019-02-08:
I'm glad you got it set up! And yeah, hopefully we figure out how to get these to work because my laptop's Ctrl button is getting hammered with all the shortcut keybindings that I'd rather be on the tablet itself.

sprite-1, 2019-02-08:
I FIGURED IT OUT

I feel so dumb now. There's a small switch on the side of the tablet pad. Switching it the other way instantly made my `xsetwacom` keybinds working!

closed 2019-02-08 by sprite-1