DIGImend/digimend-kernel-drivers #259: Huion H950P frame buttons help

realmadpuppy, 2019-02-07:
Hello I was on the iirc chat and left messages and left the window open for a day and a half without response, But adding to this thread(#235), I have the same problem in Ubuntu 18.10 I got the pad working great with the .8 release of DIGimend (.9 didn't work at all) But, I would love to get full functionality with my H950P and the buttons just don't work, Interestingly, I was in the KDE joystick config and the buttons post in that program.I used the Ulogic probe and Xinpit utilites and the post from top to bottom a 1,2,3,8,9,10,11,12. and if I press them while on the desktop random menu's pop up and windows close and browser tabs change so it is doing something, what I would like to know is do I uninstall the .8 drivers and install and configure the H950P with the wacom utilities? I really don't want to toast what is already working well to get the buttons configured properly but if it will work overall better with the wacom drivers and configuration I would like to give it a shot.
thank you in advance. #235

realmadpuppy, 2019-02-07:
I have additional information, I followed the instructions set out in the "wacom configuring xorg driver" and this is the result:
I am running **Kubuntu 18.10** with the **kernel 4.18.0-14-generic**
The location of the configuration file in my installation is "**/usr/share/X11/xorg.conf.d/70-wacom.conf**"
rather than "**/etc/X11/xorg.conf.d/50-tablet.conf**" as stated in the instructions on the DIGImend site.
the wacom drivers are already installed in my installation of Kubuntu so I skipped to the configuration section, and I made this change:
[image]
Then I restarted X and ran the command "_**xsetwacom list**_" with no results, To verify that, I went and tried it verbose with "_**xsetwacom --list devices**_" with no results. To make sure that MY tablet was still being seen by the OS I ran the "_**xinput list**_" command with this result:
[image]
Now, I am a long time linux user...but I am still just an end user, I really don't know where else to go without further assistance from someone with more knowledge on this subject.

**PS, I tried to seek assistance on the DIGImend iirc, and every post on here directs people to it but, there are people listed as logged in...and all there is is "dead air" no conversations, no replies, nothing, I even left myself logged on from the evening to well into the next day with nothing...no offense, but, it's a dead end to go to the DIGImend iirc. and I understand that it is voluntary support but no one even "stops by" to even see if anyone needs help? when it says "Leave the browser window open if you want a response" does that mean for a day or two, week or month?**

realmadpuppy, 2019-02-08:
Because the Huion H640P and the H950P get configured the same way I moved over to #257 and am working over there to create a workable plan. 
I'm at #257 now.

spbnick, 2019-02-08:
Hi @realmadpuppy, yeah there's mostly just me replying on #DIGImend, whenever I have a moment, and that can vary. If at that moment you're not logged in, I cannot answer. I'm assuming your nick was "madpuppy" and e.g. you're not logged in right now.

Anyway, generally you don't need to touch any files under /usr/share, and you better not. Especially you don't need to touch `/usr/share/X11/xorg.conf.d/70-wacom.conf` file - it belongs wholly to the Wacom driver. Don't put any new files in there - that location is managed by packages and they're free to mess with them. Put any configuration under `/etc/X11/xorg.conf.d/` - it will override whatever is in `/usr/share/X11/xorg.conf.d/` and it won't be touched by any package upgrades/downgrades.

The v9 drivers already install configuration for your tablet setting up the Wacom driver to handle it. See `usr/share/X11/xorg.conf.d/*-digimend.conf`. You should get your tablet in `xsetwacom list` output. If you don't we can debug and fix that. First thing would be looking at `/var/log/Xorg.0.log` and seeing if the Wacom driver tries to pick up your tablet.

realmadpuppy, 2019-02-08:
No worries, but if you go over to #257 you will see that I have got the H950P working with help from sprite-1 and a whole lot of reading! I am actually using the previous version of the DIGImend package ".8" because ".9" didn't work for me. I did read somewhere that (i think it was you) there is some bug in ".9" and to compile the test drivers. I really prefer using the .deb packages because I NEVER have good fortune compiling anything, matter of fact for the past 21 years of using Linux I have never successfully compiled anything.

realmadpuppy, 2019-02-08:
OK, I just re-read what you just said and I "get" what you mean now, I need to move the configuration file I made `"52-tablet.conf"` to `/etc/X11/xorg.conf.d/` because the `/usr/share` location is wholly handled by packages. 
Cool, Ill move that `.conf` file over there for now.

spbnick, 2019-02-08:
Great to hear that you got it working :) I fixed one problem in v9 related to the Wacom configuration, but that problem would result in X.org not starting at all (you not getting any graphical login), so it must be something else. Would be good to figure it out with v9, so that you and anyone else is not stuck with v8.

So, if you have the time and energy, please remove any extra configuration you made (but make sure you know how to get it working again in case we don't get to the bottom of this), try to install v9 (the .deb package would be fine), and let's try to debug and fix your issue.

realmadpuppy, 2019-02-08:
no problem, I'm going to go to bed now, it is 2:30 AM and I'll start on it this afternoon.

Thanks! :+1:

realmadpuppy, 2019-02-08:
I think I see what may be the problem, there is no folder named `xorg.conf.d` in `etc/X11/`

maybe that is why `.9` of the DIGImend package is not working on my system, unless it creates it when it installs?

spbnick, 2019-02-08:
> I think I see what may be the problem, there is no folder named `xorg.conf.d` in `etc/X11/`
No, that shouldn't be a problem. One of the main v9 features was installation of a default configuration into `/usr/share/X11/xorg.conf.d`, which should set your tablet up with the Wacom driver, so you don't have to do anything in `/etc/X11/xorg.conf.d`.

spbnick, 2019-02-08:
I.e. when installing v9 you shouldn't need to edit any X.org configuration files at all to make your pad buttons work.

realmadpuppy, 2019-02-08:
by the way, Enjoy the coffee! :+1:

realmadpuppy, 2019-02-08:
I went and did some research but, I tend to be quite obtuse when it comes to Linux command conventions, I would like to make buttons 1 and 2 be undo and redo in Krita but seeing that they are combination button presses I don't think `xsetwacom` supports them, you know like **undo** `Ctrl+Z` and button 2  **redo** `Crtl+Shift+Z` I did read that you can make scripts and assign a key to them but, I wouldn't know how that works because whay keys do you assign to execute a script that will not interfere with everyday use of Linux, the "F" keys? 
to be honest the last time I ever scripted anything was DoS 6 batch files I guess it wouldn't be that hard to create a simple script though?

realmadpuppy, 2019-02-08:
oh! and here is the Log you requested for my system:
[Xorg.0.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2846751/Xorg.0.log)

realmadpuppy, 2019-02-08:
well, this is great, I just had a kernel update and nothing works now, I even completely uninstalled V9 and reinstalled it with no results. Here is the latest Xorg.0.log
[Xorg.0.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2847157/Xorg.0.log)
I went through it and didn't see any reference to my pad or even wacom.

realmadpuppy, 2019-02-09:
here is what returned when I ran the command `lsusb -v -d `256c:006e`

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2847235/descriptors.txt)
after that I did the `uclogic-probe 3 4 | tee probe.txt" and here is the dump from that:
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2847258/probe.txt)
now the dump for the first pen test"
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2847265/pen_coords.txt)
now tilt:
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2847270/pen_tilt.txt)

now pressure:
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2847271/pen_pressure.txt)
now pen buttons:
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2847273/pen_buttons.txt)
now frame buttons:
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2847276/frame_buttons.txt)
IDK if you need this stuff, but I am posting it here so it is verified.

the thing is we know the tablet works, something just broke with the new kernel update from Ubuntu.

realmadpuppy, 2019-02-09:
I went and unplugged the tablet and did a clean uninstall of everything I did removing everything and starting clean, I shut down the computer and turned the back switch on the power supply off for 30 seconds. I reboot the computer and installed the DIGImend package and rebooted, after I rebooted I plugged in the tablet and nothing works. 
here is what comes up when I run `lsusb`
[image]
and heres `xinput`
[image]

The kernel is `4.18.0-15-generic`
this is what I have so far, I really don't understand what is going on, everything was working perfectly!!

realmadpuppy, 2019-02-09:
I went through all these diagnostics to figure out what was going on and what fixed everything was Installing the kernel headers for the new kernel. 
I read that if you run the command `sudo modprobe -r hid-kye hid-uclogic hid-polostar hid-viewsonic`
and you get this error: `modprobe: FATAL: Module hid-polostar not found` that installing the kernal headers and rebooting would fix it...and it fixed it.

Now all that is left is to start all over with trying to get the frame buttons to assign properly.

realmadpuppy, 2019-02-09:
I have to make a suggestion that will probably save a lot of people from wasting time troubleshooting getting the DIGImend package installed properly. In the instructions about installing the DIGImend package maybe it should stress installing the kernel headers before installing the DIGImend package. (even if you are installing a .deb or .rpm file) 
I was going nuts trying to get my tablet working after a kernel update for hours, just 2 commands and a reboot fixed 90% of the issues I have been having with the tablet. and maybe reminding users that the kernel headers need to be installed for every upgrade to the kernel as well. to keep the DIGImend drivers working.

realmadpuppy, 2019-02-09:
Thanks for all the  @spbnick  I was mistaken and figured out how to set up the buttons by referencing this: (for anyone who got the xsetwacom working and needs help assigning your buttons on a Huion tablet, the H640P and H950P specifically)
https://github.com/DIGImend/digimend-kernel-drivers#configuring-wacom-xorg-driver
fortunately for me,in the help file, the xsetwacom command was formatted for my huion tablet and all I had to do was change the button number and the command I wanted to assign to it.
`xsetwacom set "HID 256c:006e Pad pad" button 9 key Ctrl Z`
you just need to change the button number to the button you want to assign and change the key to what key command you want. for example `button 1 key Crtl Space Z` for "redo in krita.

realmadpuppy, 2019-02-09:
everything is working now so I'm closing down this thread.

closed 2019-02-09 by realmadpuppy

spbnick, 2019-02-09:
I'm glad to hear that you figured it all out, @realmadpuppy. That was a lot of research on your part. I'm a little concerned by the experience you had with having the driver working after the kernel upgrade. I would have expected that installing the .deb package would pull the kernel headers in automatically, but apparently it's not happening. I submitted #260 to track and try to fix that.