DIGImend/digimend-kernel-drivers #261: Huion  1060plus framebuttons not working

niedzwiadel23, 2019-02-10:
Hello, Im new to linux mint 19.1 cinnamon. After installing digimend-kernel-drivers
v9 deb my Huion 1060plus Graphic Tablet works just fine except the frame buttons and the softkeys. Please Help.

spbnick, 2019-02-10:
Hi @niedzwiadel23, I've heard a few reports like yours and would like to get to the bottom of this. How was it working before, and how did you set it up?

niedzwiadel23, 2019-02-10:
first i installed the digimend-kernel-drivers v8.deb and two of the framebuttons from the top worked, then the v9.deb make it worse and non of the button works actually.

 $ lsusb
Bus 002 Device 002: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 003: ID 5986:0295 Acer, Inc 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 003: ID 0cf3:3004 Atheros Communications, Inc. AR3012 Bluetooth 4.0
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 17ef:6053 Lenovo 
Bus 005 Device 005: ID 058f:6366 Alcor Micro Corp. Multi Flash Reader
Bus 005 Device 004: ID 256c:006e  
Bus 005 Device 003: ID 058f:6254 Alcor Micro Corp. USB Hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ xinput --list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Lenovo Wireless Optical Mouse N100      	id=11	[slave  pointer  (2)]
⎜   ↳ ETPS/2 Elantech Touchpad                	id=15	[slave  pointer  (2)]
⎜   ↳ PenTablet  Pad pad                      	id=16	[slave  pointer  (2)]
⎜   ↳ PenTablet  Pen stylus                   	id=17	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Power Button                            	id=9	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=10	[slave  keyboard (3)]
    ↳ Lenovo EasyCamera: Lenovo EasyC         	id=12	[slave  keyboard (3)]
    ↳ Ideapad extra buttons                   	id=13	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=14	[slave  keyboard (3)]

~$ xsetwacom --list
PenTablet  Pad pad              	id: 16	type: PAD       
PenTablet  Pen stylus           	id: 17	type: STYLUS 

i also tried to make the 52-tablet.conf file:
sudo mkdir /etc/X11/xorg.conf.d 
cd /etc/X11/xorg.conf.d 
sudo xed 52-tablet.conf

like in #257 but with no success. i was also searching for a small switch on my tablet but unfortunately there is none ;)

All of the 12 Buttons are working but cant map/use them.:

$ sudo usbhid-dump -es -m 256c:006e
Starting dumping interrupt transfer stream
with 1 minute timeout.

005:004:000:STREAM             1549789768.577610
 07 E0 01 01 01 00 00 00

005:004:000:STREAM             1549789768.739517
 07 E0 01 01 00 00 00 00

005:004:000:STREAM             1549789770.195573
 07 E0 01 01 02 00 00 00

005:004:000:STREAM             1549789770.323621
 07 E0 01 01 00 00 00 00

005:004:000:STREAM             1549789770.939672
 07 E0 01 01 04 00 00 00

005:004:000:STREAM             1549789771.085679
 07 E0 01 01 00 00 00 00

005:004:000:STREAM             1549789771.541570
 07 E0 01 01 08 00 00 00

005:004:000:STREAM             1549789771.653692
 07 E0 01 01 00 00 00 00

005:004:000:STREAM             1549789772.191711
 07 E0 01 01 10 00 00 00

005:004:000:STREAM             1549789772.339715
 07 E0 01 01 00 00 00 00

005:004:000:STREAM             1549789773.057621
 07 E0 01 01 20 00 00 00

005:004:000:STREAM             1549789773.177766
 07 E0 01 01 00 00 00 00

005:004:000:STREAM             1549789774.973829
 07 E0 01 01 40 00 00 00

005:004:000:STREAM             1549789775.101780
 07 E0 01 01 00 00 00 00

005:004:000:STREAM             1549789775.459803
 07 E0 01 01 80 00 00 00

005:004:000:STREAM             1549789775.563796
 07 E0 01 01 00 00 00 00

005:004:000:STREAM             1549789776.143795
 07 E0 01 01 00 01 00 00

005:004:000:STREAM             1549789776.277823
 07 E0 01 01 00 00 00 00

005:004:000:STREAM             1549789776.625834
 07 E0 01 01 00 02 00 00

005:004:000:STREAM             1549789776.755834
 07 E0 01 01 00 00 00 00

005:004:000:STREAM             1549789777.253850
 07 E0 01 01 00 04 00 00

005:004:000:STREAM             1549789777.391852
 07 E0 01 01 00 00 00 00

005:004:000:STREAM             1549789777.779866
 07 E0 01 01 00 08 00 00

005:004:000:STREAM             1549789777.925870
 07 E0 01 01 00 00 00 00

spbnick, 2019-02-10:
I see, thank you. Before we move further, could you please describe how *exactly* the buttons worked with v8, did you do any remapping and how?

With v9 you shouldn't have to create any X.org configuration files. I.e. you shouldn't need `/etc/X11/xorg.conf.d/52-tablet.conf`.

niedzwiadel23, 2019-02-10:
With v8 the first buttons worked in Krita i think "move back" or something like that, i dont remapping them. In v8 there was a config file so i made one when moving to v9 because it disappered

spbnick, 2019-02-10:
So you didn't set up remapping, OK.

Now, could you run `xinput test "PenTablet Pad pad"`, press each button on the frame one-by-one, and post the output here?

niedzwiadel23, 2019-02-10:
~$ xinput test "PenTablet Pad pad"
unable to find device 'PenTablet Pad pad'

spbnick, 2019-02-10:
OK, I must have fumbled the device name, should've left it to you to figure out how to run `xinput test` for your device. In general just look at the output of `xinput list` find the pad device id and pass it to `xinput test`. So for the `xinput list` output you posted above the command would be `xinput test 16`.

niedzwiadel23, 2019-02-10:
xinput test 16 dont work
 
xinput list 16
PenTablet  Pad pad                      	id=16	[slave  pointer  (2)]
	Reporting 8 classes:
		Class originated from: 16. Type: XIButtonClass
		Buttons supported: 16
		Button labels: None None None None None None None None None None None None None None None None
		Button state:
		Class originated from: 16. Type: XIKeyClass
		Keycodes supported: 248
		Class originated from: 16. Type: XIValuatorClass
		Detail for Valuator 0:
		  Label: Abs X
		  Range: 0.000000 - 0.000000
		  Resolution: 0 units/m
		  Mode: absolute
		  Current value: 0.000000
		Class originated from: 16. Type: XIValuatorClass
		Detail for Valuator 1:
		  Label: Abs Y
		  Range: 0.000000 - 0.000000
		  Resolution: 0 units/m
		  Mode: absolute
		  Current value: 0.000000
		Class originated from: 16. Type: XIValuatorClass
		Detail for Valuator 2:
		  Label: None
		  Range: 0.000000 - 1.000000
		  Resolution: 1 units/m
		  Mode: absolute
		  Current value: 0.000000
		Class originated from: 16. Type: XIValuatorClass
		Detail for Valuator 3:
		  Label: None
		  Range: 0.000000 - 1.000000
		  Resolution: 1 units/m
		  Mode: absolute
		  Current value: 0.000000
		Class originated from: 16. Type: XIValuatorClass
		Detail for Valuator 4:
		  Label: None
		  Range: 0.000000 - 1.000000
		  Resolution: 1 units/m
		  Mode: absolute
		  Current value: 0.000000
		Class originated from: 16. Type: XIValuatorClass
		Detail for Valuator 5:
		  Label: None
		  Range: 0.000000 - 1.000000
		  Resolution: 1 units/m
		  Mode: absolute
		  Current value: 0.000000

spbnick, 2019-02-10:
What do you mean by "xinput test 16 dont work"? Command starts, but doesn't react to button presses?

niedzwiadel23, 2019-02-10:
exactly, it doesnt react to buttons presses, no error message, nothing. Maybe i try to install v8 back again? or should i try something else?

spbnick, 2019-02-10:
Sure, you could install v8. However, I'm pretty sure we can figure out what's going on, and I'd like to have v9 and newer versions work too, so you could use them.

Could we go one level deeper now? Could you execute `sudo evtest`, select the pad device, press all the buttons in order, and post the output here?

niedzwiadel23, 2019-02-10:
$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Power Button
/dev/input/event1:	Sleep Button
/dev/input/event2:	Lid Switch
/dev/input/event3:	Power Button
/dev/input/event4:	AT Translated Set 2 keyboard
/dev/input/event5:	Video Bus
/dev/input/event6:	Video Bus
/dev/input/event7:	Lenovo Wireless Optical Mouse N100
/dev/input/event8:	ETPS/2 Elantech Touchpad
/dev/input/event9:	Ideapad extra buttons
/dev/input/event10:	HD-Audio Generic HDMI/DP,pcm=3
/dev/input/event11:	HD-Audio Generic Mic
/dev/input/event12:	HD-Audio Generic Headphone
/dev/input/event13:	Lenovo EasyCamera: Lenovo EasyC
/dev/input/event14:	PenTablet  Keyboard
/dev/input/event15:	PenTablet  Consumer Control
/dev/input/event16:	PenTablet  System Control
/dev/input/event17:	PenTablet  Mouse
/dev/input/event18:	PenTablet  Pen
/dev/input/event19:	PenTablet  Pad
Select the device event number [0-19]: 19
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6e version 0x111
Input device name: "PenTablet  Pad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 256 (BTN_0)
    Event code 257 (BTN_1)
    Event code 258 (BTN_2)
    Event code 259 (BTN_3)
    Event code 260 (BTN_4)
    Event code 261 (BTN_5)
    Event code 262 (BTN_6)
    Event code 263 (BTN_7)
    Event code 264 (BTN_8)
    Event code 265 (BTN_9)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 331 (BTN_STYLUS)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max        1
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max        1
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
Event: time 1549799079.024702, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1549799079.024702, type 1 (EV_KEY), code 256 (BTN_0), value 1
Event: time 1549799079.024702, -------------- SYN_REPORT ------------
Event: time 1549799079.282679, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1549799079.282679, type 1 (EV_KEY), code 256 (BTN_0), value 0
Event: time 1549799079.282679, -------------- SYN_REPORT ------------
Event: time 1549799079.932524, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1549799079.932524, type 1 (EV_KEY), code 257 (BTN_1), value 1
Event: time 1549799079.932524, -------------- SYN_REPORT ------------
Event: time 1549799080.148497, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1549799080.148497, type 1 (EV_KEY), code 257 (BTN_1), value 0
Event: time 1549799080.148497, -------------- SYN_REPORT ------------
Event: time 1549799081.064526, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90003
Event: time 1549799081.064526, type 1 (EV_KEY), code 258 (BTN_2), value 1
[... 33 lines cut ...]
Event: time 1549799084.658154, -------------- SYN_REPORT ------------
Event: time 1549799085.356153, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90009
Event: time 1549799085.356153, type 1 (EV_KEY), code 264 (BTN_8), value 1
Event: time 1549799085.356153, -------------- SYN_REPORT ------------
Event: time 1549799085.510133, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90009
Event: time 1549799085.510133, type 1 (EV_KEY), code 264 (BTN_8), value 0
Event: time 1549799085.510133, -------------- SYN_REPORT ------------
Event: time 1549799085.906098, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000a
Event: time 1549799085.906098, type 1 (EV_KEY), code 265 (BTN_9), value 1
Event: time 1549799085.906098, -------------- SYN_REPORT ------------
Event: time 1549799086.036086, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000a
Event: time 1549799086.036086, type 1 (EV_KEY), code 265 (BTN_9), value 0
Event: time 1549799086.036086, -------------- SYN_REPORT ------------
Event: time 1549799086.578046, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1549799086.578046, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 1
Event: time 1549799086.578046, -------------- SYN_REPORT ------------
Event: time 1549799086.712038, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1549799086.712038, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 0
Event: time 1549799086.712038, -------------- SYN_REPORT ------------
Event: time 1549799087.253992, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1549799087.253992, type 1 (EV_KEY), code 305 (BTN_EAST), value 1
Event: time 1549799087.253992, -------------- SYN_REPORT ------------
Event: time 1549799087.383979, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1549799087.383979, type 1 (EV_KEY), code 305 (BTN_EAST), value 0
Event: time 1549799087.383979, -------------- SYN_REPORT ------------

spbnick, 2019-02-10:
Thank you. This looks correct. Now the question is why the Wacom driver is not picking it up. Can you attach your `/var/log/Xorg.0.log` to a comment?

niedzwiadel23, 2019-02-10:
[Xorg.0.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2848819/Xorg.0.log)

niedzwiadel23, 2019-02-10:
Thank You very much for your Help and time @spbnick 
Finally i found a Solution remapping the Framebuttons.
i started from upper left to bottom right, just replace the a,b,c with a Shortcut you like

xsetwacom --set '16' Button 1 "key a"
xsetwacom --set '16' Button 2 "key b"
xsetwacom --set '16' Button 3 "key c"
xsetwacom --set '16' Button 8 "key d"
xsetwacom --set '16' Button 9 "key e"
xsetwacom --set '16' Button 10 "key f"
xsetwacom --set '16' Button 11 "key g"
xsetwacom --set '16' Button 12 "key h"
xsetwacom --set '16' Button 13 "key i"
xsetwacom --set '16' Button 14 "key j"
xsetwacom --set '16' Button 15 "key k"
xsetwacom --set '16' Button 16 "key l"

spbnick, 2019-02-10:
I see good to hear that you figured it out. However, I'm still confused why `xinput test` didn't output anything for your device. Perhaps something has changed the Wacom driver and now it doesn't produce anything before you map the buttons. Will investigate. Thanks for following along!

Nina265, 2019-04-04:
Hi, 
it's the same issue as here : [#76](https://github.com/DIGImend/digimend-kernel-drivers/issues/76)