DIGImend/digimend-kernel-drivers #263: question about 15.6 devices?

paint-rulz, 2019-02-15:
Hi :-)
May I ask which 15.6 graphics display is best supported, or likely to be first/how that could happen? Would a poll on which device most people want supported be good, and then all who voted for the top device can donate?

I can see there is the HUION HD Graphics 15.6 inch Monitor gt-156hd V2 (cheaper than the Kamvas version, but seems same specs), the Kamvas version, and XP-PEN Artist15.6 Pro particularly; also the Ugee HK1560 (bit difficult to purchase though). 
Also Parblo Coast, Artisul, Veikk and Gaomon (not looking to purchase those tbh).

paint-rulz, 2019-02-15:
Sorry, I forgot to include above, the device I can see that has buttons (I assume for brightness/contrast etc) is the XP-Pen Artist15.6 Pro, so I'd vote for that.

paint-rulz, 2019-02-19:
It was for health reasons I was asking about smaller devices. I'd been happy to make a large donation at the end of last year, but I can see the question has been passed over, so will sort things out another way.

spbnick, 2019-02-19:
Hi @paint-rulz, I would love to work on supporting all the tablets in the world, but with the amount of financing that I get at the moment, I can barely afford working just the occasional hour or two on weekends. Right now I'm focusing on [getting the latest drivers merged to the upstream kernel](https://www.patreon.com/posts/24618548) and researching implementing tests, to keep the driver from breaking. Hopefully soon I'll be able to switch back to adding support for more tablets and features (such as tilt reporting for Huion tablets). I'll probably collect a list of candidate models/features and put it up for voting for patrons on Patreon.

paint-rulz, 2019-02-19:
Hi @spbnick and thank you very much for responding. :-) I appreciate that you do this work to safeguard the drivers (Ugee 2150 has always worked perfectly). That sounds great about collecting a list and that people can vote ... we'd spoken on IRC at the end of last year, and I'd asked how I could do a big one-off donation; thank you that you gave me the necessary bank details. (Regular payments are difficult for various reasons.) Would it be possible to do the voting here (also)?

spbnick, 2019-02-19:
I'm not sure how to organize voting here, but I can try. I would have to give preference to people who are donating, of course, but they don't have to use Patreon. I have now brought up a [Liberapay](https://liberapay.com/spbnick/) and a [Buymeacoffee](https://www.buymeacoffee.com/) page, for people who don't want to use Patreon. The latter might be more suitable for you as it supports one-off donations.

paint-rulz, 2019-02-19:
That would be great if voting can be done here somehow; fingers crossed. Those other options sound very good; thanks it's possible to use those. :-)

spbnick, 2019-03-02:
@paint-rulz, OK, regarding voting, what I came up with so far is to sort issues by number of thumbs-up :thumbsup: reactions and pick one with the most first. I added a note about this under [Support](https://github.com/DIGImend/digimend-kernel-drivers#support) in the README.md. I will hold separate voting for patrons on Patreon and that will take priority.

paint-rulz, 2019-03-03:
Hi @spbnick and thank you very much for your message. Sounds great, and really appreciate. :-)