DIGImend/digimend-kernel-drivers #265: Artisul Driver D20181017

abrasivetroop, 2019-02-17:
i am using windows 10 right now. my specs are: 4gb ram, 2gb intel graphics card, intel i5. when i update windows it really became too slow that i cant work and sometimes it forcely close drawing software that i am using (i am using krita). i want to work freely in a free os and the only way to do this is using linux. 

i really wonder if there is a way to install this drivers to linux. i am using windows right now and it is too slow that i cant work. i am using an artisul tablet. i want to switch to linux but if drivers doesnt work i wont be able to draw. can somebody help please? i really need your help. thanks.

spbnick, 2019-02-18:
Hi @abrasivetroop! First of all, no, the Windows driver won't work with Linux. And yes, your PC is way too low-spec for modern windows. However, let's see what kind of tablet you have, maybe it won't be that hard to make it work. First of all, try to see if your tablet already works with Linux. If it doesn't, post the output of `sudo lsusb -v` for your tablet, and then the output of `sudo usbhid-dump -ed` for your tablet, for the start.

abrasivetroop, 2019-02-18:
Thank you for your reply, right now i have windows installed on my computer. i am really convinced to install linux. i think firstly i need to install linux and after that i will try that if my tablet works or not. after testing my tablet i will reply. thank you again for your help. i really appreciate what you doing!

abrasivetroop, 2019-02-19:
Well, i tried it. It just works like as a mouse no pen pressure or anything. But awkwardly the hotkeys works fine. So there are outputs for my tablet.

`Bus 002 Device 007: ID 5543:0054 UC-Logic Technology Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x5543 UC-Logic Technology Corp.
  idProduct          0x0054 
  bcdDevice            0.00
  iManufacturer           1 UC200_0816_01
  iProduct                2 AP604
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     212
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      98
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      92
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
Device Status:     0x0000
  (Bus Powered)
`
`002:007:002:DESCRIPTOR         1550568057.520143
 05 01 09 06 A1 01 85 03 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 05 07 19 00 29 FF 26 FF
 00 75 08 95 06 81 00 C0 05 0C 09 01 A1 01 85 04
 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81 00
 C0 05 01 09 80 A1 01 85 05 19 81 29 83 15 00 25
 01 75 01 95 03 81 02 95 05 81 01 C0

002:007:001:DESCRIPTOR         1550568057.520731
 05 01 09 02 A1 01 09 01 A1 00 85 01 05 09 19 01
 29 05 95 05 75 01 15 00 25 01 81 02 95 03 81 01
 05 01 09 30 09 31 95 02 75 10 16 00 80 26 FF 7F
 81 06 09 38 15 81 25 7F 95 01 75 08 81 06 05 0C
 0A 38 02 95 01 75 08 81 06 C0 C0 06 01 FF 09 00
 A1 01 85 08 15 00 26 FF 00 09 00 75 08 95 05 B1
 02 C0

002:007:000:DESCRIPTOR         1550568057.521799
 05 0D 09 02 A1 01 85 07 09 20 A1 00 09 42 09 44
 09 45 09 3C 09 43 09 44 15 00 25 01 75 01 95 06
 81 02 09 32 75 01 95 01 81 02 81 03 05 01 09 30
 75 10 95 01 A4 55 0D 65 13 35 00 46 73 18 26 1C
 3E 81 42 09 31 46 45 0F 26 CA 26 81 42 B4 05 0D
 09 30 26 FF 07 46 FF 07 55 00 66 11 E1 75 10 81
 42 09 3D 15 81 25 7F 75 08 95 01 81 02 09 3E 15
 81 25 7F 81 02 C0 C0 05 01 09 02 A1 01 85 09 09
 01 A1 00 05 09 19 01 29 03 15 00 25 01 95 03 75
 01 81 02 95 05 81 01 05 01 09 30 09 31 95 02 75
 10 35 00 46 40 1F 26 40 1F 81 02 05 0D 09 30 26
 FF 07 46 FF 07 95 01 75 10 81 02 C0 C0 06 03 FF
 09 00 A1 01 85 0B 15 00 26 FF 00 09 00 75 08 95
 3F 81 02 C0`

spbnick, 2019-02-19:
Thank you. Could you run [uclogic-probe](https://github.com/DIGImend/uclogic-tools) on your tablet as well?

abrasivetroop, 2019-02-19:
here is the output

M 55 00 43 00 32 00 30 00 30 00 5F 00 30 00 38 00 31 00 36 00 5F 00 30 00 31 00
P 41 00 50 00 36 00 30 00 34 00
S 64 0C 03 1C 3E CA 26 03 00 FF 07 EC 09
S 65 04 03 00 67
S 6E 04 03 30 00
S 79 10 03 41 72 74 69 73 75 6C 20 70 65 6E 63 69 6C
S 7A 08 03 02 04 00 14 00 00
S 7B 0C 03 48 00 4B 00 20 00 4F 00 6E 00

abrasivetroop, 2019-02-19:
Wait a min! It worked. Pen pressure works fine now. Man you are my hero!!! But hotkeys doesnt work. Actually it doesnt matter because i dont use hotkeys. Man thank you again! I really thank you!!

spbnick, 2019-02-19:
@abrasivetroop, that might be because you ran uclogic-probe on your tablet. You will probably need to run it every time to make it work.

abrasivetroop, 2019-02-19:
It works fine with Krita right now. I hope it will work fine in further. If there will be a problem i will notice you.

abrasivetroop, 2019-02-19:
I restarted my laptop and as you said it didnt worked. I ran uclogic-probe and it worked fine again. Is there permanent way to solve this?

spbnick, 2019-02-19:
Yeah, changing the driver. To help that, you can post the output of `sudo usbhid-dump -es -m 5543:0054` where you do a stroke with the pen. If the tablet has any physical buttons on the frame, then do another one, where you press each button in order left-to-right, top-to-bottom.

spbnick, 2019-02-19:
Oh, and don't forget to run `uclogic-probe` before running those `usbhid-dump` commands.

abrasivetroop, 2019-02-19:
here is the outputs.

output of stroke:

```
002:005:000:STREAM             1550589462.302708
 07 C0 65 05 50 1C 00 00 00 00

002:005:000:STREAM             1550589462.306861
 07 C0 69 05 5C 1C 00 00 00 00

002:005:000:STREAM             1550589462.314732
 07 C0 4F 05 52 1C 00 00 00 00

002:005:000:STREAM             1550589462.318739
 07 C0 43 05 49 1C 00 00 00 00

002:005:000:STREAM             1550589462.322736
 07 C0 3D 05 41 1C 00 00 00 00

002:005:000:STREAM             1550589462.330721
 07 C0 31 05 3C 1C 00 00 00 00

002:005:000:STREAM             1550589462.334736
 07 C0 2A 05 39 1C 00 00 00 00

002:005:000:STREAM             1550589462.342787
 07 C0 1C 05 32 1C 00 00 00 00

002:005:000:STREAM             1550589462.346785
 07 C0 0D 05 32 1C 00 00 00 00

002:005:000:STREAM             1550589462.350811
 07 C0 04 05 32 1C 00 00 00 00

002:005:000:STREAM             1550589462.358800
 07 C0 F4 04 32 1C 00 00 00 00

002:005:000:STREAM             1550589462.362725
 07 C0 E2 04 32 1C 00 00 00 00

002:005:000:STREAM             1550589462.366744
 07 C0 D9 04 32 1C 00 00 00 00

002:005:000:STREAM             1550589462.374738
 07 C0 CE 04 32 1C 00 00 00 00

002:005:000:STREAM             1550589462.378699
 07 C0 C5 04 32 1C 00 00 00 00

002:005:000:STREAM             1550589462.386736
 07 C0 C5 04 32 1C 00 00 00 00

002:005:000:STREAM             1550589462.390721
 07 C0 B9 04 32 1C 00 00 00 00

002:005:000:STREAM             1550589462.394705
 07 C0 B9 04 32 1C 00 00 00 00

002:005:000:STREAM             1550589462.402705
 07 C0 B9 04 32 1C 00 00 00 00

002:005:000:STREAM             1550589462.406731
 07 C1 B9 04 32 1C DA 00 00 00

002:005:000:STREAM             1550589462.414700
 07 C1 CA 04 0B 1C BC 01 00 00

002:005:000:STREAM             1550589462.418796
 07 C1 CF 04 00 1C 4E 02 00 00
[... 177 lines cut ...]
002:005:000:STREAM             1550589571.920451
 07 E0 02 01 00 00 00 00 00 00

002:005:000:STREAM             1550589573.892489
 07 E0 02 01 01 00 00 00 00 00

002:005:000:STREAM             1550589574.008491
 07 E0 02 01 00 00 00 00 00 00

002:005:000:STREAM             1550589574.012485
 07 E0 02 01 00 00 00 00 00 00

002:005:000:STREAM             1550589574.744491
 07 E0 02 01 04 00 00 00 00 00

002:005:000:STREAM             1550589574.860492
 07 E0 02 01 00 00 00 00 00 00

002:005:000:STREAM             1550589575.524475
 07 E0 02 01 08 00 00 00 00 00

002:005:000:STREAM             1550589575.644390
 07 E0 02 01 00 00 00 00 00 00

```

spbnick, 2019-02-19:
Thank you, @abrasivetroop! Does this really have an "eraser"? I.e. can you turn the pen around and use the other end? Could you post a link to the tablet description with photos? E.g. on the website you bought it from?

abrasivetroop, 2019-02-19:
Yes it has an reaser. Here the link for the photo: https://www.teknobiyotik.com/media/catalog/product/cache/4/image/c82fc2f55655ae2bd6f823b09b71ac50/u/c/ucap604-disgorsel-02.jpg

It has 4 hotkeys on tablet btw. Oh I forgot to say, also there is 2 buttons on the pen.

spbnick, 2019-02-19:
Thank you. An actual eraser is very unusual. I have little experience with making that work, and will need extra research to do it right. As to the buttons, I can't see them on the photos, where are they?

abrasivetroop, 2019-02-19:
It is really hard to distinguish from photos but the button are located at that metallic white part of the tablet which is in front of tablet also you plug the usb there. There are 4 buttons. Also there is two buttons on the pen you can distinguish it from photos if you look carefully. Thank you.

spbnick, 2019-02-19:
I see, thank you for all the information, @abrasivetroop. This is most, if not all, that is needed to make the driver support your tablet. However that will need time to actually implement and test. I'll see if I have time on the weekend to throw together a test version for you.

abrasivetroop, 2019-02-19:
spbnick, you are my hero! I really thank you! I will work with uclogic-probe code until you solve the problem permanently. I hope you will solve it easily. Thank you again!

abrasivetroop, 2019-04-19:
Hey nick! How is going the process? Have you finished adding support for my tablet?