DIGImend/digimend-kernel-drivers #268: XP-PEN Artist22 PRo

deiflou, 2019-03-12:
I'm trying to get my XP-PEN Artist22 Pro (https://www.xp-pen.com/product/62.html) working in Manjaro.
I attach the tablet diagnostics.
[xp_pen_artist22_pro.tar.gz](https://github.com/DIGImend/digimend-kernel-drivers/files/2957044/xp_pen_artist22_pro.tar.gz)

deiflou, 2019-03-20:
Hi. I added some code to the driver and get the tablet working. You can view the changes here: https://github.com/deiflou/digimend-kernel-drivers

I encountered some problems. The main code that I was changing to test different things is the fixed report descriptor in `hid-uclogic-rdesc.c` and the way the interfaces other than number 1 are handled in te case in `hid-uclogic-params.c`.
At first, every time I changed something, the tablet was recognized differently. For example the first time using xinput showed two pen devices and one touch device. The tablet worked with xsetwacom but not with the wacom gui utility. After some changes it was recognized as a pen and a touch and it also worked with the wacom utility (I could calibrate it, remap buttons and so on).
The only thing that didn't work was the top button of the stylus and after some other changes that I made trying to get it working the tablet stopped being recognized by the wacom utility again. And now no matter what I change it is recognized always as to pens and a touch, and I'm not capable to reproduce the steps that led me to the tablet being recognized by the utility. I think there is some sort of info that is not deleted when I uninstall/reinstall the drivers. Maybe info stored by xsetwacom or the wacom driver.
Any help to solve this is welcomed.

I've read thar the top button is not recognized because it is reported as eraser. Is this correct? Is there any way to bypass that?

One last thing, where is the configuration for xsetwacom stored?

Thanks a lot.

spbnick, 2019-03-20:
Sorry, I don't have time to dig into the specifics of the report descriptor you wrote and your tablet in particular right now, but xsetwacom doesn't normally store anything and doesn't have a configuration file.
Another thing is that this tablet could likely be initialized and handled according to the v2 protocol and you would probably not need to write your own report descriptor. Try running uclogic-probe on it.

deiflou, 2019-03-26:
Hi.
Running `lsusb` I get this line for the tablet (with no name at the end):
`Bus 001 Device 003: ID 28bd:900c`
Then running `sudo uclogic-probe 001 003`  I get:
```
M 58 00 50 00 2D 00 50 00 45 00 4E 00
P 41 00 72 00 74 00 69 00 73 00 74 00 32 00 32 00 20 00 50 00 72 00 6F 00
S 64 0C 03 2F BA B9 68 03 00 FF 1F EC 09
S 65 04 03 00 67
S 6E 04 03 00 30
S 79 0E 03 55 47 45 45 20 4C 41 32 32 30 31 46
S 7A 08 03 02 08 00 14 00 00
S 7B 0C 03 48 00 4B 00 20 00 4F 00 6E 00
Failed to get string descriptor 0xC8: Input/Output Error
```
And if I run lsusb again the device (28bd:900c) doen't appear in the list but a new line with the same vendor ID does as follow:
`Bus 001 Device 004: ID 28bd:1227`

deiflou, 2019-03-26:
Well I tried initializing with `uclogic_params_pen_init_v2` and it seems to work as with the previous code:
```
[...]

	case VID_PID(USB_VENDOR_ID_UGEE,
		     USB_DEVICE_ID_UGEE_XPPEN_TABLET_ARTIST22_PRO):
		/* If this is the pen and frame interface */
		if (bInterfaceNumber == 1) {
			/* Probe v2 pen parameters */
			rc = uclogic_params_pen_init_v2(&p.pen, &found, hdev);
			if (rc != 0) {
				hid_err(hdev, "pen probing failed: %d\n", rc);
				goto cleanup;
			}
		} else {
			/* TODO: Consider marking the interface invalid */
			uclogic_params_init_with_pen_unused(&p);
		}
		break;
[...]
```
Full code: https://github.com/deiflou/digimend-kernel-drivers

BloodyMess, 2021-01-19:
Do your changes get the Artist 22 Pro working reliably?