DIGImend/digimend-kernel-drivers #270: Take in charge Veikk A30

tobiasBora, 2019-03-21:
Hello,

I just bought the cheap and great tablet Veikk A30, but unfortunately Linux does not recognize the pressure. I tried your drivers but it seems not take in charge the Veikk model.

I tried to do some work [here](https://unix.stackexchange.com/questions/507687/graphic-tablet-veikk-pressure-sensitivity-on-linux) and I managed to create a python script that get all the relevant informations (position, click on the 2 buttons + normal click, pressure) directly from the file `/dev/input/by-id/usb-VEIKK.INC_A30_0000001-event-mouse`. You can find below my python script if it can help.

Do you think it would be possible to include this tablet in your drivers?

Thanks!

```
#!/usr/bin/env python3
import struct
PRINT_TIMESTAMP = True
# Open the file in the read-binary mode
f = open("/dev/input/by-id/usb-VEIKK.INC_A30_0000001-event-mouse", "rb" )

while 1:
    data = f.read(24)
    # print struct.unpack('4IHHI',data)
    ###### FORMAT = ( Time Stamp_INT , 0 , Time Stamp_DEC , 0 , 
    ######   type , code ( key pressed ) , value (press/release) )
    time_int, _, time_dec, _, ev_type, ev_code, ev_val = struct.unpack('4IHHI',data)
    t = (ev_type, ev_code)
    if ((t == (0,0) and ev_val == 0)
        or (t == (4, 4) and ev_val >= 589825 and ev_val <= 589827)):
      # Redundant as it's for normal/bottom/top clicks
      # (same code for press/release), or just garbage 0,0,0
        continue
    if PRINT_TIMESTAMP:
        print("[{:.2f}] ".format(time_int + time_dec/1e6),
              end="", flush=True)
    if t == (3,0):
        print("Pos x: {} ({:.2f}%)".format(ev_val, 100*ev_val/32767), flush=True)
    elif t == (3,1):
        print("Pos y: {} ({:.2f}%)".format(ev_val, 100*ev_val/32767), flush=True)
    elif t == (3,24):
        print("Pression: {} ({:.2f}%)".format(ev_val, 100*ev_val/8191), flush=True)
    elif t == (1,272):
        print("Normal click ({})".format("press" if ev_val else "release"), flush=True)
    elif t == (1,273):
        print("click button 2 (bottom) ({})".format("press" if ev_val else "release"), flush=True)
    elif t == (1,274):
        print("click button 3 (top) ({})".format("press" if ev_val else "release"), flush=True)
    else:
        print("Unknow: type={}, code={}, value={}".format(ev_type, ev_code, ev_val), flush=True)
```

tobiasBora, 2019-03-22:
Hum, after some more investigations, I found your page https://digimend.github.io/support/howto/trbl/locating_failure/ . So basically, `usbhid-dump` works and display pressure, same for `cat event` that reports things like:

```
report (size 8) (numbered) =  01 01 15 1d 59 44 7e 12
Button.0001 = 1
Button.0002 = 0
Button.0003 = 0
GenericDesktop.X = 7445
GenericDesktop.Y = 17497
Digitizers.TipPressure = 4734
```
then, `evtest` works as well:

```
#  evtest     
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      AT Translated Set 2 keyboard
/dev/input/event1:      Lid Switch
/dev/input/event2:      Sleep Button
/dev/input/event3:      Power Button
/dev/input/event4:      Video Bus
/dev/input/event5:      Video Bus
/dev/input/event6:      VEIKK.INC A30 Mouse
/dev/input/event7:      VEIKK.INC A30
/dev/input/event8:      VEIKK.INC A30
/dev/input/event9:      ETPS/2 Elantech Touchpad
/dev/input/event10:     Asus Wireless Radio Control
/dev/input/event11:     PC Speaker
/dev/input/event12:     Asus WMI hotkeys
/dev/input/event13:     HDA Intel HDMI HDMI/DP,pcm=3
/dev/input/event14:     HDA Intel HDMI HDMI/DP,pcm=7
/dev/input/event15:     HDA Intel HDMI HDMI/DP,pcm=8
/dev/input/event16:     HDA Intel HDMI HDMI/DP,pcm=9
/dev/input/event17:     HDA Intel HDMI HDMI/DP,pcm=10
/dev/input/event18:     HDA Intel PCH Headphone Mic
/dev/input/event19:     USB2.0 HD UVC WebCam: USB2.0 HD
/dev/input/event20:     lircd-uinput
Select the device event number [0-20]: 6
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x2feb product 0x2 version 0x100
Input device name: "VEIKK.INC A30 Mouse"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value   7287
      Min        0
      Max    32767
    Event code 1 (ABS_Y)
      Value  16764
      Min        0
      Max    32767
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max     8191
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
[...]
Event: time 1553228515.881700, -------------- SYN_REPORT ------------
Event: time 1553228515.885701, type 3 (EV_ABS), code 1 (ABS_Y), value 20692
Event: time 1553228515.885701, -------------- SYN_REPORT ------------
[... 80 lines cut ...]
motion a[0]=16999 a[1]=39439 
motion a[0]=16997 a[1]=39443 
motion a[0]=16995 a[1]=39443 
motion a[0]=16993 a[1]=39445 
motion a[0]=16991 a[1]=39445 
motion a[0]=16987 a[1]=39445 
motion a[0]=16983 a[1]=39445 
motion a[0]=16979 a[1]=39445 
motion a[0]=16975 a[1]=39445 
motion a[0]=16971 a[1]=39445 
motion a[0]=16969 a[1]=39445 
motion a[0]=16967 a[1]=39445 
motion a[0]=16963 a[1]=39445 
motion a[0]=16961 a[1]=39445
[...]
button release 1 
motion a[0]=16849 a[1]=39127 
motion a[0]=16819 a[1]=39001 
motion a[0]=16775 a[1]=38843 
motion a[0]=16717 a[1]=38647 
motion a[0]=16639 a[1]=38409
```
I hope you have an idea what can go wrong...

Thanks!

george-viaud, 2019-09-17:
I think you will need to follow the steps outlined in the link below (I just did for the VEIKK A15 and submitted a Issue with the same).

https://digimend.github.io/support/howto/trbl/diagnostics/

Best of luck

ShapeShifter499, 2019-12-11:
Any updates on this?  Currently with default drivers (I think that's libinput) the gesture pad and touch buttons on this A30 works fine but there is no pressure sensitivity. But using the driver from https://github.com/jlam55555/veikk-s640-driver Everything but the gesture pad and touch buttons work.