DIGImend/digimend-kernel-drivers #272: Huion H610Pro v2 not working

fxrs, 2019-03-26:
### Graphic Tablet: Huion H610Pro v2
Distro: openSUSE Tumbleweed
Kernel: 5.0.2-1-default

I bought this tablet hoping it would work with a recent kernel like the v1 variant should according to the tablets section on the website. 
Before checking the tools the only change was the light indicator showing up when moving the pen or pressing physical buttons.

Error in dmesg:
```
[41324.514637] usb 3-1.2: new full-speed USB device number 22 using ehci-pci
[41324.624774] usb 3-1.2: New USB device found, idVendor=256c, idProduct=006e, bcdDevice= 1.00
[41324.624778] usb 3-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[41324.624779] usb 3-1.2: Product: Huion Tablet
[41324.624781] usb 3-1.2: Manufacturer: HUION
[41324.719857] uclogic 0003:256C:006E.001C: failed to enable abstract keyboard
[41324.720501] input: HUION Huion Tablet Pen as /devices/pci0000:00/0000:00:1d.0/usb3/3-1/3-1.2/3-1.2:1.0/0003:256C:006E.001C/input/input45
[41324.720860] uclogic 0003:256C:006E.001C: input,hidraw0: USB HID v1.11 Device [HUION Huion Tablet] on usb-0000:00:1d.0-1.2/input0
[41324.742269] uclogic 0003:256C:006E.001D: No inputs registered, leaving
[41324.742423] uclogic 0003:256C:006E.001D: hidraw1: USB HID v1.11 Device [HUION Huion Tablet] on usb-0000:00:1d.0-1.2/input1
```

`lsusb` shows device `Bus 003 Device 019: ID 256c:006e`
The following is a sample of `sudo usbhid-dump -es -m 256c:006e`. Physical buttons do not prodoce output but pen movement does.

```
003:019:001:STREAM             1553623841.774553
 0A 00 30 14 6E 22 00 00 FB 00
```
Format seems to be
```
streamid             counter
 <static> <state> <x movement?> <x pos> <y movement?> <y pos> <pressure1> <pressure2>  <x tilt> <y tilt>
```
Where state is
```
C0 no contact
C1 contact
C2 button1 (lower one on pen)
C3 contact and button1
C4 button2
C5 contact and button2
```
At some point `usbhid-dump` ended with

```
003:022:000:Interrupt transfer failed
003:022:001:Interrupt transfer failed
No more interfaces to dump
003:022:001:Failed to attach to the kernel driver: Entity not found
003:022:000:Failed to attach to the kernel driver: Entity not found
```

I will try to check in again in case I can provide more information. 
(I am also considering to send the device back since I experienced different problems with Windows 10, its driver and certain programs.)

melanes, 2019-03-28:
I also purchased this tablet and it is not working for me either. The stylus does not take over the mouse pointer but through the terminal I can see the system is receiving some interaction from the device.

Distro: Xubuntu 18.04.2 LTS
Kernel: Linux 4.15.0-46-generic (x86_64)

[    3.027634] usb 4-1: Product: Huion Tablet
[    3.027636] usb 4-1: Manufacturer: HUION
[   34.467451] input: HUION Huion Tablet Pen as /devices/pci0000:00/0000:00:1a.1/usb4/4-1/4-1:1.0/0003:256C:006E.0003/input/input11
[   34.467586] uclogic 0003:256C:006E.0003: input,hidraw2: USB HID v1.11 Device [HUION Huion Tablet] on usb-0000:00:1a.1-1/input0
[   34.468482] uclogic 0003:256C:006E.0004: hidraw3: USB HID v1.11 Device [HUION Huion Tablet] on usb-0000:00:1a.1-1/input1
[  319.637688] uclogic 0003:256C:006E.0007: hidraw2: USB HID v1.11 Device [HUION Huion Tablet] on usb-0000:00:1a.1-1/input1
[  319.747655] input: HUION Huion Tablet Pen as /devices/pci0000:00/0000:00:1a.1/usb4/4-1/4-1:1.0/0003:256C:006E.0008/input/input17
[  319.807322] uclogic 0003:256C:006E.0008: input,hidraw3: USB HID v1.11 Device [HUION Huion Tablet] on usb-0000:00:1a.1-1/input0

[   34.467118] uclogic 0003:256C:006E.0003: failed to enable abstract keyboard
[   34.468425] uclogic 0003:256C:006E.0004: No inputs registered, leaving
[  319.637617] uclogic 0003:256C:006E.0007: No inputs registered, leaving
[  319.747245] uclogic 0003:256C:006E.0008: failed to enable abstract keyboard

from lsusb Bus 004 Device 002: ID 256c:006

My usbhid-dump does not give an error or end as the above user. I simply had to end the process with a ctrl-c. Pressing the stylus buttons did do something as did tilt and movement, though it's not clear how it would work in a graphics program.

lukaskotik, 2019-03-28:
The same issue here. I bought the tablet and I did not notify that v2 version is quite different hardware than v1..

Ubuntu 18.04.2 LTS, kernel 4.15.0-46

melanes, 2019-03-29:
I was able to get the device working!

For Ubuntu 18.04.2 LTS

1. Updated to Kernel 5.0.5-050005-generic, 
2. Installed the latest digimend drivers V9

Testing the tablet in various applications, the tilt appears to work in Krita as it should.

lukaskotik, 2019-03-29:
I tried digimend drivers v9 on 4.18.0-16-generic kernell and it completely disabled my keyboard and mouse. But the tablet was (at least partially) working. But no mouse, no keyboard was usable.
In recovery mode I removed the digimend and run

`sudo apt install xserver-xorg-input-all`

and everything went back to normal. Only removing digimend did not help.

So for now, I am out of luck. I should by Wacom.

lukaskotik, 2019-04-03:
It now works for me. 
Ubuntu 18.04, kernel 4.18.0-17-generic, latest Digimend compiled from source.
Everything, including tilt, works.

fxrs, 2019-04-08:
Still not working and no difference noticeable to me.
Kernel: 5.0.5-1-default

`make dkms_install` on `digiment-kernel-drivers.git` (`f7edb78`) shows in part the following:

```
`hid-uclogic.ko:
Running module version sanity check.
Error! Module version 10 for hid-uclogic.ko
is not newer than what is already found in kernel 5.0.5-1-default (D303F0C9FA46DEA0D24D612).
You may override by specifying --force.`
```

I have no idea how to proceed and from my point of view it is broken.

spbnick, 2019-04-08:
@fxrs, regarding DKMS issue please see: https://github.com/DIGImend/digimend-kernel-drivers#dkms-issues-preventing-correct-installation

fxrs, 2019-04-08:
Thanks for the info, @spbnick. Previous problems lead to to try via DKMS. I now tried with `make`, `make install` and this time it was a success.

ekortright, 2020-01-10:
For others with the same problem:

I have a laptop running Pop!_OS (flavor of Ubuntu 19.10, kernel 5.3.0).  Just purchased a Huion H610 Pro V2 and after installing the v. 9 .deb package the device was not recognized.  Running usbhid-dump showed that the tablet data could be read.  Rebooting had no effect.

I uninstalled the package, cloned the digimend-kernel-drivers repository, installed the linux headers, and ran `sudo make dkms_install` and now the tablet is detected and behaves like a mouse (including the pen buttons).  Pressure sensitivity verified to work in GIMP.  Now to figure out how to make the tablet buttons work . . . .

peterjc, 2020-04-04:
Cross reference #306, where there is a hint that the USB ID could depend on the device firmware?

peterjc, 2020-04-04:
Just got our Huion H610Pro v2 working on a Raspberry Pi after compiling latest DIGImend from source (without DKMS support).

Can confirm it behaves like a mouse when the pen is brought close to the surface, including contact for left click, and the right mouse button on the pen works as expected.

Have not confirmed the other pen button, or the 8 slate buttons work, nor tested pressure sensitivity (does not seem to be used within TuxPaint which was my primary use case).

peterjc, 2020-06-13:
I updated this Raspberry Pi 3B+ to Raspbian Buster, and after reinstalling the DIGImend kernaql driers, and now my Huion H610Pro v2 is working again.

Note ``lsusb`` does not name the device:

```
$ lsusb
Bus 001 Device 006: ID 256c:006d
...
```

I can understand not giving a full model given Huion's history of reusing the same USB ID, but why isn't it even saying something generic like "Huion Graphics Tablet"? Is this driver issue or does the problem lie elsewhere in the Linux stack?

spbnick, 2020-06-14:
@peterjc, lsusb displays the name the device itself reports, here. So this is Huion's fault of not providing the name. Check out the output of `lsusb -v` as well to see more of what's going on.

peterjc, 2020-06-14:
@spbnick how strange. Thanks.

Anyway, given the basics worked for me and @ekortright perhaps this issue can be closed - Unless any of the people posting earlier are still having problems?