DIGImend/digimend-kernel-drivers #273: Huion h950p tilt support not working on ubuntu

sunil-ghorpade, 2019-03-27:
Huion h950p tilt support not working on ubuntu , i tested the tablet on windows and tilt support works but does not work on ubuntu even with v9 drivers. i have ubuntu 18.10. I tested it on krita

spbnick, 2019-03-27:
Thank you for the report, @sunil-ghorpade. Which version of the driver are you using? Could you post the output of `sudo evtest` for your tablet, where you do a brief stroke with the pen?

sunil-ghorpade, 2019-03-27:
Hi Nikolai,
Thanks for the prompt reply. I ran evtest. There is no response on the
terminal while using the pen although apart from tilt it works. The keys on
the tablet no corrosponds to page up and key  no 3 corrosponds to / as you
can see from screenshot attached. other keys dont respond.
Regards,

Sunil Ghorpade
+91-9922298533

sunil-ghorpade, 2019-03-27:
sorry forgot to mention i have installed v9 driver , and also v10. i was
trying to uninstall v9 but could not.
Regards,

Sunil Ghorpade
+91-9922298533

sunil-ghorpade, 2019-03-27:
key no 1 corresponds to page up

sunil-ghorpade, 2019-03-27:
I have configured the keys on the tablet , now only the tilt support issue
remains. How to install v9 driver ?

Regards,

Sunil Ghorpade
+91-9922298533

sunil-ghorpade, 2019-03-28:
I saw some initial part of  your video on youtube. I am attaching the
result of sudo usbhid-dump -es -m 256c:006e
Regards,

Sunil Ghorpade
+91-9922298533

sunil-ghorpade, 2019-03-28:
Regards,

Sunil Ghorpade
+91-9922298533

sunil-ghorpade, 2019-03-28:
i am watching your video and doing the tests , i did not understand tour
message yesterday, please bear with me,
Regards,

Sunil Ghorpade
+91-9922298533

sunil-ghorpade, 2019-03-28:
result of  sudo evtest|grep --line-buffered ABS_TILT
attached
Regards,

Sunil Ghorpade
+91-9922298533

sunil-ghorpade, 2019-03-28:
i am watching your video and doing the tests, so i am sending multiple
mails, please excuse me.
Regards,

Sunil Ghorpade
+91-9922298533

sunil-ghorpade, 2019-03-28:
Test:as per your video 58.24 , on my computer
Regards,

Sunil Ghorpade
+91-9922298533

sunil-ghorpade, 2019-03-28:
I have installed the driver that you have shown in your video(
https://github.com/DIGImend/digimend-kernel-drivers.git, this one) , do i
need to uninstall the previously installed drivers?
Regards,

Sunil Ghorpade
+91-9922298533

sunil-ghorpade, 2019-03-28:
I think the twist is working fine!!!! . Thanks. But please tell me if i am
missing something.
Regards,

Sunil Ghorpade
+91-9922298533

spbnick, 2019-03-28:
Glad to hear you figured it out, @sunil-ghorpade. If you can, please write how you made it work, and how you tested that it works. This will help other users.

sunil-ghorpade, 2019-03-28:
I was inspired by your video. Earlier i did not know about digimend, I had
contacted huion support, who gave me v8 which made the tablet work minus
tilt. when i asked about tilt they said it is not supported in ubuntu. I
never give up ,so started exploring thankfully and got your response.
                    Now i am going to write to them about the great job
that you have done for them and ask them to update this in their system.
They are not aware of your latest commits , you should also update them.

This is what they said
Dear sunil Ghorpade,

{Thanks for your prompt reply.

Actually, our tablet is much more compatible with Windows and Mac than Ubuntu
at present, so you can't use tilt if you make the tablet work with
Ubuntu.

Hope you can understand. We suggest you work in Windows or Mac to use tilt.

If you have any further question, please don't hesitate to contact us again.}

so you see , unknowingly they are giving wrong information to customers .
and tilt is a fantastic tool for painting and drawing. I installed the
driver at root, i had followed some instructions from the net, i will dig
them out and share them with you, although i think you already know it.
Regards,

Sunil Ghorpade
+91-9922298533

scythianfuego, 2019-04-14:
Hello @spbnick ,
I can confirm that for Huion H950P tilt support doesn't work when using _digimend-dkms_9_all.deb_ but does when you clone the _master_ branch and do _sudo make dkms_install_

Linux Mint 19.1 Tessa

sunil-ghorpade, 2019-04-15:
You are right,  Nick's driver , when installed works. I tested it with
krita.

Nikorasu, 2021-02-20:
> Hello @spbnick ,
> I can confirm that for Huion H950P tilt support doesn't work when using _digimend-dkms_9_all.deb_ but does when you clone the _master_ branch and do _sudo make dkms_install_
> 
> Linux Mint 19.1 Tessa

Is this still the only way to get Tilt working?

sunil-ghorpade, 2021-02-20:
I think so. I have not made any changes since it is working fine.

Nikorasu, 2021-02-20:
Ok, Thanks for the info! I'll give that a try later, to see if I can get it working. ;)