DIGImend/digimend-kernel-drivers #275: Huion HS610 support?

ipsod, 2019-04-08:
Does it exist?  If not, what's the next step?

Imerion, 2019-04-19:
I am also curious about this. I managed to get my new Huion HS610 to work well after compiling this driver and adding a .conf file as per older instructions for H610. Pen, pen buttons and pressure detection works great. The buttons on the pad, however, do nothing. The problem seems to be that the pad is not detected. Running `xsetwacom --list` gives this:

`HUION Huion Tablet stylus       	id: 10	type: STYLUS    
HUION Huion Tablet eraser       	id: 15	type: ERASER `

I don't know what the eraser is for since the device doesn't seem to feature one. For the H610, people previously reported getting something like` HUION PenTablet Pad pad id: 12 type: PAD`. Could it be confusing the PAD with the ERASER? Using xsetwacom --set with the ERASER device does nothing, it does not register input from the buttons.

Imerion, 2019-04-21:
Oh, I should probably add that I had to create my own 50-huion.conf file. (Again, as per instructions from a 2016 thread). Without that the tablet did nothing despite latest drivers being installed (tried both v9 and latest source).

emmetoneillpdx, 2019-05-01:
Hey all. I just picked up one of these **Huion HS610** tablets and the basic functionality seems to work well with current version of digimend. I haven't been able to get anything from the frame buttons or dials yet, but I may have not configured things correctly. Like @Imerion said, you may have to create a configuration file for it. Mine is in `/etc/X11/xorg.conf.d/50-huion-hs610.conf` and is based on [this section](https://github.com/DIGImend/digimend-kernel-drivers#enabling-wacom-xorg-driver).

I also went through and attempted to collect some diagnostics, because why not. Hopefully they'll be useful for improving support. Feel free to ping me if you need anything else tested or redone.

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3133687/descriptors.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3133688/hid_report_descriptors.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3133689/probe.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3133690/pen_coords.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3133691/pen_tilt.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3133692/pen_pressure.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3133694/pen_buttons.txt)

The frame dials and button diagnostics didn't return anything, so let me know if there is anything further I can do or test.

Thanks!

spbnick, 2019-05-05:
Thank you for the diagnostics, @emmetoneillpdx. This seems to be something new from Huion: they're using a new product ID this time. As such, the DIGImend kernel driver wouldn't actually handle it, and there's no use installing it as is. I see that the tablet responds to uclogic-probe, which is good. Could you run uclogic-probe, and then, without reconnecting your tablet or rebooting, collect the pen diagnostics again? Frame controls would also be useful.

emmetoneillpdx, 2019-05-12:
> Thank you for the diagnostics, @emmetoneillpdx.

You're welcome, Nick. Thanks for all the great work you do.

> Could you run uclogic-probe, and then, without reconnecting your tablet or rebooting, collect the pen diagnostics again? Frame controls would also be useful.

Sure thing, here you go:

[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3169718/probe.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3169719/pen_coords.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3169720/pen_tilt.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3169721/pen_pressure.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3169722/pen_buttons.txt)
[frame_dials.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3169723/frame_dials.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3169724/frame_buttons.txt)

By the way, this tablet's frame buttons are all on one side, with six (2col x 3row) on top, one in the middle of the dial, and another six (2col x 3row) on the bottom. I pressed them in left-to-right, top to bottom order, starting with the top six, then the dial button, then the bottom six.

Imerion, 2019-05-20:
If there is need for any other diagnostics or tests for this particular tablet, I'm happy to help as well!

spbnick, 2019-05-26:
Thanks, @Imerion, this seems to be enough for the start. One question, though: does it have a touch dial, or a physical wheel?

Imerion, 2019-05-26:
Touch dial. Huion has a good image of the button layout here: https://www.huiontablet.com/media/catalog/product/cache/1/image/450x450/9df78eab33525d08d6e5fb8d27136e95/2/_/2_11_8.jpg

spbnick, 2019-05-26:
Thank you. Then that explains the touch dial having absolute coordinates and a separate report for taking the finger off.
I made a quick change to have the base functionality working, please download and try this version: https://github.com/spbnick/digimend-kernel-drivers/archive/support_huion_hs610.zip

Once we get that working, I can make the dial button work and research how to make the touch dial work with the Wacom drivers.

Imerion, 2019-05-26:
Fantastic, thanks! Your work is really really appreciated! Will try it in the coming days and report back.

Imerion, 2019-05-30:
I tried the driver now, and it works well without the need for the extra config file! That is, pressure sensitivity and the buttons on the pen. "xsetwacom --list" still only gives:
`HUION Huion Tablet stylus       	id: 10	type: STYLUS    
HUION Huion Tablet eraser       	id: 15	type: ERASER`
So I have no way to set the buttons yet. But perhaps that is the next step?

Just tell me if any more diagnostics can be helpful and I will run them. :)

spbnick, 2019-05-31:
Glad to hear that something works :) However, the buttons should've worked as well. For the start please post the `dmesg` output after plugging in the tablet, and `evdev` output - let's see if the button device is there.

Imerion, 2019-05-31:
Here you are:

[dmesg.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3240980/dmesg.txt)

[evdev.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3240981/evdev.txt)

I'm not sure I got the evdev output right. I used evtest, choose the Huion device and then clicked all buttons before touching the surface with the pen. After that I aborted the test. Only using the pen seemed to result in a response from evtest.

spbnick, 2019-05-31:
Argh, of course I meant `evtest`, not `evdev`, sorry.
OK, looking at what you attached I can see that the driver is not actually being used :)
Let's find out why. Could you post the output of you trying to install it, however you do it? Also, please post the output of `sudo modinfo hid-uclogic`.

Imerion, 2019-05-31:
Oh, sorry! I might have messed up somewhere. I downloaded the archive and did "make" and then "make install". Perhaps I should have removed the old one somehow? I thought that wasn't necessary since this was a new module. I did remove my modified "50-huion-hs610.conf" though.

Here is "sudo modinfo hid-uclogic":
`filename:       /lib/modules/5.1.0-050100rc5-generic/extra/hid-uclogic.ko
version:        10
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     9EECED3A5EC359ED80D22EE
alias:          hid:b0003g*v000028BDp00000042
alias:          hid:b0003g*v000028BDp00000094
alias:          hid:b0003g*v000028BDp00000075
alias:          hid:b0003g*v000028BDp00000071
alias:          hid:b0003g*v000028BDp00000074
alias:          hid:b0003g*v00002179p00000053
alias:          hid:b0003g*v00005543p00003031
alias:          hid:b0003g*v00005543p00000047
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        hid,usbhid
retpoline:      Y
name:           hid_uclogic
vermagic:       5.1.0-050100rc5-generic SMP mod_unload`

spbnick, 2019-05-31:
I would recommend using `sudo make dkms_install` instead. You'd need to do `sudo make uninstall` first, though.

Anyway, I noticed a problem with the code which prevented the driver from binding to the tablet. Could you please redownload it from the same location, reinstall, and try again? Thank you.

Imerion, 2019-05-31:
Awesome, now it works! :D "xsetwacom --list" gives:

> HUION Huion Tablet Pen stylus   	id: 14	type: STYLUS    
> HUION Huion Tablet Pad pad      	id: 15	type: PAD 

I can set the first 8 buttons. (Six above the wheel, two below). The final four do nothing when testing with "xinput test "HUION Huion Tablet Pad pad".

spbnick, 2019-06-04:
This sounds better. Could you post the `evtest` output for those final four buttons?

Imerion, 2019-06-04:
Here you go: 
[evdev.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3251839/evdev.txt)

spbnick, 2019-06-04:
Thank you, this looks correct. Could you now post the contenst of /var/log/Xorg.0.log after (re)connecting the tablet, or booting with it connected?

Imerion, 2019-06-06:
Sorry for the delay, here you go:

[Xorg.0.log](https://github.com/DIGImend/digimend-kernel-drivers/files/3263479/Xorg.0.log)

This is for booting with the tablet connected.

spbnick, 2019-06-07:
Everything looks correct. Strange. If you're sure you get nothing from those extra buttons, please post which exact version of Ubuntu you're using and I'll try it with another 12-button tablet.

Imerion, 2019-06-07:
I am so sorry, it does work! I had used the wrong numbers in my config file and that is why "evtest" didn't show anything while "xinput test" did. I'm really sorry for taking your time, but I can at least happily confirm that all 12 buttons and both buttons on the pen works perfectly!

If I can help with anything more for this or similar, just tell me. I really want to help since you work so hard on this. And I will be more careful in the future! :)

spbnick, 2019-06-07:
Whew, good to hear :) OK, then I can go on to adding the touch wheel and center button support on the weekend. My work on this is paid by donations somewhat. Contributions from you, or whoever else is reading are always welcome :)

[image] [image] [image]

spbnick, 2019-06-15:
Hi @Imerion, sorry didn't have the time nor energy to continue with your tablet last weekend, but am looking at it right now. It would need a bit of work to get the touch dial supported, but I have a rough plan now. Meanwhile, could you please run `uclogic-probe`, and then post the output of `usbhid-dump -es` where you move your finger around the touch dial **clockwise** several times? I need to know which way around the coordinates go to make sure the direction matches the Wacoms.

soulsplit, 2019-06-15:
Hi @spbnick , I also own this tablet and appreciate your work! I added the dump for the clockwise rotation.
[huion_HS610_clockwise_dial.log](https://github.com/DIGImend/digimend-kernel-drivers/files/3292869/huion_HS610_clockwise_dial.log)

spbnick, 2019-06-15:
Thanks a lot, @soulsplit! It looks that the coordinates increase counter-clockwise, which is opposite the Wacom's direction. So I'll have to flip it.

Imerion, 2019-06-15:
@spbnick No hurry at all, the main functionality works great after all. Would you need a dump from me as well for the dial?

spbnick, 2019-06-15:
@Imerion, I think the dump from @soulsplit would be enough. Your previous dump matches, assuming you did the clockwise rotation first, as I ask in the diagnostics HOWTO.

Imerion, 2019-06-15:
I did, but here is an extra anyway, just in case: 
[huion_HS610_clockwise_dial.log](https://github.com/DIGImend/digimend-kernel-drivers/files/3293455/huion_HS610_clockwise_dial.log)
Three clockwise dials.

C-Swick, 2019-06-26:
I have this tablet as well (running on arch linux with kde plasma) -- is there some magic trick to getting the "express keys" to work?  I have basic pen functionality but nothing else. I tried installing the driver posted above but it didn't fix it for me.

soulsplit, 2019-07-06:
Hi @spbnick,
do you happen to have the full HID report descriptor as code?
I'm working on adding the support for this tablet. I'm not super experienced yet, though. Any help and pointers are appreciated. :)
Thanks!

spbnick, 2019-07-07:
@UraharaKisuke-NGS, you'll need to use the Wacom X.org drivers for that. If you can see your tablet in `xsetwacom list` output, then you can set up your hotkeys with `xsetwacom`. See the README.md for more.

spbnick, 2019-07-07:
Hi @soulsplit, I have basics working in this branch: https://github.com/spbnick/digimend-kernel-drivers/tree/support_huion_hs610 - take a look.

sephirothtbm, 2019-07-08:
Hi @spbnick, first of all, thanks for all the amazing work, I installed the driver mentioned a above and all started to work very good, I used xsetwacom to set some shortcuts to the buttons, but my doubt is the next, can I set the scroll using xsetwacom? I tried a bit but didn't work, I'm not so experienced so maybe I did something wrong, but it is possible to do with the actual driver? if so, I will investigate a bit more, or do I have to wait for an update in the driver to make it works?

UraharaKisuke-NGS, 2019-07-08:
> @UraharaKisuke-NGS, you'll need to use the Wacom X.org drivers for that. If you can see your tablet in `xsetwacom list` output, then you can set up your hotkeys with `xsetwacom`. See the README.md for more.

@spbnick Yeahhh my xsetwacom list output is Pen and Eraser, no tablet listed even with the HS610 specific branch. looking at the settings for both says something to the tune of "custom key/button settings not supported by xorg, use -S  etc"

soulsplit, 2019-07-09:
> @spbnick Yeahhh my xsetwacom list output is Pen and Eraser, no tablet listed even with the HS610 specific branch. looking at the settings for both says something to the tune of "custom key/button [...]

Hi @UraharaKisuke-NGS, it may have happened that some `.o` files were already considered to be correct by `make`. In order to be sure all is compiled from scratch and your system has no previous installation:

1. run to clean up:
`make clean`
`sudo make dkms_uninstall`
2.  run to compile and install
`make`
`sudo make dkms_install`
3. make sure to remove all old kernel modules by running:
`sudo modprobe -r hid-kye hid-uclogic hid-polostar hid-viewsonic`
4. then (re-)connect your tablet

Mostly copied from the Readme.md and other comments.

spbnick, 2019-07-09:
@sephirothtbm, the wheel is not supported yet. Huion chose a weird way to implement it, and it will take a bit more time.

sephirothtbm, 2019-07-09:
@spbnick That's bad news, anyway with the basic functionality is enough to work now, just a quick question, is there a way to map the scroll up and down to any of the buttons on the pad? If no, well doesn't matter, thanks for all.

UraharaKisuke-NGS, 2019-07-11:
> > > @UraharaKisuke-NGS, you'll need to use the Wacom X.org drivers for that. If you can see your tablet in `xsetwacom list` output, then you can set up your hotkeys with `xsetwacom`. See the README.md for more.
> > 
> > 
> > @spbnick Yeahhh my xsetwacom list output is Pen and Eraser, no tablet listed even with the HS610 specific branch. looking at the settings for both says something to the tune of "custom key/button settings not supported by xorg, use -S etc"
> 
> Hi @UraharaKisuke-NGS, it may have happened that some `.o` files were already considered to be correct by `make`. In order to be sure all is compiled from scratch and your system has no previous installation:
> 
> 1. run to clean up:
>    `make clean`
>    `sudo make dkms_uninstall`
>    `sudo make uninstall`
> 2. run to compile and install
>    `make`
>    `sudo make dkms_install`
> 3. make sure to remove all old kernel modules by running:
>    `sudo modprobe -r hid-kye hid-uclogic hid-polostar hid-viewsonic`
> 4. then (re-)connect your tablet
> 
> Mostly copied from the Readme.md and other comments.

First off, thank you for the reply. Unfortunately it didn't work :/   I still see Stylus and Eraser in the xsetwacom list output -- no tablet.
When I do xsetwacom --get 12 (the device id for stylus) all -- it says that button actions are not supported by xorg.conf and that I should try shell format (-s) or something.

I may just wait until the next major release lol

hickscorp, 2019-07-25:
@UraharaKisuke-NGS If you still see the eraser, it's very likely that you're still using the `.deb` binary package.
Make sure that you uninstalled it, rebooted, and ran all the dkms build commands as explained above.

Lucretia, 2019-08-06:
> Hi @soulsplit, I have basics working in this branch: https://github.com/spbnick/digimend-kernel-drivers/tree/support_huion_hs610 - take a look.

I've just tried this branch.

Getting a strange output from evtest --grab /dev/input/event20:

```
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6d version 0x111
Input device name: "HUION Huion Tablet Pad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 256 (BTN_0)
    Event code 257 (BTN_1)
    Event code 258 (BTN_2)
    Event code 259 (BTN_3)
    Event code 260 (BTN_4)
    Event code 261 (BTN_5)
    Event code 262 (BTN_6)
    Event code 263 (BTN_7)
    Event code 264 (BTN_8)
    Event code 265 (BTN_9)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 331 (BTN_STYLUS)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max        1
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max        1
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
Event: time 1565054676.810894, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90009
Event: time 1565054676.810894, type 1 (EV_KEY), code 264 (BTN_8), value 1
Event: time 1565054676.810894, -------------- SYN_REPORT ------------
Event: time 1565054676.924900, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90009
Event: time 1565054676.924900, type 1 (EV_KEY), code 264 (BTN_8), value 0
Event: time 1565054676.924900, -------------- SYN_REPORT ------------
Event: time 1565054678.196952, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000a
Event: time 1565054678.196952, type 1 (EV_KEY), code 265 (BTN_9), value 1
Event: time 1565054678.196952, -------------- SYN_REPORT ------------
Event: time 1565054678.326956, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000a
Event: time 1565054678.326956, type 1 (EV_KEY), code 265 (BTN_9), value 0
Event: time 1565054678.326956, -------------- SYN_REPORT ------------
Event: time 1565054679.426992, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1565054679.426992, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 1
Event: time 1565054679.426992, -------------- SYN_REPORT ------------
Event: time 1565054679.583007, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1565054679.583007, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 0
Event: time 1565054679.583007, -------------- SYN_REPORT ------------
Event: time 1565054680.327029, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1565054680.327029, type 1 (EV_KEY), code 305 (BTN_EAST), value 1
Event: time 1565054680.327029, -------------- SYN_REPORT ------------
Event: time 1565054680.467005, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1565054680.467005, type 1 (EV_KEY), code 305 (BTN_EAST), value 0
Event: time 1565054680.467005, -------------- SYN_REPORT ------------
Event: time 1565054682.201113, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1565054682.201113, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 1
Event: time 1565054682.201113, -------------- SYN_REPORT ------------
Event: time 1565054682.319062, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1565054682.319062, type 1 (EV_KEY), code 304 (BTN_SOUTH), value 0
Event: time 1565054682.319062, -------------- SYN_REPORT ------------
Event: time 1565054682.615131, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1565054682.615131, type 1 (EV_KEY), code 305 (BTN_EAST), value 1
Event: time 1565054682.615131, -------------- SYN_REPORT ------------
Event: time 1565054682.713126, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1565054682.713126, type 1 (EV_KEY), code 305 (BTN_EAST), value 0
Event: time 1565054682.713126, -------------- SYN_REPORT ------------
```

Why are the bottom two buttons called south and east and not 10 and 11? The ring returns south and east also.

The centre button in the ring doesn't return anything.

If there's anything you want me to test, feel free to ask.

kristophesankar, 2019-09-07:
Recently bought this tablet as well. Naively thought Linux would support all tablets out of box. If you all want me to test anything I'd be happy when I have the time :)

Lucretia, 2019-09-08:
Well, it does support it out of the box, but the latest updates are from this repo, so best just include the drivers in your kernel build and install this by hand, they go into the extra directory so, there's no overwriting.

kedodrill, 2019-10-09:
Hi there, I'm looking to get this tablet at some point - any updates to this yet?

inniyah, 2019-12-05:
My Gaomon S620 seems to work too with this patch too (it seems to be identified by the same idVendor and idProduct: 256C:006D).

[1829503.077874] usb 2-8.6: new full-speed USB device number 118 using xhci_hcd
[1829503.184061] usb 2-8.6: New USB device found, idVendor=256c, idProduct=006d, bcdDevice= 1.00
[1829503.184066] usb 2-8.6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1829503.184070] usb 2-8.6: Product: Gaomon Tablet
[1829503.184073] usb 2-8.6: Manufacturer: GAOMON

inniyah, 2019-12-05:
Just for the record, Im currently using the code from master, git rev 8b228a755e44106c11f9baaadb30ce668eede5d4, which is supposed to add support for HUION HS64, which seems to share the same Id too (256c:006d). At least it works for my Gaomon S620 and, seeing that the changes are essentially the same, I would say that it should work for Huion HS610 too.

amruthsu, 2019-12-06:
###  **[Huion HS610](https://store.huion.com/products/huion-hs610?) basics working**
{ *touch ring not working }

**quick driver installation method on debian distro.**

Vendor ID : Product ID
`lsusb`
256c:006d 

pull source code
```
git clone https://github.com/spbnick/digimend-kernel-drivers.git
cd digimend-kernel-drivers
```

apply this branch
`git checkout origin/support_huion_hs610`

installation
`sudo dpkg -i ../digimend-dkms_10_all.deb`

reconnect usb tablet
```
$ xsetwacom --list
HUION Huion Tablet Pen stylus   	id: 13	type: STYLUS    
HUION Huion Tablet Pad pad      	id: 14	type: PAD   
```

optional button mapping for [GIMP](https://www.gimp.org/)
```
xsetwacom --set "HUION Huion Tablet Pad pad" Button 1 key +ctrl +z -z -ctrl
xsetwacom --set "HUION Huion Tablet Pad pad" Button 2 key +shift +e
xsetwacom --set "HUION Huion Tablet Pad pad" Button 3 key +p
xsetwacom --set "HUION Huion Tablet Pad pad" Button 8 key +shift + =
xsetwacom --set "HUION Huion Tablet Pad pad" Button 9 key + –
xsetwacom --set "HUION Huion Tablet Pad pad" Button 10 key + ]
xsetwacom --set "HUION Huion Tablet Pad pad" Button 11 key + [
xsetwacom --set "HUION Huion Tablet Pad pad" Button 12 key + m

```
optional button mapping for [Blender](https://www.blender.org/)
```
xsetwacom --set "HUION Huion Tablet Pad pad" Button 1 key +r
xsetwacom --set "HUION Huion Tablet Pad pad" Button 2 key +e
xsetwacom --set "HUION Huion Tablet Pad pad" Button 3 key +s
xsetwacom --set "HUION Huion Tablet Pad pad" Button 8 key +tab
xsetwacom --set "HUION Huion Tablet Pad pad" Button 9 key +z
xsetwacom --set "HUION Huion Tablet Pad pad" Button 10 key +w
xsetwacom --set "HUION Huion Tablet Pad pad" Button 11 key +shift +ctrl +tab
xsetwacom --set "HUION Huion Tablet Pad pad" Button 12 key +a

```

[official Huion HS610 user manual pdf](http://driver.huion.com/instruction/hs610-hs64/User_Manual_HS610.zip)
[official new firmware update available.](https://www.huion.com/firmware.html)

spbnick, 2020-01-06:
All the work on Huion HS610 support was merged into master, so you don't need to use the `support_huion_hs610` branch anymore. Further work will go directly to master.

spbnick, 2020-01-06:
I also added code to support the button in the middle of the dial and am working on support for the dial itself. However, it's getting pretty convoluted and we might need to refactor the driver again soon.

sephirothtbm, 2020-02-16:
Hi, I just uninstalled previous version and installed the master but when I do it I lost the right click button on the stylus, tried to reconfigure it using xsetwacom but didn't work, had to reinstall the copy of the dev driver, btw, uninstalled all before compiling and installed each version, so it wasn't that the cause.

spbnick, 2020-02-29:
Hi everyone, please try the latest code in #345. It should support all the features of Huion HS610 as much as possible, including the touch ring.

**NOTE**: due to Wacom driver specifics, you will need to execute `xsetwacom --set "HUION Huion Tablet Touch Ring pad" Suppress 0` to get decent speed out of your touch ring.

Please report your testing results here.

spbnick, 2020-03-01:
You can download the code in #345 as a [ZIP archive](https://github.com/DIGImend/digimend-kernel-drivers/archive/support_huion_hs610_dial_reports.zip).

Imerion, 2020-03-01:
That is fantastic work, thank you so much! Will try as soon as I get the chance.

Can I do "sudo make uninstall" from that directory or do I need to find the version that is installed and do it from there? I don't do this very often. :)

Imerion, 2020-03-01:
I believe I have the latest version installed now. (Can I check that?)
xsetwacom --list devices shows "HUION Huion Tablet Touch Ring pad	id: 16	type: PAD "

xinput test "HUION Huion Tablet Touch Ring pad" shows a long line of this when I rotate along the ring:
motion a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=1 
motion a[3]=0 a[4]=0 a[5]=1 
motion a[3]=0 a[4]=0 a[5]=1 
motion a[3]=0 a[4]=0 a[5]=1 
motion a[3]=0 a[4]=0 a[5]=1 
motion a[3]=0 a[4]=0 a[5]=1 
motion a[3]=0 a[4]=0 a[5]=1 
motion a[3]=0 a[4]=0 a[5]=1 
motion a[3]=0 a[4]=0 a[5]=0

It doesn't seem to be automatically mapped to anything though. I couldn't set it to a function in Krita, for example and it doesn't scroll or anything. But perhaps I need an additional xsetwacom entry first?

sephirothtbm, 2020-03-02:
First at all, Thanks for the work, now I have the same issue like Imerion, I installed latest version but doesn't scroll, do I need to map the ring using xsetwacom?

spbnick, 2020-03-02:
@Imerion, the device name looks right. The reports don't look right. Can you post your /var/log/Xorg.0.log?
@sephirothtbm, could you post yours as well?
Thanks!

spbnick, 2020-03-02:
Oh, and also your distros, please?

sephirothtbm, 2020-03-02:
Sure, this is the content:
[    35.943] (--) Log file renamed from "/var/log/Xorg.pid-1098.log" to "/var/log/Xorg.0.log"
[    35.985] 
X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
[    35.985] Build Operating System: Linux 4.4.0-143-generic x86_64 Ubuntu
[    35.985] Current Operating System: Linux Sakura-hm 5.0.0-38-generic #41-Ubuntu SMP Tue Dec 3 00:27:35 UTC 2019 x86_64
[    35.985] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.0.0-38-generic root=UUID=17361ec3-56a7-424e-bfef-29b4cae51933 ro quiet splash vt.handoff=1
[    35.985] Build Date: 03 April 2019  09:03:57AM
[    35.985] xorg-server 2:1.20.4-1ubuntu3 (For technical support please see http://www.ubuntu.com/support) 
[    35.985] Current version of pixman: 0.36.0
[    35.985] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[    35.985] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    35.986] (==) Log file: "/var/log/Xorg.0.log", Time: Sun Mar  1 19:26:48 2020
[    36.074] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    36.163] (==) No Layout section.  Using the first Screen section.
[    36.163] (==) No screen section available. Using defaults.
[    36.163] (**) |-->Screen "Default Screen Section" (0)
[    36.163] (**) |   |-->Monitor "<default monitor>"
[    36.171] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[    36.171] (==) Automatically adding devices
[    36.171] (==) Automatically enabling devices
[    36.171] (==) Automatically adding GPU devices
[    36.171] (==) Automatically binding GPU devices
[    36.171] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    36.204] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    36.204] 	Entry deleted from font path.
[    36.204] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    36.204] 	Entry deleted from font path.
[    36.204] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    36.204] 	Entry deleted from font path.
[    36.204] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    36.204] 	Entry deleted from font path.
[    36.204] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    36.204] 	Entry deleted from font path.
[    36.204] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	/usr/share/fonts/X11/Type1,
	built-ins
[    36.204] (==) ModulePath set to "/usr/lib/xorg/modules"
[    36.204] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[    36.204] (II) Loader magic: 0x55d691136020
[    36.204] (II) Module ABI versions:
[    36.204] 	X.Org ANSI C Emulation: 0.4
[    36.204] 	X.Org Video Driver: 24.0
[    36.204] 	X.Org XInput driver : 24.1
[    36.204] 	X.Org Server Extension : 10.0
[    36.205] (++) using VT number 1

[    36.206] (II) systemd-logind: took control of session /org/freedesktop/login1/session/c1
[    36.206] (II) xfree86: Adding drm device (/dev/dri/card0)
[    36.207] (II) systemd-logind: got fd for /dev/dri/card0 226:0 fd 12 paused 0
[    36.208] (**) OutputClass "nvidia" ModulePath extended to "/usr/lib/x86_64-linux-gnu/nvidia/xorg,/usr/lib/xorg/modules"
[    36.208] (--) PCI:*(1@0:0:0) 10de:1040:10de:0915 rev 161, Mem @ 0xf6000000/16777216, 0xe8000000/134217728, 0xf0000000/33554432, I/O @ 0x0000e000/128, BIOS @ 0x????????/131072
[    36.208] (II) LoadModule: "glx"
[    36.387] (II) Loading /usr/lib/x86_64-linux-gnu/nvidia/xorg/libglx.so
[    38.165] (II) Module glx: vendor="NVIDIA Corporation"
[    38.165] 	compiled for 4.0.2, module version = 1.0.0
[    38.165] 	Module class: X.Org Server Extension
[    38.169] (II) NVIDIA GLX Module  390.116  Sun Jan 27 06:24:32 PST 2019
[    38.195] (II) Applying OutputClass "nvidia" to /dev/dri/card0
[    38.195] 	loading driver: nvidia
[    38.435] (==) Matched nvidia as autoconfigured driver 0
[    38.435] (==) Matched nouveau as autoconfigured driver 1
[    38.435] (==) Matched modesetting as autoconfigured driver 2
[... 530 lines cut ...]
[    63.591] (**) Option "fd" "38"
[    63.591] (II) event1  - Power Button: device removed
[    63.591] (**) Option "fd" "41"
[    63.591] (II) event0  - Power Button: device removed
[    63.592] (**) Option "fd" "42"
[    63.592] (**) Option "fd" "43"
[    63.592] (**) Option "fd" "44"
[    63.592] (II) event4  - MOSART Semi. 2.4G Wireless Mouse: device removed
[    63.592] (**) Option "fd" "48"
[    63.592] (II) event7  - Eee PC WMI hotkeys: device removed
[    63.592] (**) Option "fd" "42"
[    63.592] (II) event6  - Logitech K360: device removed
[    63.593] (**) Option "fd" "43"
[    63.593] (II) event3  - Logitech K360: device removed
[    63.725] (II) systemd-logind: got pause for 13:65
[    63.725] (II) systemd-logind: got pause for 13:70
[    63.725] (II) systemd-logind: got pause for 13:71
[    63.725] (II) systemd-logind: got pause for 13:72
[    63.726] (II) systemd-logind: got pause for 13:67
[    63.726] (II) systemd-logind: got pause for 13:68
[    63.726] (II) systemd-logind: got pause for 13:74
[    63.726] (II) systemd-logind: got pause for 13:64
[    63.726] (II) systemd-logind: got pause for 13:73

And the OS is Ubuntu 19.04

spbnick, 2020-03-02:
Thanks, @sephirothtbm! I see you're using a different version of the Wacom drivers, which likely have different behavior, so I'll try Ubuntu 19.04 myself and see what's up.

[71 more replies]