DIGImend/digimend-kernel-drivers #279: XP-Pen G640 issues

mcpalmer1980, 2019-05-04:


mcpalmer1980, 2019-05-04:
I would like to report two issues I've experienced using the digimend tablet drivers for my new XP-Pen Star G640 tablet, one of which I fixed and the other ongoing. I'm using digimend-dkms_9_all.deb on Linux Mint 18.3. Note that I'm using the synaptics driver for the Elantech ETPS/2 Touchpad on my Acer Aspire E15-576G-5761 Laptop, which supposedly prevents the touchpad on my Sony Dualshock4 game controller from working. Not sure if it could also effect the tablet. In addition, I use a Logitech M570 Trackball with a wireless USB dongle as my primary mouse.

When I first connected the device without installing drivers it worked properly for about a minute. It moved the cursor to absolute coordinates when stylus entered proximity, clicked when I touched the stylus to the pad, and detected left and right clicks from the stylus buttons. After about a minute it stopped working properly for no apparent reason. It no longer detected proximity and operated using relative coordinates.

To fix the problem I installed the latest digimend tablet drivers and restarted my x-windows session. The computer froze at a black screen and I couldn't switch to a terminal, so I hard-reset the machine into recovery mode. I started x-windows from the terminal and receieved an error that the NoMatchProduct parameter is invalid within the 'InputClass' section of /usr/share/X11/xorg.conf.d/50-digimend.conf. Strangely x only started after commenting out the 'NoMatchProduct "MousePen"' line from /etc/X11/xorg.conf.d/50-digimend.conf.

X-Windows starts properly now and my XP-Pen Star G640 works, but just barely. It doesn't detect proximity and the buttons do not trigger events. Proximity and click events do work in Windows 10, which I use to play some games. Mostly I run Linux Mint, using Inkscape to produce the graphics for games I program in Python(mcpalmer1980.itch.io) . Following is the contents of my /etc/X11/xorg.conf.d/50-digimend.conf:

---
Section "InputClass"
	Identifier "Ugee/XP-Pen tablets with Wacom driver"
	MatchUSBID "28bd:0094"
	# Exclude the original WP5540U which PID is reused by Ugee M540
	# NoMatchProduct "MousePen" <- commented out to allow x session to start
	MatchDevicePath "/dev/input/event*"
	Driver "wacom"
EndSection
---

Following is some output from xsetwacom :

xsetwacom --list devices
XP-PEN STAR G640 Mouse touch    	id: 16	type: TOUCH

xsetwacom --get 16 all
Option "Area" "0 0 32767 32767"
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Option "DebugLevel" "0"
Option "CommonDBG" "0"
Option "Suppress" "2"
Option "RawSample" "4"
Option "PressCurve" "0 0 100 100"
Option "Mode" "Absolute"
Property 'Wacom Hover Click' does not exist on device.
Option "Touch" "on"
Property 'Wacom Hardware Touch Switch' does not exist on device.
Option "Gesture" "off"
Option "ZoomDistance" "1000"
Option "ScrollDistance" "0"
Option "TapTime" "250"
Property 'Wacom Proximity Threshold' does not exist on device.
Option "Rotate" "none"
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Option "Threshold" "1"
Option "Serial" "0"
Property 'Wacom Pressure Recalibration' does not exist on device.
---

relevant log entries from Xorg.0.log:

---
[   587.705] (II) config/udev: Adding input device XP-PEN STAR G640 Mouse (/dev/input/mouse2)
[   587.705] (II) No input driver specified, ignoring this device.
[   587.705] (II) This device may have been added with another device file.
[   587.866] (II) config/udev: Adding input device XP-PEN STAR G640 Mouse (/dev/input/event16)
[   587.866] (**) XP-PEN STAR G640 Mouse: Applying InputClass "evdev pointer catchall"
[   587.866] (**) XP-PEN STAR G640 Mouse: Applying InputClass "Ugee/XP-Pen tablets with Wacom driver"
[   587.866] (II) LoadModule: "wacom"
[   587.867] (II) Loading /usr/lib/xorg/modules/input/wacom_drv.so
[... 21 lines cut ...]
[   593.077] (**) XP-PEN STAR G640 Mouse: Applying InputClass "Ugee/XP-Pen tablets with Wacom driver"
[   593.077] (II) Using input driver 'wacom' for 'XP-PEN STAR G640 Mouse'
[   593.077] (**) XP-PEN STAR G640 Mouse: always reports core events
[   593.077] (**) Option "Device" "/dev/input/event17"
[   593.136] (EE) XP-PEN STAR G640 Mouse: Invalid type 'stylus' for this device.
[   593.136] (EE) XP-PEN STAR G640 Mouse: Invalid type 'eraser' for this device.
[   593.136] (EE) XP-PEN STAR G640 Mouse: Invalid type 'cursor' for this device.
[   593.136] (II) XP-PEN STAR G640 Mouse: type not specified, assuming 'touch'.
[   593.136] (II) XP-PEN STAR G640 Mouse: other types will be automatically added.
[   593.136] (--) XP-PEN STAR G640 Mouse touch: maxX=32767 maxY=32767 maxZ=8191 resX=0 resY=0 
[   593.136] (II) XP-PEN STAR G640 Mouse touch: hotplugging dependent devices.
[   593.136] (EE) XP-PEN STAR G640 Mouse touch: Invalid type 'stylus' for this device.
[... cut ...]
[   593.169] (**) XP-PEN STAR G640 Mouse touch: (accel) keeping acceleration scheme 1
[   593.169] (**) XP-PEN STAR G640 Mouse touch: (accel) acceleration profile 0
[   593.169] (**) XP-PEN STAR G640 Mouse touch: (accel) acceleration factor: 2.000
[   593.169] (**) XP-PEN STAR G640 Mouse touch: (accel) acceleration threshold: 4
---

The XP-Pen seems to be the best and most popular budget drawing tablet right now, especially for OSU players(which I'm not). Therefore any help resolving this matter will be appreciated by far more people than just myself.

gonX, 2019-12-23:
FWIW, the `libinput` Xorg input driver seems to work very well with my G640, including setting tablet area, pen pressure, and customizable pen pressure curves with `xinput`. Very usable for both osu! and  actual drawing.

The tablet does not seem to benefit from the extra features that the `wacom` driver should be providing, but correct me if I'm wrong on that.

Christian2006-Dex, 2020-03-09:
Good I have an easypen i405 and in the graphic tablet program it does not detect it. I use Linux Mint 19.3 64bits cinnamon. I would greatly appreciate a help.