DIGImend/digimend-kernel-drivers #280: GAOMON S620

ekkl, 2019-05-07:
Does Digimed support model GAOMON S620?
https://www.aliexpress.com/store/product/GAOMON-S620-6-5-x-4-Inches-Digital-Pen-Tablet-Anime-Graphic-Tablet-for-Drawing-Playing/2629011_32998473127.html?spm=2114.12010612.8148356.1.5ce71685OZeacv

spbnick, 2019-05-26:
I don't have data on support of this exact model. It's likely to be a remake of a UC-Logic-based tablet, such as Huion H430P, which is well-supported, but I cannot guarantee that.

shika7200, 2019-07-31:
Can I sand for you one?

bydavenet, 2020-01-28:
This tablet should be almost the same as Huion HS64 which is already supported. I'll send diagnostics when I get to Linux PC.

bydavenet, 2020-02-24:
Device seems functional under Linux. Here I attach diagnostics. 

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4246066/descriptors.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4246067/frame_buttons.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4246068/hid_report_descriptors.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4246069/pen_buttons.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4246070/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4246071/pen_pressure.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4246072/probe.txt)

urnenfeld, 2020-07-29:
Have a Gaomon S620, product id matches your diagnostics: 
`Bus 002 Device 005: ID 256c:006d
`

DavidHusicka, 2020-07-29:
@urnenfeld yes and the tablet is fully functional. If you need to change your active area, please follow the README file on this repo.

spbnick, 2020-08-01:
Thank you everyone for contributing, closing the issue as the tablet seems supported.

closed 2020-08-01 by spbnick