DIGImend/digimend-kernel-drivers #282: Intey NY-BG14 (Hanwang) Graphic Tablet

platinnitrat, 2019-05-13:
I got this tablet from a friend who never used it and it only is recognized as a Mouse, so neither pressure nor the buttons on the frame do anything. 
According to my search there is basically no working software for this on any system (windows, mac or linux) (the manufacturer only sells through amazon where they probably bought all the positive reviews)
Anyway I followed the basic steps to gather the input data to work out a driver for this. 
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3173139/descriptors.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3173140/frame_buttons.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3173141/hid_report_descriptors.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3173142/pen_buttons.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3173144/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3173146/pen_pressure.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/3173147/probe.txt)

Vald3m, 2019-10-28:
Hi. 
I have tablet which looks exactly the same as yours [Parblo A609](https://www.parblo.com/products/a609). It has different product ID - 0b57:9091 But the same HID reports descriptor. And by default it works only as a mouse. For switching device into full mode (with sending pressure levels and tilting information) device expect HID feature request. 
I forked current project and implemented support for this tablet. The major issue that format of full report can't be parsed as normal HID report. Vendor is using improper bits order for all parts of data in report. So report needs to be recreated from scratch. 
For example full report with all parts looks like this:
`02 E1 57 50 0F CC FF E6 40 FF`
where 
`02` Report ID
`E1` - corresponds for pen buttons status and 'In-Range' status
`57 50` - X coordinate (bytes in a BigEndian order)
`0F CC` - Y coordinate (bytes in a BigEndian order)
`FF E6 40 FF` - contains information about pressure and tilting. 10-bits for pressure, 6-bits for tilt-X, 1-bit unused, 6-bits for tilt-Y. The rest bits looks like some counter. At least value in that part increased for each next reports.
I need to clean up my code and prepare it to PR.

carcinoma, 2020-01-17:
Hi,
i bought the tablet as well. Any chance to get support for it? 
Can i help with something?

Greetings

Vald3m, 2020-01-17:
Hi. Sorry for delay. I planned to do some improvement in the code. However it looks good even in current state. I just need to mention that pressure levels limited to 1024, because I couldn't find LSB of data for pressure, in HID report. You can clone project with my implementation from my repo by next link https://github.com/Vald3m/digimend-kernel-drivers/tree/parblo-a609
Later I will try to do PR to main project.

carcinoma, 2020-01-18:
Thanks a lot for it!

i installed the version and tried to get it running. Unfortunately no programm (GIMP, Krita, Inscape) is able to use the Tablet. They recognize it (and i configured it properly (set to "screen")), but they do not recognize the pen. evtest is also not able to see events but usbhid-dump see data.

Any suggestetions?
Thanks

carcinoma, 2020-01-18:
After changing in hid-hanwag line 147 from
"USB_DEVICE_ID_SIGNOTEC_VIEWSONIC_PD1011" 
to 
"USB_DEVICE_ID_INTEY_NY_BG14"
evtest receives events.
Still no pen in GIMP or Krita

Xorg.log shows now:
`[    40.571] (II) config/udev: Adding input device HanWang Tablet (/dev/input/event17)
[    40.571] (**) HanWang Tablet: Applying InputClass "libinput tablet catchall"
[    40.571] (II) Using input driver 'libinput' for 'HanWang Tablet'
[    40.571] (**) HanWang Tablet: always reports core events
[    40.571] (**) Option "Device" "/dev/input/event17"
[    40.571] (**) Option "_source" "server/udev"
[    40.633] (II) event17 - HanWang Tablet: is tagged by udev as: Tablet
[    40.679] (EE) event17 - HanWang Tablet: libinput bug: missing tablet capabilities: resolution. Ignoring this device.
[    40.679] (II) event17 - HanWang Tablet: device is a tablet
[    40.716] (II) event17 - failed to create input device '/dev/input/event17'.
[    40.716] (EE) libinput: HanWang Tablet: Failed to create a device for /dev/input/event17
[    40.716] (EE) PreInit returned 2 for "HanWang Tablet"
[    40.716] (II) UnloadModule: "libinput"
`

amicuchu, 2020-03-30:
@Vald3m I tested your driver and seems that modern versions doesn't like that the request to enable HID feature uses a buffer on stack, triggering an opps when the usb tries do dma
Tested in 5.5.13 with a Intey NY-BG14. Btw, I have a cd with windows drivers that was included in the box if you're interested

Vald3m, 2020-03-31:
Just did test with 5.5.13. Just upgrade kernel and DKMS automatically reinstall drivers and everything are working without any fixes. No issues in my setup. I'am using Ubuntu 18.04 with lowlatency kernel from https://kernel.ubuntu.com/~kernel-ppa/mainline/

Vald3m, 2020-04-06:
@amicuchu I found one laptop in my usage where I could reproduce the issue explained by you. I discovered, that this issue happened only on the laptop with a secure boot on. Anyway I did fixes for crash, and also found some bug in dkms related part mentioned by @carcinoma, which also were fixed. If it still relevant for you, please re-test latest version.

amicuchu, 2020-04-07:
Hid events are getting recolected but seems like they miss a lot of features like pressure or tilt. Only recognises one button

Vald3m, 2020-04-07:
@amicuchu  I would recommend to follow procedure of failure detection explained [here](https://digimend.github.io/support/howto/trbl/locating_failure/).
If you want to be sure that reports with pressure are enabled properly, build usbhid-dump tool and run next command:
`sudo usbhid-dump -es -m xxxx:yyyy` where `xxxx` is vendorId and `yyyy` is productId of your tablet.
after moving pen on device you should see raw reports from device. If first byte of reports is 02 - feature is enabled and pressure and tilt should be reported. If it 01 - driver is not loaded properly and default hid driver is used.
Few words about frame buttons. They are reported by report with ID 12 and you can see them with usbhid-dump. In initial implementation buttons were mapped to BTN_0 to BTN_3 but BTN_0 is an left mouse button and BTN_1 is right mouse button, and I decided to move them up by 16 positions, to be sure that the are not cross with mouse button. Now I decide to move them back into BTN_X area starting from BTN_2 (BTN_2, BTN_3, BTN_4, BTN_5), like author of original driver do for other tablets.
Please check latest commit.

amicuchu, 2020-04-07:
usbhid-dump:
Button reports from right to left:
`008:002:000:STREAM             1586271295.035269
 0C 00 00 01 00 00 00 00 00 00

008:002:000:STREAM             1586271296.695285
 0C 00 00 02 00 00 00 00 00 00

008:002:000:STREAM             1586271297.263290
 0C 00 00 04 00 00 00 00 00 00

008:002:000:STREAM             1586271297.815297
 0C 00 00 08 00 00 00 00 00 00
`
hid debug events:
`report (size 10) (numbered) =  0c 00 00 01 00 00 00 00 00 00
Button.0010 = 1
Button.0011 = 0
Button.0012 = 0
Button.0013 = 0

report (size 10) (numbered) =  0c 00 00 02 00 00 00 00 00 00
Button.0010 = 0
Button.0011 = 1
Button.0012 = 0
Button.0013 = 0

report (size 10) (numbered) =  0c 00 00 04 00 00 00 00 00 00
Button.0010 = 0
Button.0011 = 0
Button.0012 = 1
Button.0013 = 0

report (size 10) (numbered) =  0c 00 00 08 00 00 00 00 00 00
Button.0010 = 0
Button.0011 = 0
Button.0012 = 0
Button.0013 = 1
`

evtest (only first button reported):
`
Event: time 1586271497.474324, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90010
Event: time 1586271497.474324, type 1 (EV_KEY), code 271 (?), value 1
Event: time 1586271497.474324, -------------- SYN_REPORT ------------
`
This was tested before your button remap

amicuchu, 2020-04-07:
Reports from pen:
usbhid-dump
Without any presure
`008:002:000:STREAM             1586271955.340276
 02 E0 18 C0 0D 38 00 20 48 90`
Tilt is correctly reported on bytes 8 and 9

With full presure:
`008:002:000:STREAM             1586272244.949180
 02 E1 0D 6E 0D 20 FF E2 39 65`
Left button indicated in first half of byte 2 and presure is reported on byte 7

Pressing auxiliar buttons of pen:
`008:002:000:STREAM             1586272552.417407
 02 E4 0F 8B 1A 1C 00 20 49 90`
Auxiliar button indicated in first half of byte 2 as 4. While pressing the tip, presure is maxed so when byte 2 is E5 then byte 7 is 00 or FF always. Pen has one button that can be pressed on the bottom part or the upper part, in both cases report the same button

Sometimes events get on evtest show all parameters and some times only I get generic mouse ones. Seems that the delay you put in init can explain this

Vald3m, 2020-04-07:
Button reports looks correct (correspond to version what was before current). Even more HID descriptor was replaced correctly. Shame on me that I didn't test buttons in xinput, because I'm not using buttons in work. I just notice that they are detectable in evtest and forgot about them. I just retest my tablet and discover that xinput doesn't show any events from frame buttons.

amicuchu, 2020-04-08:
after add resolution parameter to hwdb udev so libinput can be happy, the tablet works correctly and xinput is getting data right like:

> xinput test 18
motion a[0]=16776464 a[1]=6158010 a[2]=0 a[3]=1 a[4]=-64 
motion a[0]=15289609 a[1]=7207561 a[2]=0 a[3]=0 a[4]=-48 
motion a[0]=13793748 a[1]=8243903 a[2]=0 a[3]=0 a[4]=-33 
motion a[0]=12294884 a[1]=9269438 a[2]=0 a[3]=-1 a[4]=-18 
motion a[0]=10792268 a[1]=10285365 a[2]=0 a[3]=-2 a[4]=-3

So I think that the driver is ready for PR and I suggest you send the new hwdb config to the udev mail list

[61-edev-local.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/4452681/61-edev-local.txt)

amicuchu, 2020-04-08:
About the buttons, libinput won't eat them if they aren't on a separate input device which udev recognises as `EVDEV_UDEV_TAG_TABLET_PAD`

Vald3m, 2020-04-09:
@amicuchu Let me start from your `xinput test` output. It looks weird, values is so huge. They should be in ranges reported by `evtest`. 
Now about `udev` rules.
I have been doing some research during last two days and I can provide you some recommendations.
1. Fix xorg rules to be sure that device recognized by xorg wacom driver. In this case two xinput devices will be created automatically without additional udev rules as additional benefit you can do the tablet configuration in the system settings. The only an issue what I found that in Gnome doesn't show button configuration in wacom setup menu. In KDE it possible to configure number of buttons and after that they will be available in Settings to remap.
2. I have done some investigation how tablet recognized by wacom configuration tools in KDE and Gnome. For KDE I reach the point that tablet is automatically recognized and display it in configuration menu (key point is content `/usr/share/wacomtablet`  directory). Gnome is using `libwacom` library to detect tablet configuration (`/usr/share/libwacom/`) but in my setup I couldn't reach proper detection (device detected by libwacom but unfortunately doesn't provide proper button configuration). What is possible for me now on Gnome is using `xsetwacom` to configure buttons.
I did new commits where I reverted buttons mapping back to BTN0...BTN3 and added `extra` directory which contains instructions how to add recognition in KDE and Gnome settings. I hope, I will fine some time to finish research in Gnome.

amicuchu, 2020-04-09:
About the xinput values, these are values calculated by libinput after consider the resolution because libinput garantees that values are in mm.

I was trying make libinput works because then can work on wayland, I know that probably wacom driver on X server can do a better job. Libinput provides functionality to libwacom when a tablet is used. Anyways, I will reconfigure my X server to use wacom drivers so everything works as expected

AlphaCraft9658, 2021-09-04:
I also have the INTEY NY-BG14 and I can't find any drivers. Please add this tablet!

ddmitry, 2022-08-13:
@platinnitrat, @AlphaCraft9658, @carcinoma, @Vald3m 

I know the topic is old, but I hope that can still be useful.
I bought this tablet in 2017 and haven't used it for a while. Wanted to install it on a new PC, googled it and found this topic. I dug out original drivers (downloaded back in the days) from my other PC and made a few screenshots how it looks like.

It's Windows and Mac only. Works fine on Windows 7 x64 and Windows 10 x64. Haven't tested anything else.

Drawing tablet INTEY NY-BG14 Drivers (Product id 9095)
https://www.dropbox.com/sh/7zgwtg9no73ougu/AADP0Jc1oAnZmpizVKzwVMTWa?dl=0

Hope it helps

mikulabc, 2023-11-11:
@ddmitry  thanks for your dropbox link
**Literally the only driver download i found on the entire internet :D**

AlphaCraft9658, 2024-01-08:
> @platinnitrat, @AlphaCraft9658, @carcinoma, @Vald3m
>
> I know the topic is old, but I hope that can still be useful. I bought this tablet in 2017 and haven't used it for a while. Wanted to install it [...]

I'm quite late to this, but thank you for the link. I might install it on my Windows install. Hopefully I can find something compatible for Linux.

AlphaCraft9658, 2025-09-15:
Has there been anything new in regards to this?

ddmitry, 2025-09-16:
> Has there been anything new in regards to this?

I doubt, but since you raised it I might try make it work.
I've got something to test it on, since the last time I posted here the drivers.