DIGImend/digimend-kernel-drivers #283: Huion HS64 tablet not supported

nekoellie, 2019-05-14:
Are there any plans to add support for the new Huion HS64. And is there any way I could maybe help contribute? (I don't know much about programming but I can follow instructions quite well ;D)

spbnick, 2019-05-26:
No plans so far, but if you could [collect some diagnostics](https://digimend.github.io/support/howto/trbl/diagnostics/) that will move its support closer to reality :) Thanks!

nekoellie, 2019-06-04:
I've collected some diagnostics here, not sure if they're 100% correct but I will gladly provide you with updated files if needed. Here's the product page: https://www.amazon.com/HUION-HS64-Battery-Free-Sensitivity-6-3x4inch/dp/B07PZV29S6 (The Huion page wouldn't load for me)
[HuionHS64.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/3251625/HuionHS64.zip)

kylegodbey, 2019-06-15:
I added the device id (0x006d) in the header file and treated it like the existing Huion Tablet and it seems to work fine, but I haven't tested the buttons or if pressure levels are being accurately measured, etc.

I'd submit a pull request, but I'd want to be sure everything works beforehand.

spbnick, 2019-06-17:
The PR is merged, closing now. Thank you, @kylegodbey.

spbnick, 2019-06-17:
I would appreciate a commit to the [tablets repo](https://github.com/DIGImend/tablets) describing this tablet.

closed 2019-06-17 by spbnick