DIGImend/digimend-kernel-drivers #284: RPM packages?

darkhog, 2019-05-15:
Can you create a set of RPM packages for distros based on that format (Fedora, openSuSE, Mandriva, etc.)? Compiling anything related to kernel from sources is risky and I don't want to end up with an unbootable system.

spbnick, 2019-05-16:
I did some research on this a while back, and haven't found a way to package a DKMS-based, or any module in an RPM. E.g. Fedora and RHEL discourage distribution of out-of-tree modules, and don't support them officially. If you know of any well-done RPM-packaged modules please give me a link, maybe I'll be able to use it as a reference. Otherwise, you'll have to continue using plain DKMS with DIGImend kernel drivers.

darkhog, 2019-05-16:
AFAIK nVidia&#39;s kernel drivers are packaged in a rpm on openSuSE, though I&#39;ve just always installed them via .run files from official nVidia&#39;s website due to the versions in the repo being usually few releases out of date.

spbnick, 2019-05-26:
NVIDIA drivers are an alien beast to Linux and an RPM package is just a hack on top to make it manageable. I'd rather not resort to that kind of approach, where a DKMS install would work more reliably and require less maintenance. Please write if you know of any open-source out-of-tree drivers being packaged into an RPM officially.

jn64, 2023-03-08:
[medusalix/xone](https://github.com/medusalix/xone) is a Linux kernel driver for Xbox One / X/S controllers. The upstream project uses dkms but there is a [Fedora Copr package sentry/xone](https://copr.fedorainfracloud.org/coprs/sentry/xone/) ([source repo](https://pagure.io/xone-rpm/tree/master)) that uses kmod/akmods.

As I understand it, dkms and [akmods](https://rpmfusion.org/Packaging/KernelModules/Akmods) are different ways to achieve the same thing. akmods is generally preferred by Fedora community because it builds actual rpm packages, i.e. using `dnf`, I can uninstall a specific version of the module that was built for a specific kernel.

While dkms builds and installs the kernel modules in its own way, which of course is perfectly usable, but it's just less familiar to the average Fedora user.

I don't have the knowledge for akmods packaging yet, but I intend to look into it further as I'm interested in using DIGImend on Fedora.

[Fedora does not allow external kernel module packages in their repos](https://docs.fedoraproject.org/en-US/packaging-guidelines/what-can-be-packaged/#_no_external_kernel_modules), but [Copr is ok](https://docs.pagure.org/copr.copr/user_documentation.html#what-i-can-build-in-copr), as are third-party repos like RPMFusion.