DIGImend/digimend-kernel-drivers #289: huion h420 works but is unusable

sumandroid12, 2019-06-18:
h420 is detected and all but i find it unusable to work with, I can barely write anything in linux.
On windows everything is smooth and responsive. Its like the curves and strokes are barely registered. 

Is the H420 not fully supported?

spbnick, 2019-06-18:
Hi @sumandroid12, the Huion H420 should be fully supported. Could you post the output of `xsetwacom list` to check if the driver is handling your tablet?

sumandroid12, 2019-06-18:
`xsetwacom list` is not detecting the tablet. i assumed that the wacom driver isnt supported.

sumandroid12, 2019-06-18:
I am using arch linux and had installed the digmend drivers from the AUR repository btw. are there any other steps?

spbnick, 2019-06-18:
I have no idea what the AUR repository contains, and I never used Arch, sorry. Could you post the output of `sudo modinfo hid-uclogic` and also `cat /sys/module/hid_uclogic/version`?

sumandroid12, 2019-06-18:
`sudo modinfo hid-uclogic` gives
```
filename:       /lib/modules/5.1.9-arch1-1-ARCH/extra/hid-uclogic.ko.xz
version:        10
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     DB4281FA059008C37426C66
alias:          hid:b0003g*v000028BDp00000042
alias:          hid:b0003g*v000028BDp00000094
alias:          hid:b0003g*v000028BDp00000075
alias:          hid:b0003g*v000028BDp00000055
alias:          hid:b0003g*v000028BDp00000071
alias:          hid:b0003g*v000028BDp00000074
alias:          hid:b0003g*v00002179p00000053
alias:          hid:b0003g*v00005543p00003031
alias:          hid:b0003g*v00005543p00000047
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006D
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        hid,usbhid
retpoline:      Y
name:           hid_uclogic
vermagic:       5.1.9-arch1-1-ARCH SMP preempt mod_unload
```
`cat /sys/module/hid_uclogic/version` gives
```cat: /sys/module/hid_uclogic/version: No such file or directory```

spbnick, 2019-06-18:
I can see that you have a fresh version of the driver installed, but the stock version loaded. Please unplug the tablet, run `sudo modprobe -r hid-uclogic`, reconnect the tablet, and see if it works.

sumandroid12, 2019-06-18:
after doing that 
`cat /sys/module/hid_uclogic/version` => `10`
but xsetwacom list still shows nothing. And the performance of tablet is still the same.

spbnick, 2019-06-18:
OK, please post the output of `xinput list` and your `/var/log/Xorg.0.log`.

sumandroid12, 2019-06-18:
`xinput list` =>
```
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ SynPS/2 Synaptics TouchPad                id=17   [slave  pointer  (2)]
⎜   ↳ PixArt USB Optical Mouse                  id=11   [slave  pointer  (2)]
⎜   ↳ HUION 420 Consumer Control                id=12   [slave  pointer  (2)]
⎜   ↳ HUION 420 Pad                             id=14   [slave  pointer  (2)]
⎜   ↳ HUION 420 Pen Pen (0)                     id=21   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Video Bus                                 id=7    [slave  keyboard (3)]
    ↳ Video Bus                                 id=8    [slave  keyboard (3)]
    ↳ Power Button                              id=9    [slave  keyboard (3)]
    ↳ HP Truevision HD: HP Truevision           id=15   [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard              id=16   [slave  keyboard (3)]
    ↳ HP Wireless hotkeys                       id=18   [slave  keyboard (3)]
    ↳ HP WMI hotkeys                            id=19   [slave  keyboard (3)]
    ↳ LITE-ON Technology USB NetVista Full Width Keyboard.      id=10   [slave  keyboard (3)]
    ↳ HUION 420 Consumer Control                id=13   [slave  keyboard (3)]
    ↳ HUION 420 Pen                             id=20   [slave  keyboard (3)]
```
[/var/log/Xorg.0.log](https://pastebin.com/S7faWbBu)

spbnick, 2019-06-18:
This should work now, even though you're using the libinput and not wacom driver. Try running `xinput test 21` (where 21 is the ID of the tablet's "Pen Pen" device), and see if it correlates with your input properly.

sumandroid12, 2019-06-18:
yup its working I can see it correlates with pressure but its responsiveness is nowhere near on windows. using the xournal++ app and i cant write anything as it wont respond properly. At first i thought that maybe i sucked at using it but after using the device on windows it worked perfectly, I could write with ease. Aside from the xournal app i have also tried krita, Its the same. Any way to get similar performance like on windows.

spbnick, 2019-06-18:
I doubt it could be something related to the driver then. When did you last try it on Windows? Could the pen battery be low, for example?

sumandroid12, 2019-06-18:
No i had just checked now.. anyway to configure or calibrate the device?

spbnick, 2019-06-18:
There is a way to calibrate the coordinates, but I doubt it will help you. Could you describe your problem with "responsiveness" exactly, or perhaps make a video showing the problem?

sumandroid12, 2019-06-18:
Hi i have found out the problem. I am using a dual monitor setup and after limiting the stylus to one screen i am able to draw with ease now. Since the control area was large hence i found it unnatural to write. On windows the stylus is limited to single screen. Feeling very dumb now.

Very sorry about wasting your time and thanks for helping me reach this conclusion :)

closed 2019-06-18 by sumandroid12

spbnick, 2019-06-18:
Glad to hear you figured it out :) Lesson to me: ask more about the specific problem :)

ktrace, 2019-06-18:
> Hi i have found out the problem. I am using a dual monitor setup and after limiting the stylus to one screen i am able to draw with ease now. Since the control area was large hence i found it unnatural to write.

Please tell the solution :) My son had same problem with dual monitor setup.

sumandroid12, 2019-06-18:
Hi @ktrace to set the input to a single screen I used `xinput map-to-output 21 HDMI1` 
Here `21` was my xinput id for the stylus pen as you can see above from `xinput list` and `HDMI1` the name of display from `xrandr`

ktrace, 2019-06-19:
@sumandroid12 great thanks!

AleMunin, 2022-10-03:
Apologies if my crude knowledge of external drivers is the real matter at hand, but I felt I should report this.

`sudo modprobe -r hid-uclogic` and plugging it back again did not solve the problem unfortunately. 

Unfortunately, while Huion 420 is listed as supported by digimend since version 5, it is only detecting the position of the pen when holding a button.

I made tests on three different machines:

- Ideapad with zorin 16 ( v11 .deb file from github)
- Very old itautec laptop with MX linux (v10 from their own repository)
- gen 3 i5 machine with zorin 15 (no drivers, just control group)

The only difference from having the drivers is that when I press the pen against the tablet, it knows where it is, but never ever detects any movement, otherwise without them, I needed to click the pen button for it to localize it, and constantly hold the button to move it.

Here are the details from Zorin 16 (.deb v11):

`xsetwacom --list devices
HUION 420 Pad pad                   id: 16    type: PAD       
HUION 420 Pen stylus                id: 17    type: STYLUS `

the cat command returns the same thing from the issue, only in my language

`cat: /sys/module/hid_uclogic/version: Arquivo ou diretório inexistente`

And here is the xinput
`xinput list
⎡ Virtual core pointer                        id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                  id=4    [slave  pointer  (2)]
⎜   ↳ Elan Touchpad                               id=10    [slave  pointer  (2)]
⎜   ↳ Logitech USB Optical Mouse                  id=8    [slave  pointer  (2)]
⎜   ↳ HUION 420 Pad pad                           id=14    [slave  pointer  (2)]
⎜   ↳ HUION 420 Pen stylus                        id=15    [slave  pointer  (2)]
⎜   ↳ HUION 420 Consumer Control                  id=16    [slave  pointer  (2)]
⎣ Virtual core keyboard                       id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard                 id=5    [slave  keyboard (3)]
    ↳ Video Bus                                   id=6    [slave  keyboard (3)]
    ↳ Power Button                                id=7    [slave  keyboard (3)]
    ↳ EasyCamera: EasyCamera                      id=9    [slave  keyboard (3)]
    ↳ Ideapad extra buttons                       id=11    [slave  keyboard (3)]
    ↳ HDA Intel PCH Headphone                     id=12    [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard                id=13    [slave  keyboard (3)]
    ↳ HUION 420 Consumer Control                  id=17    [slave  keyboard (3)]`

Do you guys feel posting the cat command here would help?

Am I doing something different and did not notice?