DIGImend/digimend-kernel-drivers #292: Stopped Working With Huion GT 191

BlenderBurst, 2019-07-11:
Hello,

I recently bought the Huion KAMVAS GT 191.

I installed this driver and it worked fine for a few hours. The pen was not mapped to the single screen though. I have a multiple screen setup and the pen mapping was stretched between all three screens. I was working on finding a terminal command to make it map to the single screen. If I turned off all my other monitors, It worked great.

A few hours after I installed it, It stopped working. The screen still works like a regular monitor, but the pen isn't working anymore. I plugged the tablet into a different computer and installed the driver, but it did not work.

I am on Linux Mint 19 Cinnamon.

If anyone can help I'd appreciate it. Thanks.

spbnick, 2019-07-11:
Could you please reconnect your tablet and post the output of `sudo dmesg|tail -n20`, `xinput list` and `xsetwacom list`?

BlenderBurst, 2019-07-11:
Ok sure.
Just this morning, I reinstalled the Driver again and it started working again. Although I don't know how long it will continue to work.

When I put, sudo dmesg|tail -n20, into the terminal:

[   23.663664] uclogic 0003:256C:006E.0002: No inputs registered, leaving
[   23.663684] uclogic 0003:256C:006E.0002: hidraw4: USB HID v1.11 Device [HID 256c:006e] on usb-0000:00:14.0-2/input1
[   26.213396] IPv6: ADDRCONF(NETDEV_UP): enp2s0: link is not ready
[   26.350812] r8169 0000:02:00.0 enp2s0: link down
[   26.350895] IPv6: ADDRCONF(NETDEV_UP): enp2s0: link is not ready
[   26.514736] IPv6: ADDRCONF(NETDEV_UP): wlp3s0: link is not ready
[   26.527308] IPv6: ADDRCONF(NETDEV_UP): wlp3s0: link is not ready
[   26.575191] IPv6: ADDRCONF(NETDEV_UP): wlp3s0: link is not ready
[   27.540066] wlp3s0: authenticate with 84:16:f9:fc:fd:51
[   27.551910] wlp3s0: send auth to 84:16:f9:fc:fd:51 (try 1/3)
[   27.553963] wlp3s0: authenticated
[   27.556094] wlp3s0: associate with 84:16:f9:fc:fd:51 (try 1/3)
[   27.559812] wlp3s0: RX AssocResp from 84:16:f9:fc:fd:51 (capab=0x31 status=0 aid=1)
[   27.559883] wlp3s0: associated
[   27.791135] IPv6: ADDRCONF(NETDEV_CHANGE): wlp3s0: link becomes ready
[   56.883224] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD data byte 8
[   57.044729] snd_hda_codec_hdmi hdaudioC1D0: HDMI: invalid ELD data byte 7
[   62.506438] Bluetooth: RFCOMM TTY layer initialized
[   62.506450] Bluetooth: RFCOMM socket layer initialized
[   62.506459] Bluetooth: RFCOMM ver 1.11

When I put, xinput list, Into the Terminal:

⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ HID 256c:006e Pen stylus                	id=10	[slave  pointer  (2)]
⎜   ↳ HID 256c:006e Pad pad                   	id=11	[slave  pointer  (2)]
⎜   ↳ PixArt Dell MS116 USB Optical Mouse     	id=14	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Power Button                            	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ Dell KB216 Wired Keyboard               	id=12	[slave  keyboard (3)]
    ↳ Dell KB216 Wired Keyboard               	id=13	[slave  keyboard (3)]
    ↳ Dell WMI hotkeys                        	id=15	[slave  keyboard (3)]

When I put, xsetwacom list, Into the Terminal:

HID 256c:006e Pen stylus        	id: 10	type: STYLUS    
HID 256c:006e Pad pad           	id: 11	type: PAD

spbnick, 2019-07-11:
This looks alright. Next time it stops working, please post the same and make sure to reconnect the tablet right before running `sudo dmesg | tail -n20`, so that the tablet initialization is captured. Thanks.

spbnick, 2019-07-11:
Oh, and could you tell how you were installing it?

BlenderBurst, 2019-07-11:
Thanks so Much for helping! Do you have any idea how I can get the pen mapping to work?
I installed it though Gdebi package installer.

BlenderBurst, 2019-07-11:
Never mind. I got the Pen mapping to work! I used the command:

xsetwacom set "HID 256c:006e Pen stylus" MapToOutput HEAD-2

BlenderBurst, 2019-07-11:
The tablet stopped working again. It seems like if I unplug the USB cable and then plug it back in, that's whats making it stop working.

Here's the Terminal Commands:

sudo dmesg | tail -n20:

[ 4416.314711] usb 1-2: USB disconnect, device number 3
[ 4581.151953] usb 1-2: new full-speed USB device number 9 using xhci_hcd
[ 4581.300954] usb 1-2: config 1 has an invalid interface number: 1 but max is 0
[ 4581.300956] usb 1-2: config 1 has no interface number 0
[ 4581.301699] usb 1-2: New USB device found, idVendor=0416, idProduct=a316
[ 4581.301701] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 4581.301703] usb 1-2: Product: WPM USB
[ 4581.301704] usb 1-2: Manufacturer: NUVOTON
[ 4581.303141] input: NUVOTON WPM USB as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/0003:0416:A316.0006/input/input27
[ 4581.360717] hid-generic 0003:0416:A316.0006: input,hidraw3: USB HID v1.10 Device [NUVOTON WPM USB] on usb-0000:00:14.0-2/input1
[ 4584.503496] usb 1-2: USB disconnect, device number 9
[ 4584.836073] usb 1-2: new full-speed USB device number 10 using xhci_hcd
[ 4585.013371] usb 1-2: New USB device found, idVendor=256c, idProduct=006e
[ 4585.013378] usb 1-2: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[ 4585.023621] input: HID 256c:006e Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:256C:006E.0007/input/input28
[ 4585.080380] input: HID 256c:006e Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:256C:006E.0007/input/input29
[ 4585.080991] uclogic 0003:256C:006E.0007: input,hidraw3: USB HID v1.11 Keypad [HID 256c:006e] on usb-0000:00:14.0-2/input0
[ 4585.085229] uclogic 0003:256C:006E.0008: No inputs registered, leaving
[ 4585.085452] uclogic 0003:256C:006E.0008: hidraw4: USB HID v1.11 Device [HID 256c:006e] on usb-0000:00:14.0-2/input1
[ 4589.973895] usb 1-2: USB disconnect, device number 10

xinput list:

⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ PixArt Dell MS116 USB Optical Mouse     	id=14	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Power Button                            	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ Dell KB216 Wired Keyboard               	id=12	[slave  keyboard (3)]
    ↳ Dell KB216 Wired Keyboard               	id=13	[slave  keyboard (3)]
    ↳ Dell WMI hotkeys                        	id=15	[slave  keyboard (3)]

When I put in, xsetwacom list, Nothing came up.

Thanks,

spbnick, 2019-07-12:
Unless you disconnected the tablet yourself before capturing the `dmesg` output above, it looks like the tablet disconnects itself, which is likely a hardware problem.

BlenderBurst, 2019-07-12:
So do you think its a hardware problem with the tablet or my computer?

BlenderBurst, 2019-07-12:
Hey, its working, now, and I think I know what the problem was.

I was plugging the USB cable into a Sony USB cable extender, and then plugging that into my Computer. But when I plugged the USB cable that came with the tablet, directly into my computer, Everything seems to work fine.

Thanks for helping.

spbnick, 2019-07-13:
Wonderful :) Glad to hear you figured it out!

closed 2019-07-13 by spbnick