DIGImend/digimend-kernel-drivers #296: Parblo A640: UC-Logic pen pressure works once Wacom X.org driver is enabled

hyiltiz, 2019-07-25:
A friend got this, and I wanted to try it out. The pen on the tablet acted as a mouse, but the pressure didn't work in GIMP even after the Input Device was set to Screen and the brush dynamics was set to Pressure Opacity despite `evtest` reported pressure values in entries like this:
```
Event: time 1564083030.673190, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 361
```

Tried installing the DIGImend dkms drivers, unloaded the drivers, unplugged the USB cable then reloaded the drivers then plugged back in. No success.

What worked is the simple `Xorg.conf` template on this site (including below to keep the post complete):
```
Section "InputClass"
    Identifier "Tablet"
    Driver "wacom"
    MatchDevicePath "/dev/input/event*"
    MatchUSBID "<VID>:<PID>"
EndSection
```
where  "<VID>:<PID>" is the ID from `lsusb` output.

The device parameters as reported by `uclogic-probe` were as follows:
```BASH
# lsusb
...
Bus 002 Device 002: ID 5543:0061 UC-Logic Technology Corp. A640
...
# uclogic-probe 2 2
M 55 00 43 00 2D 00 4C 00 6F 00 67 00 69 00 63 00
P 41 00 36 00 34 00 30 00
S 64 12 03 B0 73 12 43 03 00 FF 1F D8 13 0A 00 00 00 00 00
S 65 04 03 91 67
S 6E 04 03 30 00
S 79 07 03 BE FF AF 00 00
S 7A 08 03 02 04 00 54 00 00
S 7B 0C 03 48 00 4B 00 20 00 4F 00 6E 00
S C8 20 03 46 00 36 00 31 00 30 00 5F 00 4F 00 30 00 38 00 31 00 30 00 5F 00 41 00 50 00 5F 00 54 00
S C9 20 03 46 00 36 00 31 00 30 00 5F 00 4F 00 30 00 38 00 31 00 30 00 5F 00 41 00 50 00 5F 00 54 00
S CA 20 03 46 00 36 00 31 00 30 00 5F 00 4F 00 30 00 38 00 31 00 30 00 5F 00 41 00 50 00 5F 00 54 00
# uclogic-probe 2 2|uclogic-decode
    |           Manufacturer: UC-Logic
    |                Product: A640
 64 |        Params block #1: ????????
    |                          Max X: 29616
    |                          Max Y: 17170
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 79 |         Internal model: ??
 7b |         Buttons status: HK On
 c8 |        Params block #2: F610_O0810_AP_T
    |                          Max X: 3539014
    |                          Max Y: 12544
    |                   Max pressure: 48
    |                     Resolution: 95
    |
 c9 |       Firmware version: F610_O0810_AP_T
 ca |  Internal manufacturer: F610_O0810_AP_T
```

RudyTheNinja, 2019-07-31:
> What worked is the simple `Xorg.conf` template on this site (including below to keep the post complete):
>
> ```
> Section "InputClass"
>     Identifier "Tablet"
>     Driver "wacom"
>     MatchDevicePath [...]

Sorry but I'm kinda lost, how did you get it to work?
I can get the template myself, fill it out correctly but then.. where do I put it?

hyiltiz, 2019-08-04:
Where what file should be saved depends on what distribution you are running, but in most cases, it is definitely under `/etc/` not `/usr/`, and furthermore most likely to be at this path: `/etc/X11/xorg.conf.d/50-tablet.conf`. 

I wrote the report above as a Thank You note to the devs, and if they so desire and if all info provided above is enough, they can add in support to Parblo A640 by default so in newer versions of this package would include config files by default. Thus, some standard system admin tasks and know-hows were assumed, such as how to write a config file for `Xorg` and where to put it.

RudyTheNinja, 2019-08-05:
I can't seem to find the `xorg.conf.d` folder
Perhaps I'm missing something in order to have anything at that file path, any idea as to what that could be?

I installed the .deb provided here, could it be something that the driver package doesn't include?

hyiltiz, 2019-08-06:
Simply create the parent folders whenever necessary for the file path, e.g. to write to the file `/etc/X11/xorg.conf.d/50-tablet.conf`, do:
```bash
mkdir -p /etc/X11/xorg.conf.d/
gedit /etc/X11/xorg.conf.d/50-tablet.conf
```